thesofproject/sof #3068: [TGL][007-drop-stable]"error: ipc error for 0x50040000 size 124" occurred when system boot up.

YvonneYang2, 2020-06-19:
**Describe the bug**
"error: ipc error for 0x50040000 size 124" occurred when system boot up. Piplines cannot work after issue happened.

**To Reproduce**
1. Boot up system.
2. Check dmesg.

**Reproduction Rate**
100%

**Expected behavior**
system boots up without any error.

**Impact**
"error: ipc error for 0x50040000 size 124" occurred when system boot up.

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:1b2f2606
Revert 6a7c533
Revert cd0a323
FW: https://github.com/thesofproject/sof/tree/tgl-007-drop-stable commit: 3d8a0ac809
TPLG: same path as FW.  sof-tgl-sdw-max98373-rt5682-4ch.tplg rename to  sof-tgl-sdw-max98373-rt5682.tplg

**dmesg**
```
[    4.010819] intel-sdw intel-sdw.1: Slave status change
[    4.192153] asix 3-1.1:1.0 eth0: register 'asix' at usb-0000:00:14.0-1.1, ASIX AX88772B USB 2.0 Ethernet, 00:0e:c6:c6:86:b2
[    4.192179] usbcore: registered new interface driver asix
[    4.195196] asix 3-1.1:1.0 enx000ec6c686b2: renamed from eth0
[    4.281567] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:17:0
[    4.294920] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    4.297907] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50040000 size 124
[    4.297911] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 13
[    4.297913] sof-audio-pci 0000:00:1f.3: error: kcontrol value get for widget: 13
[    4.299469] sof-audio-pci 0000:00:1f.3: codec_mask = 0x4
[    4.299472] sof-audio-pci 0000:00:1f.3: display power enable
[    4.299473] sof_sdw sof_sdw: snd_hda_codec_device_new: entry
[    4.304974] snd_hda_codec_hdmi ehdaudio0D2: hdmi: pcm_num set to 9
[    4.304977] sof-audio-pci 0000:00:1f.3: display power disable
[    4.305187] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 4

[    4.305189] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 5

[    4.305190] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 6

[    4.305191] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 7

[    4.305192] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 8

[    4.305193] sof-audio-pci 0000:00:1f.3: display power enable
[    4.305197] sof-audio-pci 0000:00:1f.3: enable codec wakeup
[    4.307759] sof-audio-pci 0000:00:1f.3: disable codec wakeup
[    4.313418] sof-audio-pci 0000:00:1f.3: display power disable
[    4.314791] sof-audio-pci 0000:00:1f.3: display power enable
[    4.314795] sof-audio-pci 0000:00:1f.3: enable codec wakeup
[    4.317780] sof-audio-pci 0000:00:1f.3: disable codec wakeup
```
[dmesg.log](https://github.com/thesofproject/sof/files/4802306/dmesg.log)
[all-logger.log](https://github.com/thesofproject/sof/files/4802307/all-logger.log)

keyonjie, 2020-06-19:
The error message below should actually not impact the subsequent test? There is no initial model->data for smart_amp, that is the reason why we see this error. 

```
[    4.287123] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50040000 size 124
[    4.287127] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 13
[    4.287129] sof-audio-pci 0000:00:1f.3: error: kcontrol value get for widget: 13
```

We should either initialize the model->data for smart_amp, or change to just return a warning here.
@bkokoszx

keyonjie, 2020-06-19:
something like this, @bkokoszx 

```
diff --git a/src/audio/component.c b/src/audio/component.c
index f79fbc8e0..7963322ab 100644
--- a/src/audio/component.c
+++ b/src/audio/component.c
@@ -508,8 +508,10 @@ int comp_get_model(struct comp_dev *dev, struct comp_model_data *model,
                model->data_pos += bs;
 
        } else {
-               comp_err(dev, "comp_get_model(): !model->data");
-               ret = -EINVAL;
+               comp_warn(dev, "comp_get_model(): model data not allocated yet");
+
+               cdata->data->abi = SOF_ABI_VERSION;
+               cdata->data->size = 0;
        }
 
        return ret;
```

ClarexZhou, 2020-06-19:
Tested with PR #3069  ,  no ipc error after system boot up.
But still failed with same ipc error when trigger aplay

```
~$ aplay -Dhw:0,0 -f s16_le -c 2 -r 48000 sine_48k_32bit_10s_stereo_l5_r2.wav -vv -i
aplay: main:788: audio open error: Device or resource busy

```
dmesg
```
[  123.430422] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[  123.431264] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[  123.431278] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[  123.522089] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[  123.540902] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[  123.541093] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[  123.541097] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[  123.555229] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  123.555255] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  123.555258] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[  123.555260] sof-audio-pci 0000:00:1f.3: periods:1
[  123.557115] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2014001e successful
[  123.601853] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50030000 size 124
[  123.601862] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 13
[  123.601866] sof-audio-pci 0000:00:1f.3: error: failed kcontrol value set for widget: 13
[  123.601868] sof-audio-pci 0000:00:1f.3: error: restoring kcontrols after resume
[  123.601871] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -12
[  123.601897] intel-sdw intel-sdw.0: pm_runtime_get_sync failed in intel_startup, ret -16
[  123.601902] intel-sdw intel-sdw.0: ASoC: error at snd_soc_dai_startup on SDW0 Pin2: -16
[  123.601905] intel-sdw intel-sdw.0: ASoC: can't open DAI SDW0 Pin2: -16
[  123.601915]  SDW0-Playback: ASoC: BE open failed -16
[  123.601919]  Headphone: ASoC: failed to start some BEs -16
[  152.609529] sof-audio-pci 0000:00:1f.3: Copying cached debugfs
```
[dmesg.txt](https://github.com/thesofproject/sof/files/4803758/dmesg.txt)
[all-logger.txt](https://github.com/thesofproject/sof/files/4803759/all-logger.txt)

Test recipe:
Kernel:  Sof dev [commit: 1b2f2606] 
+ PR https://github.com/thesofproject/linux/pull/2208 for DP fix 
+ PR https://github.com/thesofproject/linux/pull/2184 for wov S0i2 
+ codec hack commit to verify this captured data 
+ revert regression on USB-C https://github.com/thesofproject/linux/commit/6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
+ revert https://github.com/thesofproject/linux/commit/d4b41f8bb  for reverting conflicts
+ revert regression on PINCTRL https://github.com/thesofproject/linux/commit/cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer

FW and tplg:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc1
https://github.com/thesofproject/sof/tree/releases/jsl/v3.0-rc1
+ PR #3053 for headset output issue after runtime PM suspended on 373 I2S
+ PR #2987 (373 SDW tplg)
+ PR #3069

sinahuang, 2020-06-19:
Test on Chromebook  with codec 98373 in I2S mode with PR https://github.com/thesofproject/sof/pull/3069, runtime PM status keep active after do some playback/capture, and then ""error: ipc error for 0x50030000 size 124" occurred.
```
[  913.378965] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  913.378974] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[  913.379169] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50030000 size 124
[  913.379173] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 1
[  913.379175] sof-audio-pci 0000:00:1f.3: error: failed kcontrol value set for widget: 1
[  913.379176] sof-audio-pci 0000:00:1f.3: error: restoring kcontrols after resume
[  913.379177] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -12
[  913.379187] sof-audio-pci 0000:00:1f.3: pcm: open stream 1 dir 0
[  913.379189] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes

```
Test recipe:
Kernel:  Sof dev [commit: 1b2f2606] 
+ PR https://github.com/thesofproject/linux/pull/2208 for DP fix 
+ PR https://github.com/thesofproject/linux/pull/2184 for wov S0i2 
+ codec hack commit to verify this captured data 
+ revert regression on USB-C https://github.com/thesofproject/linux/commit/6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
+ revert https://github.com/thesofproject/linux/commit/d4b41f8bb  for reverting conflicts
+ revert regression on PINCTRL https://github.com/thesofproject/linux/commit/cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer

FW and tplg:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc1
https://github.com/thesofproject/sof/tree/releases/jsl/v3.0-rc1
+ PR #3053 for headset output issue after runtime PM suspended on 373 I2S
+ PR #2987 (373 SDW tplg)
+ PR #3069
[dmesg.log](https://github.com/thesofproject/sof/files/4804512/dmesg.log)

bkokoszx, 2020-06-19:
@ClarexZhou @sinahuang 
I've just updated #3069. Could you try to reproduce error with it?

ClarexZhou, 2020-06-22:
Error can be fixed via latest #3069

mmaka1, 2020-06-22:
@YvonneYang2 Why is the smart-amp component provisioned with settings of size = 0? Is not the FW responding with error correct?

PR #3069 (2020-06-25): comp: return warning instead of error in comp_get_model()
In comp_get_model() function we should return warning in
case not allocated model->data.

fixes #3068
closed 2020-06-25 by lgirdwood