thesofproject/sof #3073: [BUG][TGL][DSM]Sound from speaker0,0 is distortion when playback and capture on speaker at the same time

sinahuang, 2020-06-22:
**Describe the bug**
Sound from speaker0,0 is distortion when playback and capture on speaker at the same time

**To Reproduce**
1.Run "aplay -Dhw:0,0 -f s32_le -c 2 -r 48000 sine_48k_32bit_10s_stereo_l5_r2.wav -d 5 & arecord -Dhw:0,0 -f s32_le -c 4 -r 48000 record.wav -d 5"
2.Check the sound output from speaker.

**Reproduction Rate**
5/5

[image]

**Expected behavior**
Sound from speaker0,0 should be normally when playback and capture on speaker at the same time

**Impact**
Sound from speaker0,0 is distortion when playback and capture on speaker at the same time

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix 
+ PR#2184 for wov S0i2 
+ codec hack commit to verify this captured data.
For volteer:
+ revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
+ revert  d4b41f8bb
+ revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc2 commit: b36dabe4
+ revert e35e91de652e06183c51ecc9814e7165f780c116
+ PR 3069
FW:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc2 commit: b36dabe4
+ revert e35e91de652e06183c51ecc9814e7165f780c116
+ PR 3069
TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

[sof_trace.log](https://github.com/thesofproject/sof/files/4811896/sof_trace.log)
[dmesg.log](https://github.com/thesofproject/sof/files/4811897/dmesg.log)

sinahuang, 2020-06-22:
Error not seen with tgl-006-drop-stable released FW + tplg on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc2 +pr3069 branch and 
 kernel https://github.com/thesofproject/linux/commits/topic/sof-dev +PR#2208 
PR#2184 +0001-hack-hack-98373-for-feedback-data-validation.patch.txt .

ClarexZhou, 2020-06-22:
Error only occurred if build fw with chrome_override changes in config
make tigerlake_chrome_overridedefconfig

Error not seen if build fw with only defaultconfig

jajanusz, 2020-06-22:
Problem is caused by not fitting into 120MIPS (there is 400mhz clock with defconfig and 120mhz in chrome config)

mengdonglin, 2020-06-22:
> Problem is caused by not fitting into 120MIPS (there is 400mhz clock with defconfig and 120mhz in chrome config)

@jajanusz So this is a performance restriction caused by lower clock frequency?

jajanusz, 2020-06-22:
> @jajanusz So this is a performance restriction caused by lower clock frequency?

Yes, we are testing some optimization fix, but if it's not enough we will need to use dynamic clock switching.

ClarexZhou, 2020-06-22:
issue only occurred after disable LPS.
Error not seen if build with LPS enabled, and disable LPRO.

lgirdwood, 2020-06-22:
@sinahuang @ClarexZhou are we using any demux in the topology ? We have an updated DEMUX pending that reduces from 70MCPS to 8.5 MCPS

sathya-nujella, 2020-06-22:
this is a blocker for us: we can not enable WoV + DSM feature at this moment..

RDharageswari, 2020-06-22:
We see I/O error when we open the (feedback)capture path while speaker playback
localhost ~ # arecord -Dhw:0,0 try.wav -c4 -fs32_le -d10 -r48000
Recording WAVE 'try.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: pcm_read:2151: read error: Input/output error

sathya-nujella, 2020-06-22:
> > @jajanusz So this is a performance restriction caused by lower clock frequency?
> 
> Yes, we are testing some optimization fix, but if it's not enough we will need to use dynamic clock switching.

Is it like using LPRO for only S0ix+Wov and for all other cases use HPRO?

keyonjie, 2020-06-23:
> @sinahuang @ClarexZhou are we using any demux in the topology ? We have an updated DEMUX pending that reduces from 70MCPS to 8.5 MCPS

Yes, we use demux there.  @sinahuang @ClarexZhou can you help to try if Bartosz's PR help for this? 
https://github.com/thesofproject/sof/pull/3000

sinahuang, 2020-06-23:
Error can be seen with PR https://github.com/thesofproject/sof/pull/3000 when build with LPS and LPRO enabled.

[image]
[record.zip](https://github.com/thesofproject/sof/files/4817063/record.zip)

ClarexZhou, 2020-06-23:
Error still can be reproduced with https://github.com/thesofproject/sof/commits/tgl-007-drop-stable, commit f16bf1eae6f6a.  Error can be reproduced even after disable LPRO in config with this commit.

Error not seen with PR#3000, even with LPRO enabled config

slawblauciak, 2020-06-23:
The mentioned PR will not fix the problem if the DSM processing is enabled.

lgirdwood, 2020-06-23:
> The mentioned PR will not fix the problem if the DSM processing is enabled.

Btw, there was another DEMUX corruption fix applied yesterday (sorry, cant remeber PR), we should be applying that and #3000 as dependencies in order to fix this issue (since they fix other related issues that can impact this issue).

ClarexZhou, 2020-06-24:
Error not seen with tgl-007-drop-stable, tag: releases/tgl/v7.0-rc5, commi: 1dd07a8a

ClarexZhou, 2020-06-29:
Keep open, we are waiting for PR on Master.

slawblauciak, 2020-07-28:
PR to be merged: https://github.com/thesofproject/sof/pull/3205

slawblauciak, 2020-08-18:
@ClarexZhou PR merged, good to close?

keqiaozhang, 2020-09-03:
Close for not reproducible.

closed 2020-09-03 by keqiaozhang