thesofproject/sof #3076: [BUG] firmware load failure on JSL Chrome

keqiaozhang, 2020-06-22:
**Describe the bug**
on JSL chrome, we found that with jasperlake_defconfig, firmware can be loaded successfully. but after changing the config with jasperlake_chrome_overridedefconfig, firmware failed to load with errors.

**To Reproduce**
build JSL firmware with xcc:
1. mkdir build
2. cd build 
3. cmake -DTOOLCHAIN=xt -DROOT_DIR=$your_path/xcc/install/builds/RG-2017.8-linux/X6H3CNL_2017_8/xtensa-elf ..
4. make jasperlake_defconfig
5. make jasperlake_chrome_overridedefconfig
6. make bin -j 8

**Reproduction Rate**
10/10
**Expected behavior**
firmware can be loadded even with chrome_override changes in config

**Impact**
Failed to load firmware 

**Environment**
firmware: 
https://github.com/thesofproject/sof/tree/releases/jsl/v3.0-rc1(commit: 3d8a0ac8)
+PR #https://github.com/thesofproject/sof/pull/3053 
+PR # https://github.com/thesofproject/sof/pull/2987  
* Platform: JSL chromebook

**Screenshots or console output**
```
[    7.459867] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000024 timedout
[    7.459871] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    7.460412] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[    7.460431] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[    7.460490] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000024
[    7.460498] sof-audio-pci 0000:00:1f.3: error: status = 0x4c4c4e55 panic = 0x00000000
[    7.460519] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000024
[    7.460523] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    7.460526] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    7.510874] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1d003c timedout
[    7.510877] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    7.510881] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 3
[    7.511124] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```

jajanusz, 2020-06-22:
ROM error says `Mailbox upstream has invalid alignment`.
Also FW loads with diagnostic driver + our tests, so there is some integration problem.

keqiaozhang, 2020-06-30:
Confirmed that this issue is related to MEU version and building process. 
below command line works for me. This issue can be closed.
cmake -DTOOLCHAIN=xt -DROOT_DIR=$CONFIG_PATH/xtensa-elf -DMEU_PATH=$MEU_PATH -DMEU_PRIVATE_KEY=$PRIVATE_KEY_PATH -DMEU_OPENSSL=/usr/local/ssl/bin/openssl ..

closed 2020-06-30 by keqiaozhang