thesofproject/sof #3092: [BUG][TGL]Arecord0,0 sound wav files is not smooth.

sinahuang, 2020-06-24:
**Describe the bug**
Arecord0,0 sound wav files is not smooth.

**To Reproduce**
1.Boot up system
2.aplay -Dhw:0,0 -f s32_le -c 2 -r 48000 sine_48k_32bit_10s_stereo_l5_r2.wav -d 5 & arecord -Dhw:0,0 -f s32_le -c 4 -r 48000 record.wav -d 5
3.Check the arecord files via audacity.

[32.zip](https://github.com/thesofproject/sof/files/4823414/32.zip)

**Reproduce rate**
5/5

**Expected result**
Arecord0,0 sound wav files is not smooth.

**Actual result**
Arecord0,0 sound wav files is not smooth.

[image]

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix 
+ PR#2184 for wov S0i2 
+ [0001-hack-hack-98373-for-feedback-data-validation.patch.txt](https://github.com/thesofproject/sof/files/4812034/0001-hack-hack-98373-for-feedback-data-validation.patch.txt)

For volteer:
+ revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
+ revert  d4b41f8bb
+ revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc2 commit: b36dabe4
+ revert e35e91de652e06183c51ecc9814e7165f780c116
+ PR 3069
FW:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc5 commit: https://github.com/thesofproject/sof/commit/1dd07a8aa27d22df56a43635975f4e151cbb4dbf

TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

jajanusz, 2020-06-24:
What if you revert all topology changes? 
These : https://github.com/thesofproject/sof/commit/462acfc3f59bc0e88b869359514281db31af388c
https://github.com/thesofproject/sof/commit/b36dabe499775ea063301af429cf31ab1aed1a2e

jajanusz, 2020-06-24:
Or just checkout to https://github.com/thesofproject/sof/commit/e35e91de652e06183c51ecc9814e7165f780c116 and use defconfig (not chrome)

sinahuang, 2020-06-24:
Error also can be seen on tgl-006-drop-stable .

[image]
[image]

bkokoszx, 2020-06-24:
@sinahuang 
Yours attached tgl-006-drop-stable waves looks correct. We capture current and voltage data from codec and they can be "less smooth".

On tgl-007-drop-stable (https://github.com/thesofproject/sof/issues/3092#issue-644336407), we can see small glithes:
[image]
which do not exist in tgl-006.

Can you retest this branch considering @jajanusz comments?

YvonneYang2, 2020-06-28:
Tested fw and tplg with latest tgl-007-stable(1dd07a8aa27d) + reverted 462acfc and b36dabe, issue can still be reproduced.
007 wav file : [007.zip](https://github.com/thesofproject/sof/files/4841702/007.zip)
[image]

and tested tgl-006-stable branch again
006 wav file: [006.zip](https://github.com/thesofproject/sof/files/4841704/006.zip)
[image]

YvonneYang2, 2020-06-30:
Tested it with nocodec kernel and tplg with same test recipe as https://github.com/thesofproject/sof/issues/3092#issue-644336407 , issue cannot be reproduced.

wav file: [nocodec.zip](https://github.com/thesofproject/sof/files/4849122/nocodec.zip)
[image]

bkokoszx, 2020-06-30:
@YvonneYang2 
If we cannot reproduce this issue with nocodec topology, it looks rather like some issue codec (wrong configuration?).
@keyonjie 
What do you think?

YvonneYang2, 2020-06-30:
> @YvonneYang2
> If we cannot reproduce this issue with nocodec topology, it looks rather like some issue codec (wrong configuration?).
> @keyonjie
> What do you think?

The test recipe is same as https://github.com/thesofproject/sof/issues/3092#issue-644336407, and kconfig please use https://github.com/thesofproject/kconfig/blob/master/kconfig-sof-nocodec.sh .
And the tplg please select sof-smart-amplifier-nocodec.tplg and rename to sof-tgl-nocodec.tplg.

bkokoszx, 2020-06-30:
@YvonneYang2 
Do you have speakers connected to the setup? Can you try to verify the correctness of the streaming wave?

ClarexZhou, 2020-07-01:
> @YvonneYang2
> Do you have speakers connected to the setup? Can you try to verify the correctness of the streaming wave?

Yes, speakers are connected. Yvonne used device is the same one which was used by Sina when reporting this bug.  But speakers are changed since old speakers are broken.

bkokoszx, 2020-07-01:
@ClarexZhou 
Ok, but are you able to hear stream from speakers properly?

ClarexZhou, 2020-07-02:
yes, max98373 speakers can output successfully. but has issue L speaker output R channel audio, R speaker output Left channel audio. This issue is tracking in https://github.com/thesofproject/linux/issues/2045#issuecomment-648636256

bkokoszx, 2020-07-02:
@ClarexZhou @YvonneYang2 
Can you try to reproduce this bug with following change:
``` diff
diff --git a/tools/topology/sof-smart-amplifier.m4 b/tools/topology/sof-smart-amplifier.m4
index 593b5afa2..07aa8b568 100644
--- a/tools/topology/sof-smart-amplifier.m4
+++ b/tools/topology/sof-smart-amplifier.m4
@@ -222,7 +222,7 @@ DAI_CONFIG(ALH, eval(SMART_ALH_INDEX + 1), eval(SMART_BE_ID + 1), SMART_ALH_CAPT
 #SSP SSP_INDEX (ID: SMART_BE_ID)
 DAI_CONFIG(SSP, SMART_SSP_INDEX, SMART_BE_ID, SMART_SSP_NAME,
        SSP_CONFIG(DSP_B, SSP_CLOCK(mclk, SSP_MCLK, codec_mclk_in),
-                     SSP_CLOCK(bclk, 9600000, codec_slave),
+                     SSP_CLOCK(bclk, 12288000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(8, 25, 15, 255),
                      SSP_CONFIG_DATA(SSP, SMART_SSP_INDEX, 24, 0, SMART_SSP_QUIRK)))
```

With that change I got wave without glitches:
[image]

ClarexZhou, 2020-07-03:
> @ClarexZhou @YvonneYang2
> Can you try to reproduce this bug with following change?
>
> ```diff
> diff --git a/tools/topology/sof-smart-amplifier.m4 b/tools/topology/sof-smart-amplifier.m4
> index [...]

Verified, seems error not seen with tplg update. 
[image]
[image]

Attach 007 tplg error seen wav and current updated tplg error not seen wav in below
[wav.zip](https://github.com/thesofproject/sof/files/4867789/wav.zip)

bkokoszx, 2020-07-03:
@keyonjie 
Do you remember the reason why we changed bclk from 9600000 to 12288000 before?

keyonjie, 2020-07-03:
we change from 32/32 bit to 24/25 bit format, that's why we need change the bclk from 12.288M to 9.6M.

Just checked the codec side, looks the codec expects pure 24bit (not 24bit in 25bit container), @ryans-lee can you confirm this?

sathya-nujella, 2020-07-06:
+ @RDharageswari

bkokoszx, 2020-07-07:
@keyonjie 
I've changed only container from 25 to 24 (without bclk) and it did not help.

keyonjie, 2020-07-08:
> @keyonjie
> I've changed only container from 25 to 24 (without bclk) and it did not help.

The reason why we set it to 25 rather than 24 is to avoid using MN divider. We have dependency between bclk and the container, you should change them together.

plbossart, 2020-07-08:
Can you clarify why a 12.288MHz bit clock is required? That's 8 channels of 32-bit data in each direction, do we really have such a configuration?

bkokoszx, 2020-07-14:
@ClarexZhou I've updated PR #3142. Can you try to reproduce above issue with updated PR?

@plbossart There is a codec requirement for bclk/fclk ratio, which should be equal to 256 when the number of channels is equal to 8.

ClarexZhou, 2020-07-17:
PR #3142 can fix

closed 2020-07-21 by lgirdwood

keqiaozhang, 2020-11-13:
Verified this issue with tgl 1.0 fix branch https://github.com/bkokoszx/sof/tree/tgl-010-hot-fix-1-revert-97377c3 on TGL Chromebook/I2S mode, this issue is not reproduced.

PR #3142 (2020-07-21): smart_amp: fix smart amplifier tplg
This commit changes SSP bclk rate from 9600000 to
12288000 ind ssp container to 32 bits order to fix
glitch issue.

fixes #3092