thesofproject/sof #3095: [FEATURE] Add multi sink support to testbench

cujomalainey, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
Currently testbench only supports single output from process component, this makes it hard to test components that split streams like crossover

**Describe the solution you'd like**
Add multiple sink support to testbench

**Describe alternatives you've considered**
N/A all other solutions require hacking the code block under test

**Additional context**
@singalsu CC for input

lgirdwood, 2020-06-24:
@ranj063 @singalsu any pointers ?

singalsu, 2020-06-25:
@cujomalainey Superb, thanks! This has been in my wish list of things to do! I think this should be fairly straightforward to do. Maybe add support for multiple sources too when convenient. Also feel free to clean up command line interface if needed. There's not much dependencies to fix them into scripts.

cujomalainey, 2020-06-25:
Excellent, that is good to hear, this will allow us to better test the crossover while the UUID stuff works its way through the PR process.

cujomalainey, 2020-09-28:
@johnylin76 this is done right?

closed 2020-12-03 by cujomalainey