thesofproject/sof #3096: [BUG] Firmware returns an error on TplgCompFree on test with mixer on capture pipline.

jdrzaszcz, 2020-06-25:
**Describe the bug**
When running simultaneous 3 x capture and 3 x playback on loopback signal on each SSP's with mixer component on one of the capture pipeline, firmware returns an error on TplgCompFree. Error is visible since commit a21cba16d99d642dd55fd3ebdd04201fb1017c40 which check state != COMP_STATE_READY in comp_free.
When sending ipc TplgCompFree on mixer component firmware return -22.

**Topology**
```
+-------------------------+
| +----+   +----+   +---+ |              +----+
| |Host|-->|Buff|-->|Dai| +------------->|SSPx|-----------------+
| +----+   +----+   +---+ |              +----+                 |
+-------------------------+                                     |
                                                                |
+-------------------------+                                     |
| +----+   +----+   +---+ |              +------+               |
| |Host|-->|Buff|-->|Dai| +------------->|SSPx+1|------------+  |
| +----+   +----+   +---+ |              +------+            |  |
+-------------------------+                                  |  |
                                                             |  |
+-------------------------+                                  |  |
| +----+   +----+   +---+ |              +------+            |  |
| |Host|-->|Buff|-->|Dai| +------------->|SSPx+2|---------+  |  |
| +----+   +----+   +---+ |              +------+         |  |  |
+-------------------------+                               |  |  |
                                                          |  |  |
                                                          |  |  |
                                                          |  |  |
                                                          |  |  |
                                    +----------------+    |  |  |
                                    | +----+   +---+ |    |  |  |
                        +-----------+ |Buff|<--|Dai| <----+  |  |
                        |           | +----+   +---+ |       |  |
                        |           +----------------+       |  |
                        |                                    |  |
                        |           +----------------+       |  |
                        |           | +----+   +---+ |       |  |
                        | +---------+ |Buff|<--|Dai| <-------+  |
                        | |         | +----+   +---+ |          |
                        | |         +----------------+          |
                        | |                                     |
+-----------------------v-v-----------------------+             |
| +----+    +----+    +-----+    +----+   +-----+ |             |
| |Host|<---|Buff|<---|Mixer|<---|Buff|<--| Dai |<--------------+
| +----+    +----+    +-----+    +-- -+   +-----+ |
+-------------------------------------------------+
```
**To Reproduce**
Run one of slim driver python tests from file:
test_01_mixer_capture.py

**Reproduction Rate**
100 %

**Expected behavior**
The released components should be in proper state and released without errors.

**Environment**
WHL RVP

Commit: 20ab16c36ece474fc3a55e87d7daa945bdcc9089 from SOF master branch.

**Example logs**
[dma_trace_1.log](https://github.com/thesofproject/sof/files/4829524/dma_trace_1.log)

[2020_06_25_08_03_00.zip](https://github.com/thesofproject/sof/files/4829519/2020_06_25_08_03_00.zip)

mmaka1, 2020-06-30:
Please note the test environment's sof-logger is one major version behind the fw side

logger ABI Version is 4:2:0
ldc_file ABI Version is 5:0:0

jdrzaszcz, 2020-07-14:
This also happens with topology with demux component in capture pipeline.
```
        pipe_plb
    +-------------------------------+
    | +------+   +------+   +-----+ |                                    +------+
    | | Host |-->| Buff |-->| Dai |------------------------------------->| SSPx |
    | +------+   +------+   +-----+ |                                    +------+
    +-------------------------------+                                        |
                                                                             |
                                                                             |
        pipe_cap                                                             |
    +----------------------+                                                 |
    | +-------+   +------+ |                                                 |
    | | Host1 |<--| Buff |<---+                                              |
    | +-------+   +------+ |  |      pipe_cap                                |
    +----------------------+  |  +----------------------------------+        |
                              +-----+-------+   +------+   +-----+  |     +------+
                                 |  | DeMux |<--| Buff |<--| Dai |<-------| SSPx |
        pipe_cap              +-----+-------+   +------+   +-----+  |     +------+
    +----------------------+  |  +----------------------------------+
    | +-------+   +------+ |  |
    | | Host2 |<--| Buff |<---+
    | +-------+   +------+ |
    +----------------------+
```
Sof master commit: 
Example log files: 318dc9f7e12ca48458ae3186d9a178c335e3b60a
[14_02_TestDemuxSimpleCapture48000Hz16b16b2ch.zip](https://github.com/thesofproject/sof/files/4917801/14_02_TestDemuxSimpleCapture48000Hz16b16b2ch.zip)

jdrzaszcz, 2020-07-16:
Yes , this PR #3197 fix this issue.

PR #3197 (2020-08-07): pipeline: change pipe reset propagation condition
This commit allows propagating pipeline reset to adjacent pipelines
assuming that they have the same schedule component.

fixes #3096
closed 2020-08-07 by lgirdwood