thesofproject/sof #3102: [BUG][TGL-tplg]"error: ipc error for 0x60010000 size 20" occurred when arecord with unsupported format.

YvonneYang2, 2020-06-28:
**Describe the bug**
"error: ipc error for 0x60010000 size 20" occurred when arecord with unsupported format.  Now it only supports s16 format.

**To Reproduce**
1. Boot up system
2. arecord -Dhw:0,5 -f s32_le -c 4 -r 48000 record.wav -vv -i
3. Check dmesg

**Reproduce rate**
5/5

**Expected result**
No ipc error in dmesg and terminal shows warning like:
```
~$ arecord -Dhw:0,99 -f s16_le -c 4 -r 48000 record.wav -vv -i
Recording WAVE 'record.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1299: Sample format non available
Available formats:
- S32_LE 
```
**Actual result**
Ipc error occurred in dmesg and terminal showed like:
```
~$ arecord -Dhw:0,5 -f s32_le -c 4 -r 16000 capture.wav -d 5 -vv -i
Recording WAVE 'capture.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 128
CHANNELS: 4
RATE: 16000
PERIOD_TIME: 63750
PERIOD_SIZE: 1020
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 255750
BUFFER_SIZE: 4092
BUFFER_BYTES: 65472
TICK_TIME: 0
```

dmesg
```
[  103.903465] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  103.903467] sof-audio-pci 0000:00:1f.3: periods:4
[  103.903755] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  103.903768] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  103.903777] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[  103.903785]  DMIC16kHz: ASoC: hw_params FE failed -12
[  103.903793]  dmic16k: ASoC: hw_free BE dmic16k
[  103.903801]  DMIC16kHz: ASoC: hw_free FE DMIC16kHz
[  103.903827]  dmic16k: ASoC: hw_free BE dmic16k
[  103.904025]  dmic16k: ASoC: close BE dmic16k
[  103.904080]  DMIC16kHz: ASoC: close FE DMIC16kHz
[  103.904100]  dmic16k: ASoC: BE dmic16k event 2 dir 1
[  103.904123]  DMIC16kHz: ASoC: BE capture disconnect check for dmic16k
[  103.904126]  DMIC16kHz: freed DSP capture path DMIC16kHz <- dmic16k
```

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix 
+ PR#2184 for wov S0i2 
+ [0001-hack-hack-98373-for-feedback-data-validation.patch.txt](https://github.com/thesofproject/sof/files/4812034/0001-hack-hack-98373-for-feedback-data-validation.patch.txt)
For TGL chrome book: 
+ revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   
+ revert  d4b41f8bb
+ revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  

FW:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc5 commit: https://github.com/thesofproject/sof/commit/1dd07a8aa27d22df56a43635975f4e151cbb4dbf

TPLG:sof-tgl-sdw-max98373-rt5682-4ch.tplg rename to sof-tgl-sdw-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in sdw mode.

[all_logger_.log](https://github.com/thesofproject/sof/files/4841768/all_logger_.log)
[dmesg.log](https://github.com/thesofproject/sof/files/4841769/dmesg.log)

YvonneYang2, 2020-06-28:
Tested it with tplg on tgl-006-stable branch, error not seen.

YvonneYang2, 2020-06-29:
Tested it with #3103 , format 24_le/32_le could work and ipc error occurred.

keyonjie, 2020-06-29:
Hi @naveen-manohar can you help to confirm if we will use the dmic16k for WoV on this SDW platform? if so, we might need your help to update this topology .m4 file, you can refer to sof-tgl-max98373-rt5682.m4 for the needed changes.

closed 2020-07-01 by lgirdwood