thesofproject/sof #3104: [BUG][TGL][HDA]Wov  pipeline will  be triggered automatically  no need clap hands or make some noise

Liviali155, 2020-06-29:
**Describe the bug**
Wov  pipeline will  be triggered automatically  no need clap hands or make some noise no matter using format  s32le/ s16le/ s24le

[image]

**To Reproduce**
1."sudo reboot" to reboot system
2.arecord -Dhw:0,7 -M -N -f s16_le -r 16000 -i -vvv -c2 --buffer-size=68000 ./tmp.wav
3.Do not clap hands or make some noise to trigger

**Reproduction Rate**
10/10

**Expected behavior**
Wov  pipeline will  not be triggered automatically  

**Impact**
Wov  pipeline will  be triggered automatically  no need clap hands or make some noise 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev(f6eb2d31)}
    * SOF: {Master(b49080c7)}
2) Name of the topology file
    * Topology: rename sof-hda-generic-4ch-kwd.tplg to sof-hda-generic-4ch.tplg

3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-U RVP with onboard codec ALC711 in HDA mode

[dmesg0629.log](https://github.com/thesofproject/sof/files/4843890/dmesg0629.log)
[sof-logger0629.log](https://github.com/thesofproject/sof/files/4843891/sof-logger0629.log)

lgirdwood, 2020-06-29:
@mrajwa any ideas ?

mrajwa, 2020-06-29:
@lgirdwood, @Liviali155 sounds like wrong config. Please verify `config.activation_threshold` which is defined in `detect_test_coef.m4`

You can also read it using sof-ctl and paste here and I will take a look

Liviali155, 2020-06-30:
After applied  #3109,there is wov volume control in alsamixer,Wov pipeline can be triggered  easier  and easier along with  the volume louder,but has another issue  https://github.com/thesofproject/sof/issues/3112,    close this issue  and continue follow up https://github.com/thesofproject/sof/issues/3112

closed 2020-06-30 by Liviali155