thesofproject/sof #3105: [BUG][TGL][HDA] Need to add PGA volume control on HDA kwd tplg

Liviali155, 2020-06-29:
**Describe the bug**
Need to add PGA volume control on HDA kwd tplg

**To Reproduce**
1."sudo reboot" to reboot system
2.alsamixer
3.Tab

**Reproduction Rate**
10/10

**Expected behavior**
There is PGA volume control on HDA kwd tplg

**Impact**
There is no PGA volume control on HDA kwd tplg
[image]

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).

Kernel: {sof-dev(f6eb2d31)}
SOF: {Master(b49080c)}
Name of the topology file
Topology: rename sof-hda-generic-4ch-kwd.tplg to sof-hda-generic-4ch.tplg
Name of the platform(s) on which the bug is observed.
Platform: TGL-U RVP with onboard codec ALC711 in HDA mode

[dmesg0629.log](https://github.com/thesofproject/sof/files/4843966/dmesg0629.log)
[sof-logger0629.log](https://github.com/thesofproject/sof/files/4843967/sof-logger0629.log)

lgirdwood, 2020-06-29:
@mrajwa @keyonjie any comments here ?

mrajwa, 2020-06-29:
@lgirdwood, it is enough to set it in the topology -> `# define(KFBM_TYPE, `vol-kfbm') `

keyonjie, 2020-06-30:
yes, I can submit a quick fix for this.

keyonjie, 2020-06-30:
@Liviali155 https://github.com/thesofproject/sof/pull/3109 is to fix this.

Liviali155, 2020-06-30:
> @Liviali155 #3109 is to fix this.

Tried with #3109,error not seen
[image]

ClarexZhou, 2020-07-01:
Close for PR merged

closed 2020-07-01 by ClarexZhou