thesofproject/sof #3112: [BUG][TGL]It's hard to trigger wov when set volume under 30 with format 16_le/24_le but s32_le can be auto triggered.

YvonneYang2, 2020-06-30:
**Describe the bug**
When set  wov capture volume under 60, wov can barely be triggered with format s16_le. 
set volume under 30,  wov can barely be triggered with format s24_le.  
At the same time, wov with format s32_le is so easy to be triggered even no need to make some obvious noise.
 

**To Reproduce**
1. Boot up system
2. Set capture volume "PGA9.0 9" 50/30/20.
3. Run wov test with format 16/24/32_le via command "arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -i -vvv -c2 --buffer-size=80000 ./tmp.wav"

**Reproduce rate**
5/5

**Expected result**
wov with format s16/s24/s32 can be triggered.

**Actual result**
When set  wov capture volume under 60, wov can barely be triggered with format s16_le. 
set volume under 30,  wov can barely be triggered with format s24_le.  
At the same time, wov with format s32_le is so easy to be triggered even no need to make some obvious noise.

**Environment**
Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix 
+ PR#2184 for wov S0i2 
+ [0001-hack-hack-98373-for-feedback-data-validation.patch.txt](https://github.com/thesofproject/sof/files/4812034/0001-hack-hack-98373-for-feedback-data-validation.patch.txt)
For TGL chrome book:
+ revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   
+ revert  d4b41f8bb
+ revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  

FW:
https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc5 commit: https://github.com/thesofproject/sof/commit/1dd07a8aa27d22df56a43635975f4e151cbb4dbf

TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

[sof-trace.log](https://github.com/thesofproject/sof/files/4849634/sof-trace.log)
[dmesg.log](https://github.com/thesofproject/sof/files/4849635/dmesg.log)

tlauda, 2020-06-30:
@YvonneYang2 Is it a regression?

ClarexZhou, 2020-06-30:
This isn't a regression, always behavior like this.

keyonjie, 2020-06-30:
Hi @tlauda @ClarexZhou I just root caused and will provide a PR to fix this soon, so you can not put effort to it atm.

keyonjie, 2020-06-30:
PR #3116 is submitted to fix the issue here.

ClarexZhou, 2020-06-30:
Verified, error not seen with PR #3116

keyonjie, 2020-07-08:
Close it as #3116 was merged.

closed 2020-07-08 by keyonjie