thesofproject/sof #3115: [BUG][TGL][HDA] WOV can  not be triggered after suspend/resume

Liviali155, 2020-06-30:
**Describe the bug**
 WOV can  not be triggered after suspend/resume

**To Reproduce**
1."sudo reboot" to reboot system
2.arecord -Dhw:0,7 -M -N -f s16_le -r 16000 -i -vvv -c2 --buffer-size=68000 ./tmp.wav
3.open another terminal to run "sudo rtcwake -m mem -s 10"
4.After wake up clap hands or make some noise to trigger wov

**Reproduction Rate**
10/10

**Expected behavior**
Wov can be triggered successfully

**Impact**
WOV can  not be triggered after suspend/resume
[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev(38849516)}
    * SOF: {Master(461f7d55)}
2) Name of the topology file
    * Topology: {rename  sof-hda-generic-4ch-kwd.tplg to sof-hda-generic-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL-U RVP with onboard codec ALC711 in HDA mode}

[dmesg0630.log](https://github.com/thesofproject/sof/files/4849853/dmesg0630.log)

Liviali155, 2020-07-01:
Issue also can be reproduced when use  "echo freeze > /sys/power/state" to suspend

1."sudo reboot" to reboot system
2.arecord -Dhw:0,7 -M -N -f s16_le -r 16000 -i -vvv -c2 --buffer-size=68000 ./tmp.wav
3.open another terminal to run  "sudo su" then  "echo freeze > /sys/power/state" to suspend
4. clap hands or make some noise to trigger wov

WOV can not be triggered,at the same time if press keyboard to wake up device,ipc error occured in dmesg

<details>
 <summary>dmesg_error.log</summary>

````
:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158548] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158851] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158853] sof-audio-pci 0000:00:1f.3: error: pcm params failed for DETECT12.0
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158854] sof-audio-pci 0000:00:1f.3: error: failed to set pcm params for widget DETECT12.0
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158855] sof-audio-pci 0000:00:1f.3: ASoC: PRE_PMU: DETECT12.0 event failed: -22
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.158857] sof-audio-pci 0000:00:1f.3: received event 2 for widget DETECT12.0
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.248807] OOM killer enabled.
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.248809] Restarting tasks ... 
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250681] sof-audio-pci 0000:00:1f.3: pcm: prepare stream 7 dir 1
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250684] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250689] sof-audio-pci 0000:00:1f.3: generating page table for 000000006e668646 size 0x426c0 pages 67
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250709] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250727] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250730] sof-audio-pci 0000:00:1f.3: period_bytes:0x10980
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250731] sof-audio-pci 0000:00:1f.3: periods:4
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250746] sof-audio-pci 0000:00:1f.3: stream_tag 2
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.250758] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.253509] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.253539] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.253833] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.253835] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 791136
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.253891] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 1
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.254848] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
Jul  1 05:35:29 ubuntu-Tiger-Lake-Client-Platform kernel: [   74.254854] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_ST````
</details>

Kernel: {sof-dev(38849516)}
SOF: {Master(461f7d5)}

After applied https://github.com/thesofproject/sof/pull/3116 issue still can be reproduced
[kern0701.log](https://github.com/thesofproject/sof/files/4856980/kern0701.log)
[sof-logger0701.log](https://github.com/thesofproject/sof/files/4856981/sof-logger0701.log)

keyonjie, 2020-07-01:
Thanks @Liviali155 

@mrajwa FYI.

Liviali155, 2020-07-03:
On TGL Volteer I2S with codec 98357 also has this issue  when use "echo freeze > /sys/power/state" to suspend with
 Kernel:
           Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
         + PR #2208 for DP fix 
         + PR #2184 for wov S0i2 
         + PR #2238
         + codec hack commit to verify this captured data.
         + revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
         + revert  d4b41f8bb
         + revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer
SOF:https://github.com/thesofproject/sof/tree/tgl-007-drop-stable commit: 1dd07a8+PR #3116 

<details>
 <summary>dmesg_error.log</summary>

````
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.157582] sof_rt5682 tgl_max98357a_rt5682: ASoC: resume work completed
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158138] sof-audio-pci 0000:00:1f.3: received event 1 for widget DETECT12.0
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158152] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158464] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158467] sof-audio-pci 0000:00:1f.3: error: pcm params failed for DETECT12.0
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158469] sof-audio-pci 0000:00:1f.3: error: failed to set pcm params for widget DETECT12.0
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158470] sof-audio-pci 0000:00:1f.3: ASoC: PRE_PMU: DETECT12.0 event failed: -22
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.158472] sof-audio-pci 0000:00:1f.3: received event 2 for widget DETECT12.0
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159688]  DMIC16k: ASoC: prepare FE DMIC16k
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159691]  dmic16k: ASoC: prepare BE dmic16k
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159717] sof-audio-pci 0000:00:1f.3: pcm: prepare stream 100 dir 1
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159718] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 100 dir 1
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159720] sof-audio-pci 0000:00:1f.3: generating page table for 00000000ca5b6aae size 0x426c0 pages 67
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159726] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159741] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
Jun 24 00:48:41 ubuntu-Volteer kernel: [  172.159743] sof-audio-pci 0000:00:1f.3: period_bytes:0x10980
````
</details>

[kern.log](https://github.com/thesofproject/sof/files/4867884/kern.log)
[sof-logger0703.log](https://github.com/thesofproject/sof/files/4867886/sof-logger0703.log)

Liviali155, 2020-07-08:
Error also can be reproduced with https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e on TGL-U RVP with onboard codec ALC711 in HDA mode and    TGL chrome laptop I2S with codec 98357

Liviali155, 2020-07-09:
Error still  can be reproduced with https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc8 ,commit: 2057409a on  TGL chrome laptop I2S with codec 98357

ClarexZhou, 2020-07-10:
Error also seen with https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc9 ,commit: be6c34790.  
Verified on 5682-max98357a on chrome OS, error also can be reproduced (frequency: 3/5) and frequcey :2/5 system will hang and can't wake up after enter s0ix.

ClarexZhou, 2020-07-10:
This issue is caused by alsa lib too old when building sof.  Error seen when build tplg with  libasound2 verison 1.1.0. Error not seen when build tplg with libasound2 1.1.3-5ubuntu0.2. Close

closed 2020-07-10 by ClarexZhou