thesofproject/sof #3120: [BUG] Glitches in ASRC playback stress test

singalsu, 2020-07-01:
**Describe the bug**
Glitch in playback stress test with sof-diagnostic-driver environment. The issue was created because it is not same that could happen in capture (#2981).

**To Reproduce**
Run the playback test with sufficient number of iterations for 8 to 8 kHz conversion, e.g. 100 iterations.

**Reproduction Rate**
About 5% occurrence.

**Expected behavior**
Clean sine waveform output.

**Impact**
If the glitch would happens in normal audio use case there could be a very short audible corruption happening every few minutes. 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.8.0-rc1-test+
    * SOF: git master (7b532d3712ff15c523f4dd52462f19aac5130e61)
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML

**Screenshots or console output**
[image]

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

singalsu, 2020-07-01:
@jdrzaszcz FYI

slawblauciak, 2020-08-10:
@singalsu any updates?

singalsu, 2020-08-10:
@slawblauciak I have tested a lot of things and work with debugging this nearly all the work time but I haven't found the root cause. I have a draft PR that makes the issue rare but it can still happen.

singalsu, 2020-08-12:
@slawblauciak @jdrzaszcz I'm not getting this error with git master firmware if I leave out option to force internal ssp. Now four rounds of 100x stress test passed. I wonder this issue can be closed?

slawblauciak, 2020-08-26:
@singalsu what does that option do exactly? When do we use it? Why would we want to use it?

lgirdwood, 2021-01-12:
@singalsu ping - any update ?

lgirdwood, 2021-01-22:
@singalsu ping - maybe for v1.8 ?

singalsu, 2021-01-22:
@lgirdwood Yep, unfortunately v1.8.

lgirdwood, 2021-05-10:
@singalsu any update here ?

singalsu, 2021-05-27:
I'm no more able to run this test. The python test scripts have changed beyond repairable for Linux host.

lgirdwood, 2021-05-27:
ok, lets close for now

closed 2021-05-27 by lgirdwood