thesofproject/sof #3128: [BUG][TGL]"DSP panic" when do pause/release of headset playbak&dmic capture

Liviali155, 2020-07-06:
**Describe the bug**
"DSP panic" when do pause/release of headset playbak&dmic capture
After error occured,all the pipeline can not work ,reboot can recovery

**To Reproduce**
1."sudo reboot" to reboot sysytem
2.open terminal2 to run "aplay -Dhw:0,1 -r48000 -c2 -fs24_le wav/48California_Dreaming.wav  -vv -i"
3 .open terminal1 to run " arecord -Dhw:0,99 -r48000 -c4 -fs32_le s30.wav  -vv  -i"
4.do pause/release of arecord(0,99) about 20 times
5.do pause/release of aplay(0,1)about 30times

**Reproduction Rate**
5/5

**Expected behavior**
There is no error occured when do pause/release

**Impact**
Connection timed out when do pause/release of aplay(0,1),dmesg show "DSP panic"

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
                 + PR#2208  for DP fix 
                 + PR#2184 for wov S0i2 
                 +PR#2238 for max98373-sdw jack switch
                  + codec hack commit to verify this captured data.
                  + revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a   for volteer
                 + revert  d4b41f8bb
                 + revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d  for volteer
    * SOF:   https://github.com/thesofproject/sof/tree/tgl-007-drop-stable   v7.0-rc6,commit: a40b46c1
                  + PR#3116 for KWD trigger volume
2) Name of the topology file
    * Topology: sof-tgl-max98357a-rt5682.tplg 
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL laptop  with codec 98357 in I2S mode

Liviali155, 2020-07-06:
Issue also occured when do pause/release of speaker  playbak&dmic capture

ClarexZhou, 2020-07-08:
Error not seen on TGL i2s max98373 5682

Liviali155, 2020-07-08:
"DSP panic" error not seen with  https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e  on TGL chromebook with codec 98357 in I2S mode
but still has issue https://github.com/thesofproject/sof/issues/2749  in  below scenerio

when do pause/release of speaker playbak&dmic capture
2.export TPLG=sof-tgl-max98357a-rt5682.tplg
./multiple-pause-resume.sh -r 100 (https://github.com/thesofproject/sof-test/blob/master/test-case/multiple-pause-resume.sh)

closed 2020-07-09 by Liviali155