thesofproject/sof #3129: [BUG] [TGL] No PGA volume contorl for max98373 in I2S mode speaker

ClarexZhou, 2020-07-06:
**Describe the bug**
No PGA volume contorller for max98373 I2S

**To Reproduce**
1. boot up system
2. ```alsamixer``` to check PGA volumes for speaker playback
3. No PGA volume controller for speaker playback.
4. Press tab to Capture to check, also no PGA volume for speaker catpure and DMIC48kHz 

```

ubuntu@TGL-Volteer-01:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: smart373-spk (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 2: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 3: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 4: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 5: HDMI4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@TGL-Volteer-01:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: smart373-spk (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 99: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 100: DMIC16k (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
[image]
[image]

**Reproduction Rate**
3/3

**Expected behavior**
Should provide PGA volume control

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev 94896acb
    * SOF: https://github.com/thesofproject/sof/tree/tgl-007-drop-stable   v7.0-rc6,commit: a40b46c1
2) Name of the topology file
    * Topology: sof-tgl-max98373-rt5682.tplg 
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL chrome laptop with max98373 in I2S mode

[amixerContents.txt](https://github.com/thesofproject/sof/files/4877598/amixerContents.txt)

ClarexZhou, 2020-07-06:
@keyonjie  could you please check first?

Liviali155, 2020-07-07:
On TGL  chrome laptop with codec 98357 in I2S mode,there is no PGA volume control in alsamixer for  EchoRef  and  DMIC48kHz 
[image]

```
ubuntu@ubuntu-Volteer:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 6: EchoRef (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 99: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 100: DMIC16k (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

plbossart, 2020-07-07:
Why would we need a volume control on an echo reference? the echo reference should be unmodified - though acoustic echo cancellation should adjust.

keyonjie, 2020-07-08:
We don't have to have PGA for every pipeline/stream, it depends on how do we expect to tune the volume for the specific stream. @sathya-nujella what is the requirement from the end user about this?

sathya-nujella, 2020-07-08:
> We don't have to have PGA for every pipeline/stream, it depends on how do we expect to tune the volume for the specific stream. @sathya-nujella what is the requirement from the end user about this?

We got the PGA/vol-gain ask from Customers: in DMIC48K cap and wov/DMIC16K pipelines.. not in echo ref stream.  As Pierre explained, it may be good we give un-modified data in echo ref stream?

keyonjie, 2020-07-28:
@sathya-nujella are you good to own and follow this issue?

sathya-nujella, 2020-07-28:
> @sathya-nujella @RDharageswari are you good to own and follow this issue? It will be a simple tplg change needed only.

Sure @keyonjie . Let me confirm if we really need gain in echo ref path or not. As per initial understanding, not needed. But will cross check once.

keyonjie, 2020-07-28:
> > @sathya-nujella @RDharageswari are you good to own and follow this issue? It will be a simple tplg change needed only.
> 
> Sure @keyonjie . Let me confirm if we really need gain in echo ref path or not. As per initial understanding, not needed. But will cross check once.

Thanks, so at least adding PGA between host and smart_amp for playback is aligned needed, please let us know if you need help on that.

sathya-nujella, 2020-08-10:
Hi @keyonjie 
1)
I confirm we don't need PGA/volume module in playback path or echo ref path. Confirmed with Google.
We don't need especially for this amp-Max98373 as it supports H/W gain control.
2)
FYI,
Only gain needed in WoV capture path, which is already there.

We can close this issue IMO.

sathya-nujella, 2020-08-10:
> Why would we need a volume control on an echo reference? the echo reference should be unmodified - though acoustic echo cancellation should adjust.

Thank you Pierre. I double confirmed with Google too , we don't need vol component in echo ref path: @keyonjie -FYI

closed 2020-08-10 by sathya-nujella