thesofproject/sof #3136: [BUG][TGL]"error: ipc error for 0x60060000 size 12"when check DSM  with  quickly pause/release

Liviali155, 2020-07-07:
**Describe the bug**
check echo reference with pause/release 
After error occured all the pipeline can not work,reboot can recovery

**To Reproduce**
1."sudo reboot" to reboot
2.open  terminal 1 to run "arecord -Dhw:0,3 -f s32_le -c 4 -r 48000 capture.wav -vv -i"
3.open terminal 2 to run "aplay -Dhw:0,2 -r 48000 -c 2 -fs32_le sine_48k_32bit_10s_stereo_l5_r2.wav -vv -i"
4.do pause/release of aplay quickly

**Reproduction Rate**
5/5

**Expected behavior**
No error occured when do pause/release

**Impact**
Input/output error at about 10th of pause/release,dmesg show "error: ipc error for 0x60060000 size 12"

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix
+ PR#2184 for wov S0i2
+PR#2238 for max98373-sdw jack switch
+ codec hack commit to verify this captured data.

SOF: https://github.com/thesofproject/sof/tree/tgl-007-drop-stable v7.0-rc6,commit: a40b46c
+ PR#3116 for KWD trigger volume
Name of the topology file
Topology: sof-tgl-sdw-max98373-rt5682.tplg
Name of the platform(s) on which the bug is observed.
Platform: TGL  with codec 98373 in SDW mode

Liviali155, 2020-07-07:
@RanderWang  please help to check

RanderWang, 2020-07-07:
> @RanderWang please help to check

@Liviali155  can it be reproduced on I2S mode ?

ClarexZhou, 2020-07-08:
Error not seen on TGL **I2S** max98373 5682

Liviali155, 2020-07-08:
Issue still can be reproduced  with https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e on  TGL chromebook with codec 98373 in SDW mode

<details>
 <summary>dmesg_error.log</summary>

````
[ 4498.499574] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[ 4498.877431] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2014001e successful
[ 4499.488779] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 4499.489015] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[ 4499.489018]  smart373-spk: ASoC: trigger FE cmd: 3 failed: -22
[ 4499.489560] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x14001e successful
[ 4499.489693] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 4499.489693]  smart373-spk: ASoC: trigger FE cmd: 4 failed: -22
[ 4500.584365] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 4500.584373] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[ 4500.584377]  echo: ASoC: trigger FE cmd: 0 failed: -22
[ 4500.584827] sdw_deprepare_stream: subdevice #0-Capture: inconsistent state state 3
[ 4500.584830] intel-sdw intel-sdw.1: sdw_deprepare_stream: failed -22
[ 4500.600071] sdw_disable_stream: subdevice #0-Playback: inconsistent state state 4
[ 4500.600075] intel-sdw intel-sdw.1: intel_trigger trigger 0 failed: -22
[ 4500.600077] intel-sdw intel-sdw.1: ASoC: error at snd_soc_pcm_dai_trigger on SDW
````
</details>

<details>
 <summary>sof_error.log</summary>

````
0 buffer       4.21        src/audio/buffer.c:172  comp_update_buffer_produce(), no bytes to produce, source->comp.id = 20, source->comp.type = 18, sink->comp.id = 19, sink->comp.type = 1
[8997758547.083334] (       26.927084) c0 buffer       4.22        src/audio/buffer.c:211  comp_update_buffer_consume(), no bytes to consume, source->comp.id = 23, source->comp.type = 2, sink->comp.id = 20, sink->comp.type = 18
[8997758589.010418] (       41.927082) c0 host         4.19          src/audio/host.c:382  host_copy_normal(): no bytes to copy
[8997758643.281250] (       54.270832) c0 host         3.12          src/audio/host.c:382  host_copy_normal(): no bytes to copy
[8997762467.395834] (     3824.114502) c0 dw-dma                 src/drivers/dw/dma.c:1056 ERROR dw_dma_get_data_size(): xrun detected
[8997762489.947918] (       22.552084) c0 dai          4.23           src/audio/dai.c:653  ERROR dai_report_xrun(): overrun due to no data available
[8997762510.416668] (       20.468750) c0 dai          4.23           src/audio/dai.c:694  ERROR comp_overrun(): sink->free = 224, copy_bytes = 0
[8997762533.593750] (       23.177084) c0 pipe         4.24      src/audio/pipeline.c:784  pipe trigger cmd 8
[8997762563.072918] (       29.479166) c0 dai          4.23           src/audio/dai.c:624  dai_comp_trigger(), XRUN
[8997762590.364584] (       27.291666) c0 dw-dma                 src/drivers/dw/dma.c:406  dw_dma_stop(): dma 1 channel 1 stop
````
</details>

[dmesg_0708.log](https://github.com/thesofproject/sof/files/4888751/dmesg_0708.log)
[sof-logger_0708.log](https://github.com/thesofproject/sof/files/4888752/sof-logger_0708.log)

Liviali155, 2020-07-10:
Issue still can be reproduced with https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc9,commit: be6c3479 on TGL chromebook with codec 98373 in SDW mode

WYL-12, 2020-12-01:
On TGL-Delbin-I2S-02 , using https://sof-ci-storage.sh.intel.com/release/sof/releases/tgl/v11.0-rc3/c6249fff6493_chrome_release/  ,issue can't be reproduced in below steps: 
1."sudo reboot" to reboot
2.open terminal 1 to run "arecord -Dhw:0,0 -f s32_le -c 4 -r 48000 capture.wav -vv -i"
3.open terminal 2 to run "aplay -Dhw:0,0 -r 48000 -c 2 -fs32_le sine_48k_32bit_10s_stereo_l5_r2.wav -vv -i"
4.do pause/release of aplay quickly

dmesg not found ipc error info, another error occured thesofproject/linux#2584.
[image]

lgirdwood, 2020-12-02:
@WYL-12 @slawblauciak can we close ?

slawblauciak, 2020-12-03:
Issue no longer reproduced.

closed 2020-12-03 by slawblauciak