thesofproject/sof #3149: [BUG][TGL][max98357-I2S]Continuous buzzing output from speaker when aplay via speaker then run Wov

Liviali155, 2020-07-08:
***Describe the bug***
Continuous buzzing output from speaker when aplay via speaker then run Wov

***To Reproduce***
1."sudo reboot" to reboot system
2.open terminal1 to run "aplay -Dhw:0,0 -r48000 -c2 -fs24_le  sine_48k_32bit_10s_stereo_l5_r2.wav -vv -i"
3.open  terminal2 to run "arecord -Dhw:0,100 -M -N -f s16_le -r 16000 -i -vvv -c2 --buffer-size=68000 ./tmp.wav"

```
ubuntu@ubuntu-Volteer:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 0: max357a-spk (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 2: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 3: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 4: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 5: HDMI4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```
ubuntu@ubuntu-Volteer:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 6: EchoRef (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 99: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofrt5682 [sof-rt5682], device 100: DMIC16k (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

***Reproduction Rate***
10/10

***Expected behavior***
No noise output from speaker when run Wov

***Impact***
Continuous buzzing output from speaker when run Wov

<details>
 <summary>sof_error.log</summary>

````
0 dai          11.59          src/audio/dai.c:362  dai_capture_params() fifo 10108
[539551824.114583] (       54.531250) c0 pipe         11.60     src/audio/pipeline.c:590  pipe prepare
[539552168.177083] (      344.062500) c0 kpb          11.55          src/audio/kpb.c:434  kpb_prepare()
[539552197.395833] (       29.218750) c0 kpb                         src/audio/kpb.c:218  kpb_allocate_history_buffer()
[539556352.135417] (     4154.739746) c0 memory                      src/lib/alloc.c:283  ERROR 169 blocks needed for allocation but only 168 blocks are remaining
[539557337.552083] (      985.416687) c0 kpb                         src/audio/kpb.c:245  kpb new memory block: 64256
[539557836.875000] (      499.322906) c0 kpb                         src/audio/kpb.c:245  kpb new memory block: 70144
[539557861.197917] (       24.322916) c0 kpb                         src/audio/kpb.c:294  kpb_allocate_history_buffer(): allocated 134400 bytes
[539557886.927083] (       25.729166) c0 kpb                         src/audio/kpb.c:1491 kpb_reset_history_buffer()
[539557912.187500] (       25.260416) c0 kpb                         src/audio/kpb.c:1386 kpb_clear_history_buffer()
[539570313.802083] (    12401.614258) c0 pipe         12.66     src/audio/pipeline.c:459  pipe params dir 1 frame_fmt 0 buffer_fmt 0 rate 16000
[539570347.083333] (       33.281250) c0 pipe         12.66     src/audio/pipeline.c:463  pipe params stream_tag 0 channels 2 sample_valid_bytes 2 sample_container_bytes 0
[539570434.531250] (       87.447914) c0 selector     12.62 ..../selector/selector.c:225  selector_params()
[539570484.895833] (       50.364582) c0 kpb          11.55          src/audio/kpb.c:400  ERROR kpb_params(): kpb has been already configured.
````
</details>

Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix
+ PR#2184 for wov S0i2
+PR#2238 for max98373-sdw jack switch
+ codec hack commit to verify this captured data.
+ revert regression on USB-C 6a7c533 for volteer
+ revert d4b41f8bb
+ revert regression on PINCTRL cd0a323 for volteer
SOF:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e

Name of the topology file
Topology: sof-tgl-max98357a-rt5682.tplg
Name of the platform(s) on which the bug is observed.
Platform: TGL chromebook with codec 98357 in I2S mode

[dmesg0708.log](https://github.com/thesofproject/linux/files/4889149/dmesg0708.log)
[sof-logger_0708.log](https://github.com/thesofproject/linux/files/4889157/sof-logger_0708.log)

Liviali155, 2020-07-08:
Error not seen when use firmware:
SOF: https://github.com/thesofproject/sof/tree/tgl-007-drop-stable v7.0-rc6,commit: a40b46c
+ PR#3116 for KWD trigger volume

ClarexZhou, 2020-07-08:
Error also seen on i2s 5682-max98373, pop noise can be observed.

Liviali155, 2020-07-09:
Error still can be reproduced with https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc8 ,commit: 2057409 on TGL chrome laptop I2S with codec 98357

ClarexZhou, 2020-07-09:
Verified on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc8 ,commit: 2057409 on i2s 5682-max98373 , only has once pop on max98373 output after "arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -i -vvv -c2 --buffer-size=80000 1.wav"

ClarexZhou, 2020-07-10:
Error not seen on I2S-5682-max98373 on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc9 ,commit: be6c34790 after remove KWD pipline PGA volume control

ClarexZhou, 2020-07-10:
Error not seen on I2S-5682-max98357a on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc9 ,commit: be6c347 after remove KWD pipline PGA volume control. Close

closed 2020-07-10 by ClarexZhou