thesofproject/sof #3160: [BUG][v7.0-rc9]Wov pipeline  can’t be started  when using format f24_le/s32_le for “Unable to install hw params:” during speaker playback

Liviali155, 2020-07-10:
**Describe the bug**
Wov pipeline  can’t be started  when using format f24_le/s32_le for “Unable to install hw params:” during speaker playback 

**To Reproduce**
1."sudo reboot" to reboot system
2.aplay -Dhw:0,0 -f s16_le -c 2 -r 48000 sine_48k_16bit_10s_stereo_l5_r2.wav  -vv -i
3.open another terminal to run “arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -i -vvv -c80000 1.wav” or “arecord -Dhw:0,100 -M -N -f s24_le -r 16000 -i -vvv -c80000 1.wav“

**Expected behavior**
Wov can be triggered normally

**Impact**
Wov pipeline  can’t be started for “Unable to install hw params:”

[image]

**Environment**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix 
+PR#2238 for max98373-sdw jack switch
+ https://github.com/mrajwa/linux/commit/5acaef7ee24d1fdfd368294b5a65a79518484155
+ https://github.com/mrajwa/linux/commit/fa927384c558ebecc07e5b35660569d323ef6e3c
+ https://github.com/mrajwa/linux/commit/34b60f83d781b1797823e462a2228f43b00a2dc5
For volteer:
+ codec hack commit to verify this captured data.

Firmware:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc9,commit: be6c3479

TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

<details>
 <summary>dmesg_error.log</summary>

````

[  102.509492] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  102.509509] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  102.514164] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  102.514170] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  102.514173] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  102.514177]  DMIC16k: ASoC: hw_params FE failed -22
[  102.514180]  dmic16k: ASoC: hw_free BE dmic16k
[  102.514184]  DMIC16k: ASoC: hw_free FE DMIC16k
[  102.514187] sof-audio-pci 0000:00:1f.3: pcm: free stream 100 dir 1
[  102.514194]  dmic16k: ASoC: hw_free BE dmic16k
[  102.514372]  dmic16k: ASoC: close BE dmic16k
[  102.514422]  DMIC16k: ASoC: close FE DMIC16k
[  102.514425] sof-audio-pci 0000:00:1f.3: pcm: close stream 100 dir 1
[  102.514435]  dmic16k: ASoC: BE dmic16k event 2 dir 1
[  102.514448]  DMIC16k: ASoC: BE capture disconnect check for dmic16k
[  102.514449]  DMIC16k: freed DSP capture path DMIC16k <- dmic16k
[  103.439771] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  103.439783] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  106.733721]  SSP1-Codec: ASoC: trigger BE SSP1-Codec cmd 0
[  106.747714]  smart373-spk: ASoC: post trigger FE smart373-spk cmd 0
[  106.747726] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  106.747736] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  106.747896] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_ST
````
</details>

<details>
 <summary>sof_error.log</summary>

````
dai          9.53           src/audio/dai.c:362  dai_capture_params() fifo 10108
[205008514.635417] (       16.093750) c0 pipe         9.54      src/audio/pipeline.c:590  pipe prepare
[205008604.531250] (       89.895836) c0 kpb          9.49           src/audio/kpb.c:434  kpb_prepare()
[205008613.541667] (        9.010417) c0 kpb                         src/audio/kpb.c:218  kpb_allocate_history_buffer()
[205011234.635417] (     2621.093750) c0 memory                      src/lib/alloc.c:283  ERROR 169 blocks needed for allocation but only 168 blocks are remaining
[205011345.364583] (      110.729164) c0 kpb                         src/audio/kpb.c:245  kpb new memory block: 64256
[205011663.697917] (      318.333344) c0 memory                      src/lib/alloc.c:283  ERROR 900 blocks needed for allocation but only 887 blocks are remaining
[205011976.562500] (      312.864594) c0 memory                      src/lib/alloc.c:283  ERROR 899 blocks needed for allocation but only 887 blocks are remaining
[205012289.427083] (      312.864594) c0 memory                      src/lib/alloc.c:283  ERROR 898 blocks needed for allocation but only 887 blocks are remaining
[205012755.572917] (      466.145844) c0 memory                      src/lib/alloc.c:283  ERROR 897 blocks needed for allocation but only 887 blocks are remaining
[205013068.645833] (      313.072906) c0 memory                      src/lib/alloc.c:283  ERROR 896 blocks needed for allocation but only 887 blocks are remaining
[205013381.093750] (      312.447906) c0 memory                      src/lib/alloc.c:283  ERROR 895 blocks needed for allocation but only 887 blocks are remaining
[205013693.489583] (      312.395844) c0 memory                      src/lib/alloc.c:283  ERROR 894 blocks needed for allocation but only 887 blocks are remaining
[205014005.885417] (      312.395844) c0 memory                      src/lib/alloc.c:283  ERROR 893 blocks needed for allocation but only 887 blocks are remaining
````
</details>

[dmesg0710.log](https://github.com/thesofproject/sof/files/4900611/dmesg0710.log)
[sof-logger_0710.log](https://github.com/thesofproject/sof/files/4900612/sof-logger_0710.log)

keyonjie, 2020-07-10:
Thanks for reporting @Liviali155 Looks the SRAM memory is used up in this scenario, @mrajwa can you can a look to this?

ClarexZhou, 2020-07-10:
Error not seen after remove WOV pipline PGA volume control via below:
```
diff --git a/tools/topology/sof-tgl-max98373-rt5682.m4 
b/tools/topology/sof-tgl-max98373-rt5682.m4
index fa96cb5bf..43a83bcd5 100644
--- a/tools/topology/sof-tgl-max98373-rt5682.m4
+++ b/tools/topology/sof-tgl-max98373-rt5682.m4
@@ -68,7 +68,7 @@ include(`sof-smart-amplifier.m4')
  # define channel
  define(CHANNELS, `4')
  # define kfbm with volume
-define(KFBM_TYPE, `vol-kfbm')
+define(KFBM_TYPE, `kfbm')
  # define pcm, pipeline and dai id
  define(DMIC_PCM_48k_ID, `99')
  define(DMIC_PIPELINE_48k_ID, `4')

```

Liviali155, 2020-07-15:
error not seen on  https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc10(fb801c8e5e3090406955d82e06c20fcdbfd302b4),closed it

closed 2020-07-15 by Liviali155