thesofproject/sof #3164: [BUG] Headphone microphone unavailable

ryzokuken, 2020-07-12:
Pulseaudio sink:

```
Source #11
	State: RUNNING
	Name: alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__hw_sofhdadsp__source
	Description: sof-hda-dsp Headphones Stereo Microphone
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 48000Hz
	Channel Map: front-left,front-right
	Owner Module: 6
	Mute: no
	Volume: front-left: 65533 / 100% / -0.00 dB,   front-right: 65533 / 100% / -0.00 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor of Sink: n/a
	Latency: 2672 usec, configured 40000 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = ""
		alsa.id = "HDA Analog (*)"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "sof-hda-dsp"
		alsa.long_card_name = "LENOVO-20R1S05400-ThinkPadX1Carbon7th"
		alsa.driver_name = "snd_soc_skl_hda_dsp"
		device.bus_path = "pci-0000:00:1f.3-platform-skl_hda_dsp_generic"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "02c8"
		device.string = "hw:sofhdadsp"
		device.buffering.buffer_size = "65472"
		device.buffering.fragment_size = "16320"
		device.access_mode = "mmap+timer"
		device.profile.name = "HiFi: hw:sofhdadsp: source"
		device.profile.description = "Headphones Stereo Microphone"
		device.description = "sof-hda-dsp Headphones Stereo Microphone"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	Ports:
		[In] Mic2: Headphones Stereo Microphone (priority: 200, not available)
	Active Port: [In] Mic2
	Formats:
		pcm
```

Emphasis on `[In] Mic2: Headphones Stereo Microphone (priority: 200, not available)`

The headphone speakers do work, and show up as "plugged", while the microphone shows up as "unplugged" in `pavucontrol`.

kv2019i, 2020-07-14:
@ryzokuken Can you provide output of alsa-info.sh when you have the headset connected?

ryzokuken, 2020-07-14:
@kv2019i the output of `alsa-info.sh` while the headphones are connected and I can listen to music on it (microphone still unavailable, as I mentioned)

```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.65
!!################################

!!Script ran on: Tue Jul 14 07:51:49 UTC 2020

!!Linux Distribution
!!------------------

Arch Linux \r (\l) Arch Linux release DISTRIB_ID=Arch DISTRIB_DESCRIPTION="Arch Linux" NAME="Arch Linux" PRETTY_NAME="Arch Linux" ID=arch HOME_URL="https://www.archlinux.org/" DOCUMENTATION_URL="https://wiki.archlinux.org/" SUPPORT_URL="https://bbs.archlinux.org/" BUG_REPORT_URL="https://bugs.archlinux.org/" LOGO=archlinux

!!DMI Information
!!---------------

Manufacturer:      LENOVO
Product Name:      20R1S05400
Product Version:   ThinkPad X1 Carbon 7th
Firmware Version:  N2QET19W (1.13 )
Board Vendor:      LENOVO
Board Name:        20R1S05400

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0003:00/status 	 15
/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/ACPI000E:00/status 	 15
/sys/bus/acpi/devices/INT0E0C:00/status 	 15
/sys/bus/acpi/devices/INT3400:00/status 	 15
/sys/bus/acpi/devices/INT3403:00/status 	 15
/sys/bus/acpi/devices/INT3403:01/status 	 15
/sys/bus/acpi/devices/INT34BB:00/status 	 15
/sys/bus/acpi/devices/INT3F0D:00/status 	 15
/sys/bus/acpi/devices/LEN0100:00/status 	 15
/sys/bus/acpi/devices/LEN0111:00/status 	 15
/sys/bus/acpi/devices/LEN0268:00/status 	 15
/sys/bus/acpi/devices/LEN0300:00/status 	 15
/sys/bus/acpi/devices/LNXPOWER:00/status 	 1
/sys/bus/acpi/devices/LNXPOWER:01/status 	 1
/sys/bus/acpi/devices/LNXPOWER:02/status 	 15
/sys/bus/acpi/devices/LNXPOWER:03/status 	 1
/sys/bus/acpi/devices/LNXPOWER:05/status 	 1
/sys/bus/acpi/devices/LNXPOWER:06/status 	 1
/sys/bus/acpi/devices/LNXPOWER:07/status 	 1
/sys/bus/acpi/devices/LNXPOWER:08/status 	 1
/sys/bus/acpi/devices/PNP0103:00/status 	 15
/sys/bus/acpi/devices/PNP0C02:00/status 	 3
/sys/bus/acpi/devices/PNP0C02:01/status 	 3
/sys/bus/acpi/devices/PNP0C02:06/status 	 3
/sys/bus/acpi/devices/PNP0C0A:00/status 	 31
/sys/bus/acpi/devices/PNP0C60:00/status 	 15
/sys/bus/acpi/devices/PRP00001:00/status 	 11
/sys/bus/acpi/devices/STM7308:00/status 	 15
/sys/bus/acpi/devices/SYNA8004:00/status 	 15
/sys/bus/acpi/devices/USBC000:00/status 	 15
/sys/bus/acpi/devices/device:17/status 	 15
/sys/bus/acpi/devices/device:26/status 	 15
/sys/bus/acpi/devices/device:27/status 	 15

!!Kernel Information
!!------------------

Kernel release:    5.7.8-arch1-1
Operating System:  GNU/Linux
Architecture:      x86_64
Processor:         unknown
[... 1492 lines cut ...]
[   40.595885] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI2 4 mapping ok
[   40.595890] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI3 5 mapping ok
[   40.595937] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
--
[   40.595972] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[   40.607253] input: sof-hda-dsp Headset Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input26
[   40.607295] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input27
[   40.607335] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input28
[   40.607522] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input29
[   40.607557] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input30
[   40.607591] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input31
[   40.616000] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
--
[ 7262.868188] process '/usr/bin/llpp' started with executable stack
[ 7612.883120] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883128] snd_hda_codec_realtek ehdaudio0D0: ASoC: can't open DAI Digital Codec DAI: -22
[ 7612.883133] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883139]  Digital Playback and Capture: ASoC: BE open failed -22
[ 7612.883142]  HDA Digital: ASoC: failed to start some BEs -22
[ 7612.883978] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883986] snd_hda_codec_realtek ehdaudio0D0: ASoC: can't open DAI Digital Codec DAI: -22
[ 7612.883992] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883998]  Digital Playback and Capture: ASoC: BE open failed -22
[ 7612.884002]  HDA Digital: ASoC: failed to start some BEs -22
```

ryzokuken, 2020-07-14:
@kv2019i I wonder if it has anything to do with

```
[ 7612.883978] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883986] snd_hda_codec_realtek ehdaudio0D0: ASoC: can't open DAI Digital Codec DAI: -22
[ 7612.883992] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[ 7612.883998]  Digital Playback and Capture: ASoC: BE open failed -22
[ 7612.884002]  HDA Digital: ASoC: failed to start some BEs -22
```

kv2019i, 2020-07-15:
@ryzokuken The problem is that headset is not reported to be connected:
```
state.sofhdadsp {
....
		name 'Headset Jack'
		value false
...
		name 'Mic Jack'
		value false
```

Former is the traditional headphone with a mic (aka headset) while latter is for plugging in a plain mic.  So this looks like a codec driver problem. There have been multiple fixes to the Thinkpad X1 codecs in upstream. Trying a recent live-image of Fedora or Ubuntu might be a good idea, as folks from these distros have been active in sending patches related to Thinkpad codecs.

If the live-images won't help, please raise this as a bug to ALSA (https://www.alsa-project.org/wiki/Bug_Tracking). Based on data so far, this is not an issue with SOF driver, but rather a codec issue, so ALSA bug tracker is the more appropriate place to track this.

ryzokuken, 2020-07-15:
@kv2019i I'll take this up with archlinux maintainers, thanks!

slawblauciak, 2020-07-21:
Confirmed codec configuration issue, closing.

closed 2020-07-21 by slawblauciak