thesofproject/sof #3165: [BUG] build-tools.sh fails after merging 3064

mrajwa, 2020-07-12:
**Describe the bug**
build-tools.sh fails to build topology parser.

**To Reproduce**
Just call ./scripts/build-tools.sh

**Reproduction Rate**
100%

**Expected behavior**
./scripts/build-tools.sh build all tools successfully.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: 318dc9f7e12ca48458ae3186d9a178c335e3b60a
2) Name of the topology file
    * Topology: n/a
3) Name of the platform(s) on which the bug is observed.
    * Platform: n/a

**Screenshots or console output**
` 50%] Building C object CMakeFiles/sof_tplg_parser.dir/tplg_parser.c.o
In file included from /home/admin/Dev/fork/sof/tools/tplg_parser/tplg_parser.c:20:0:
/home/admin/Dev/fork/sof/tools/tplg_parser/include/tplg_parser/topology.h:13:10: fatal error: sound/asoc.h: No such file or directory
 #include <sound/asoc.h>
          ^~~~~~~~~~~~~~
compilation terminated.`

mrajwa, 2020-07-12:
This is because #3064 adds fuzzer as a default build and fuzzer depends on tplg_parser.c which has never been built before. This file tplg_parser.c has compilation problems due to missing library, namely asoc.h. I am adding it here #3166

lgirdwood, 2020-07-14:
@jajanusz I assume we have a cmake variable that points to the kernel header path ? This needs to be used here too.

jajanusz, 2020-07-14:
> @jajanusz I assume we have a cmake variable that points to the kernel header path ? This needs to be used here too.

No we don't have. These headers are usually in 'system headers path' so they are picked up be compiler without specyfing include dirs (fe. when you install linux headers with apt-get). Probably that's why it worked for everyone that was using it in the past - the headers were there, and also are there on CI.

If you don't know what package to install and just want to add include dirs locally, you can do something like `CFLAGS=-I/path/to/folder/with/sound/soc`.

lgirdwood, 2020-07-14:
@mrajwa the package to install is usually called linux-headers. I wonder if this package is missing from install guide ?

jajanusz, 2020-07-14:
I haven't seen it in install guide.

marc-hb, 2020-07-16:
> I haven't seen it in install guide.

I think it's there, see https://github.com/thesofproject/sof/pull/3166#issuecomment-659581461

Cc: @crazoes

closed 2020-07-29 by mrajwa