thesofproject/sof #3169: [BUG][BDW]"error: firmware boot failure" when do module reload test

Liviali155, 2020-07-13:
**Describe the bug**
"error: firmware boot failure" when do module reload test

**To Reproduce**

1."sudo reboot" to reboot system
2.cd sof-test
3. cd test-case
4. export TPLG=/lib/firmware/intel/sof-tplg/sof-cht-max98090.tplg

**Reproduction Rate**
1 round:failed at 8/500
2 round:failed 4/500

**Expected behavior**
No error occured when do module reload test

**Impact**
"error: firmware boot failure" when do module reload test
[image]

<details>
 <summary>dmesg_error.log</summary>

````
[  114.520988] sof-audio-acpi 808622A8:00: booting DSP firmware
[  114.617728] usbcore: registered new interface driver snd-usb-audio
[  116.548560] sof-audio-acpi 808622A8:00: error: firmware boot failure
[  116.548606] sof-audio-acpi 808622A8:00: invalid header size 0xffffffff. FW oops is bogus
[  116.548610] sof-audio-acpi 808622A8:00: error: unexpected fault 0x90020000 trace 0x00003100
[  116.548617] sof-audio-acpi 808622A8:00: error: ipc host -> DSP: pending no complete no raw 0x4000000000003100
[  116.548620] sof-audio-acpi 808622A8:00: error: mask host: pending no complete yes raw 0x1
[  116.548623] sof-audio-acpi 808622A8:00: error: ipc DSP -> host: pending yes complete no raw 0x90020000
[  116.548627] sof-audio-acpi 808622A8:00: error: mask DSP: pending yes complete yes raw 0x3
[  116.548630] sof-audio-acpi 808622A8:00: error: failed to boot DSP firmware -5
[  116.548692] sof-audio-acpi 808622A8:00: error: sof_probe_work failed err: -5
[  116.715101] usbcore: deregistering interface driver snd-usb-audio
````
</details>

when catch sof-logger has below error
```
ubuntu@jf-bsw-cyn-max98090-3:~/sof-test/test-case$ sudo sof-logger -l /etc/sof/sof-*.ldc
error: Unable to open in file /sys/kernel/debug/sof/etrace
```

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev fa7850de}
    * SOF: {master:318dc9f7}
2) Name of the topology file
    * Topology: {sof-cht-max98090.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {BSW with onboard codec MAX98090 in I2S mode}

[dmesg0713.log](https://github.com/thesofproject/sof/files/4911694/dmesg0713.log)

slawblauciak, 2020-07-14:
@ktrzcinx mind sharing your view on this, could this be caused by the manifest changes?

ktrzcinx, 2020-07-14:
I suppose you are talking about extended manifest - it is disabled by default by now.

plbossart, 2020-07-14:
no it's unrelated to the firmware format proper, re-trying the boot works usually so it's a timing issue at the driver/hardware level.

plbossart, 2020-07-30:
After talking with the Coreboot folks, this is clearly a problem with Coreboot not initializing the PCI BAR1 and not programming the PMC registers. This will have to be re-checked if/when there is a Coreboot version, for now we will skip these tests in CI. Moving to P3 since it's not something we can fix.

slawblauciak, 2020-08-10:
Thanks for confirming this, @plbossart

plbossart, 2021-06-10:
let's close this, there is no line of sight for a new CoreBoot update

closed 2021-06-10 by plbossart