thesofproject/sof #3171: [BUG][BYT-NOCODEC]Input/output error  when aplay via (0,1) using format s24_le/s32_le

Liviali155, 2020-07-13:
**Describe the bug**
Input/output error  when aplay via (0,1) using format s24_le/s32_le
After error occured all the pipeline can work

**To Reproduce**
1."sudo reboot" to reboot system
2.cd sof-test
3.cd test-case
4.export TPLG=sof-byt-nocodec.tplg 
5.check-playback.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
10/10

**Expected behavior**
No error occured

**Impact**
Input/output error  when aplay via (0,1) using format s24_le/s32_le

[image]

```
ubuntu@jf-byt-mb-nocodec-1:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: PCM (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: PCM Deep Buffer (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

ubuntu@jf-byt-mb-nocodec-1:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: PCM (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
No error dmesg,no error sof error trace

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: {sof-dev fa7850de}
SOF: {master:318dc9f7}
Name of the topology file
Topology: {sof-byt-nocodec.tplg }
Name of the platform(s) on which the bug is observed.
Platform: {BYT MB with nocodec}
[sof-logger_0713.log](https://github.com/thesofproject/sof/files/4911968/sof-logger_0713.log)

[dmesg_0713.log](https://github.com/thesofproject/sof/files/4911967/dmesg_0713.log)

Liviali155, 2020-07-13:
BYT_MB_RT5682 also has this issue

mengdonglin, 2020-07-14:
sof-byt-nocodec.tplg is created from "sof-cht-nocodec.m4", updated in https://github.com/thesofproject/sof/pull/3080 to remove SRC for Baytrail and CherryTrail by @plbossart

lgirdwood, 2020-07-14:
@Liviali155 as @mengdonglin says lets reduce the BYT/CHT tests to reflect the new topologies.

plbossart, 2020-07-14:
we use the same pipe type of the two PCM devices, so not sure why one would support a format and not the other. This looks like a fundamental bug or a topology edit issue.

plbossart, 2020-07-29:
Can you retry with SOF PR #https://github.com/thesofproject/sof/pull/3245
Thanks!

Liviali155, 2020-07-29:
After applied #3245,issue still can be reproduced on BYT MB with nocodec
[image]
[0729dmesg.log](https://github.com/thesofproject/sof/files/4992291/0729dmesg.log)
[sof-logger0729.log](https://github.com/thesofproject/sof/files/4992293/sof-logger0729.log)

plbossart, 2020-07-29:
@Liviali155 work-around suggested with https://github.com/thesofproject/sof/pull/3257

Can you double-check the results? Also shouldn't this test be part of CI? It's been reporting no issues on this.

Liviali155, 2020-07-30:
Tried with #3257,error not seen

slawblauciak, 2020-08-10:
@Liviali155 that change has been merged, good to close now?

plbossart, 2020-08-10:
> @Liviali155 that change has been merged, good to close now?

I'd like to keep it for now, there's no rational explanation as to why reducing the period size solves the problem. I think we have more fundamental problem with scheduling.

slawblauciak, 2020-09-15:
@mengdonglin please assign it to somebody who maintains legacy platforms.

keqiaozhang, 2020-09-15:
Confirmed that this issue is not reproducible for a long time.

mengdonglin, 2020-09-15:
@plbossart @keqiaozhang Close this issue . We can monitor if this issue will happen again on BYT.

closed 2020-09-15 by mengdonglin