thesofproject/sof #3173: [Bug][RKL]FW load failed

ClarexZhou, 2020-07-14:
**Describe the bug**
FW load failed 

**To Reproduce**
"sudo reboot" to reboot

**Reproduction Rate**
5/5

**Expected behavior**
sof load successfully

**Impact**
Load failed

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel:Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 9eb3d58a)
+ https://github.com/thesofproject/linux/pull/2270
FW: master, commit 5564a902
Topology: rename sof-hda-generic-4ch.tplg
Name of the platform(s) on which the bug is observed.
Platform: RKL desktop with ALC711 in HDA mode

**Demsg**
```
[   62.526700] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   62.527220] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   62.527243] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x3010202 successful
[   62.720240] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   65.720746] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000006 timedout
[   65.720750] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   65.721293] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[   65.721300] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   65.721333] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000006
[   65.721346] sof-audio-pci 0000:00:1f.3: error: status = 0x00000014 panic = 0x00000000
[   65.721357] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000006
[   65.721365] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   65.721369] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   65.772247] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1d003c timedout
[   65.772250] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   65.772256] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[   65.772416] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```

ClarexZhou, 2020-07-14:
@RanderWang  @xiulipan Any ideas about this?

RanderWang, 2020-07-14:
> @RanderWang @xiulipan Any ideas about this?

now feature bit is 0x1FF, but only 0xFFF can work on RKL. So we need to discuss our feature bit setting for future chips.

ClarexZhou, 2020-07-15:
PR #3178 will fix this

lgirdwood, 2020-07-15:
Merged #3178

ClarexZhou, 2020-07-16:
Error still can be reproduced after PR #3178 merged. @RanderWang  Any ideas?

xiulipan, 2020-07-16:
@ClarexZhou Root cause to be the rimage submoudle update issue.
@jajanusz The git sub module issue

I see two PR https://github.com/thesofproject/sof/pull/3178 and https://github.com/thesofproject/rimage/pull/19
The latest rimage can work, but the deafult version rimage in sof did not have the latest update.
Any idea here with the git submoudule setting

below is log when I do a fresh git clone for sof and git submodule update.
```
$git submodule init
Submodule 'rimage' (https://github.com/thesofproject/rimage.git) registered for path 'rimage'
pxl@pxl-skle:/tmp/sof$ git submodule update
Cloning into '/tmp/sof/rimage'...
Submodule path 'rimage': checked out '9f3f2d7d714908c1f8f3f77baebe9175682f02bc'
```

xiulipan, 2020-07-16:
root cause to be rimage submodule issue. Fixed by update rimage to latest version.
But still see every when FW boot up. @RanderWang Please help to confirm and we may need to report a new issue.

```
[   62.798622] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[   64.808616] sof-audio-pci 0000:00:1f.3: error: firmware boot failure
[   64.808631] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.808648] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000211
[   64.808658] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.808661] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -5
```

xiulipan, 2020-07-17:
@jajanusz @RanderWang Update test result with rimage submodule fix.

FW with `make tigerlake_defconfig` could not boot on RKL RVP faill with 
`[   64.808648] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000211`

FW with `make alderlake_s_defconfig` could boot successfully.

But both FW has name `sof-tgl.ri`, will send a patch to fix the name confusing.

RanderWang, 2020-07-17:
thanks @xiulipan.  @ClarexZhou can you confirm and close the issue ? Thanks!

ClarexZhou, 2020-07-20:
Confirmed issue not seen after update rimage and build with ```make alderlake_s_defconfig ```.  Close

closed 2020-07-20 by ClarexZhou