thesofproject/sof #3176: [BUG][RKL] Audio output slowly than expected

ClarexZhou, 2020-07-14:
**Describe the bug**
Audio output on headset is slower than expected 

**To Reproduce**
1. "sudo reboot" to reboot
2.   time aplay -Dhw:0,0 -r 48000 -c2 -fs16_le wav/48California_Dreaming.wav -d 20 -vv -i

**Reproduction Rate**
5/5

**Expected behavior**
Audio output normally

**Impact**
Audio output slowly than expected
```

ubuntu@ubuntu-Rocket-Lake-Client-Platform:~$ time aplay -Dhw:0,0 -r 48000 -c2 -fs16_le wav/48California_Dreaming.wav -d 20 -vv -i
Playing WAVE 'wav/48California_Dreaming.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-hda-dsp' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
############################+                      | 54%

real    0m33.473s
user    0m0.009s
sys     0m0.014s

```
**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel:Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 9eb3d58a)
+ https://github.com/thesofproject/linux/pull/2270
FW: master, commit 5564a902, sof-tgl.ri, workarround issue https://github.com/thesofproject/sof/issues/3173
Topology: rename sof-hda-generic-4ch.tplg
Name of the platform(s) on which the bug is observed.
Platform: RKL desktop with ALC711 in HDA mode

[dmesg_aplay.txt](https://github.com/thesofproject/sof/files/4917718/dmesg_aplay.txt)

kv2019i, 2020-07-14:
Hmm, @plbossart @jajanusz , this could be the issue with the XTAL clock (24-versus-38.4Mhz). The ratio matches with test results (takes ~33secs to play out a 20sec clip). FYI for @RanderWang

plbossart, 2020-07-14:
We said we would also support the case of a RKL-CPU with a TGL-H chipset.
What CPU is that, CML?

RanderWang, 2020-07-15:
> We said we would only support the case of a RKL-CPU with a TGL-H chipset.
> What CPU is that, CML?

en, hot topic in BBS, what`s the configuration of RKL?  so...

ClarexZhou, 2020-07-20:
Error still seen on sof-8675d4cb. After issue #3173 fixed and build sof-tgl.ri with "make alderlake_s_defconfig"

ClarexZhou, 2020-07-21:
Error not seen on legacy HDA

RanderWang, 2020-07-21:
full of underrun log:
[694849172.604167] (     2000.000000) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[694851172.656250] (     2000.052124) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[694853172.656250] (     2000.000000) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695041175.781250] (   188003.125000) c0 dma-trace             src/trace/dma-trace.c:480  ERROR dtrace_add_event(): number of dropped logs = 93
[695041172.864583] (             nan) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695043172.708333] (     1999.843750) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695045172.604167] (     1999.895874) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695047172.552083] (     1999.947876) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695049172.604167] (     2000.052124) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected
[695051172.604167] (     2000.000000) c0 hda-dma            ..../intel/hda/hda-dma.c:872  ERROR hda_dma_link_check_xrun(): underrun detected

RanderWang, 2020-07-21:
The issue can be fixed by following patch. So what is the clk frequency on TGL-H ?
````C
diff --git a/src/platform/tigerlake/lib/clk.c b/src/platform/tigerlake/lib/clk.c
index 276fded13..8c6fab295 100644
--- a/src/platform/tigerlake/lib/clk.c
+++ b/src/platform/tigerlake/lib/clk.c
@@ -36,7 +36,7 @@ const struct freq_table *cpu_freq = platform_cpu_freq;
  */
 const struct freq_table platform_ssp_freq[] = {
        { 24576000, 24576 },
-       { 38400000, 38400 },
+       { 19200000, 19200 },
        { 96000000, 96000 },
 };
````

plbossart, 2020-07-21:
> The issue can be fixed by following patch. So what is the clk frequency on TGL-H ?

38.4 MHz. The last platform with 19.2 MHz was APL/GLK.

keyonjie, 2020-07-22:
is this for SSP playback? can we check the SCR and other SSP settings also? Probing to the SSP BCLK to probe the assumption that the output clock is wrong will help directly.

RanderWang, 2020-07-22:
> is this for SSP playback? can we check the SCR and other SSP settings also? Probing to the SSP BCLK to probe the assumption that the output clock is wrong will help directly.

NO, pure HDA platform. Thanks!

plbossart, 2020-07-22:
> > is this for SSP playback? can we check the SCR and other SSP settings also? Probing to the SSP BCLK to probe the assumption that the output clock is wrong will help directly.
> 
> NO, pure HDA platform. Thanks!

then why would changing the SSP settings impact the HDaudio playback? something's not quite right...

mengdonglin, 2020-07-24:
@plbossart @lgirdwood  suggest to introduce IPC to send clock information from driver on FW initialization

slawblauciak, 2020-09-08:
Platform not currently supported by SOF.

closed 2020-09-08 by slawblauciak