thesofproject/sof #3180: [BUG] SOF_VOLUME_LINEAR_ZC initial ramp is not as expected.

jdrzaszcz, 2020-07-14:
**Describe the bug**
When running simultaneous capture and playback on external loopback (same SSP) with Volume component on playback pipeline, inital ramp length is not equal to given initial ramp parameter.
It happens with SofVolumeRamp = SOF_VOLUME_LINEAR_ZC in about 50 % test cases.
It occurs after commit 3dc0ae97ac837bc2d6d41361180d24072ae9a37e.

Example:
INITIAL_RAMP = 400
Format 48kHz 24b/32/ 1ch:

[image]

Expected 19152.0 samples in initial ramp, but got: 18718
Ramp time 0.38995833333333335 is not equal expected ramp time = 0.399 on channel 1

**Topology**
```
+------------------------------------------------------+
| +------+   +-----+   +-- --+   +-----+   +---------+ |
| | Host |-->| Buf |-->| Vol |-->| Buf |-->| SSP Dai |---+
| +------+   +-----+   +-----+   +-----+   +---------+ | |
+------------------------------------------------------+ |
                                                         |
+-----------------------------------+                    |
| +------+   +-----+   +---------+  |                    |
| | Host |<--| Buf |<--| SSP Dai |<----------------------+
| +------+   +-----+   +---------+  |
+-----------------------------------+
```
**To Reproduce**
Run one of fails slim driver python tests:
[image]
[image]

**Reproduction Rate**
100 %

**Expected behavior**
Ramp length should be equal to given initial_ramp parameter

**Environment**
I checked it with WHL RVP platform.
On CI it occurs on also on APL_RVP and ICL_RVP. 

Commit: 318dc9f7e12ca48458ae3186d9a178c335e3b60a SOF master branch

**Example logs**
[dma_trace_1.log](https://github.com/thesofproject/sof/files/4918815/dma_trace_1.log)
[03_05_TestVolumeInitialRampSimple16000Hz32b32b2ch.zip](https://github.com/thesofproject/sof/files/4918819/03_05_TestVolumeInitialRampSimple16000Hz32b32b2ch.zip)

lgirdwood, 2020-07-14:
@jdrzaszcz Fwiw, Ramp lengths are not always guaranteed to be equal with ZC enabled, it tries to pick the best zero crossing point per channel up to a certain limit. This can be +/- a certain tolerance.

jdrzaszcz, 2020-07-14:
@lgirdwood The tolerance in our test is set to +/- 2 %, if it too strict, what value do you propose?

lgirdwood, 2020-07-14:
@singalsu what do you recommend ?

singalsu, 2020-08-06:
I tried to avoid 64 bit arithmetic in ramp calculations for speed. Also I recall I kept possibility for pretty slow ramps, maybe near 5s, I don't remember but can check. By going to 64 bit or decreasing max ramp length it could be made more accurate. Halving max. time would 1/2 the length difference.

Though I'm not aware of use case that requires specific ramp length. Are there? Would +/- 10% be reasonable?

jdrzaszcz, 2020-08-07:
I'm sure what the actual use cases look like (in some topology with src I saw 20 ms for type 2), in our tests the ramp length is between 100 and 1000 ms. I'll change tolerance to +/- 10 % and see how it works.

slawblauciak, 2020-08-10:
@jdrzaszcz any updates?

jdrzaszcz, 2020-08-11:
@slawblauciak I check it with 10 % tolerance and its ok.

closed 2020-08-11 by jdrzaszcz