thesofproject/sof #3189: [BUG][TGL][v7.0-rc10]Wov will repeat about 1~2s of the end of recorded file

Liviali155, 2020-07-15:
**Describe the bug**
Wov will repeat about 1~2s of the end of recorded file 

**To Reproduce**
1."sudo reboot" to reboot system
2.arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -i -vvv -c2 --buffer-size=96000 ./tmp.wav

**Expected behavior**
Recorded file normal

**Impact**
Wov will repeat about 1~2s of the end of recorded file 

**Environment**
Kernel:https://github.com/keyonjie/linux/commits/keyon-002(commit:082d5f6f2334824388b2e80de98e1f501106b756)
Firmware:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc10 ,(commit: fb801c8e5e3090406955d82e06c20fcdbfd302b4)

TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

[dmesg0715.log](https://github.com/thesofproject/sof/files/4922988/dmesg0715.log)
[sof-logger0715.log](https://github.com/thesofproject/sof/files/4922989/sof-logger0715.log)
[tmp.zip](https://github.com/thesofproject/sof/files/4922991/tmp.zip)

ClarexZhou, 2020-07-20:
Error also can be reproduced with kernel  https://github.com/thesofproject/linux/tree/tgl-007-20200716 and Firmware:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc10

ClarexZhou, 2020-07-20:
Error only occurred when stop capture via CTRL+C, error not occurred if run and stop stream via -d 10, ```arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -d 10-i -vvv -c2 --buffer-size=96000 ./tmp.wav```, verified 10 interactions

mrajwa, 2020-07-27:
@ranj063, can you look into that from driver perspective also?

keyonjie, 2020-07-28:
Hi @mrajwa this issue can be reproduced with 98357a machine on S0 also, please try it more times(3~5 times) to reproduce it.

slawblauciak, 2020-07-28:
To be clarified with the Chrome team, if forceful stopping of the pipeline is to be supported.

Liviali155, 2020-07-29:
On Chrome JSL with codec  in i2s mode also has this issue
[dmesgjsl.log](https://github.com/thesofproject/sof/files/4992882/dmesgjsl.log)
[sof-loggetjsl.log](https://github.com/thesofproject/sof/files/4992884/sof-loggetjsl.log)
[tmp.zip](https://github.com/thesofproject/sof/files/4992885/tmp.zip)

slawblauciak, 2020-07-31:
When the user forcefully stops the pipeline during draining, the flow of data is aborted. There's no likely scenario that could cause the firmware to keep sending repeating data afterwards. I would recommend verifying the way the buffer read in the driver.

mwasko, 2020-08-03:
@ranj063 can we ask for SOF driver expertise? The initial FW debug indicate problem at driver/kernel layer

ranj063, 2020-08-03:
> @ranj063 can we ask for SOF driver expertise? The initial FW debug indicate problem at driver/kernel layer

@mrajwa checking at my end but can you please elaborate on which part of your debug indicated a driver problem?

mrajwa, 2020-08-04:
@ranj063, FW pipeline is stopped but yet application gets additional data - clearly layers above FW copy more than they should.

mrajwa, 2020-08-07:
This is the application bug I have already submitted patches to ALSA, @lgirdwood  I think we can close it.

keyonjie, 2020-08-10:
Thanks for spotting this and providing solution, it is really nice catch! I added some comments about refining the PR https://github.com/alsa-project/alsa-utils/pull/51, @perexg can you help to take a look?

slawblauciak, 2020-08-10:
Will close when the PR gets merged.

mrajwa, 2020-10-12:
PR on alsa-utils has been merged, can we close it now?

keyonjie, 2020-10-12:
> PR on alsa-utils has been merged, can we close it now?

yes, think so, thanks for proceeded it.

closed 2020-10-12 by mrajwa