thesofproject/sof #3190: [BUG][tgl][v7.0-rc10]"error: ipc error for 0x60010000 size 20" after wake up when wov is paused

Liviali155, 2020-07-15:
**Describe the bug**
"error: ipc error for 0x60010000 size 20" after wake up when wov is paused
After error the pipeline of Wov still can work

**To Reproduce**
1."sudo reboot" to reboot system
2. open terminal1 to run"arecord -Dhw:0,100 -M -N -f s32_le -r 16000 -i -vvv -c2 --buffer-size=96000 ./tmp.wav -i"
3.trigger wov then pause it 
3.open terminal 2 to run "sudo rtcwake -m mem -s 10" or "sudo su ,echo freeze>/sys/power/state"
4.wait device to wake up 
5.press space key to release wov 

**Expected behavior**
No error occured

**Impact**
"error: ipc error for 0x60010000 size 20",wov can not be released
[image]

<details>
 <summary>dmesg_error.log</summary>

````
[  117.134636] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  117.134655] sof_rt5682 tgl_max98373_rt5682: ASoC: Scheduling resume work
[  117.134663] sof_rt5682 tgl_max98373_rt5682: ASoC: starting resume work
[  117.141602] sof_rt5682 tgl_max98373_rt5682: ASoC: resume work completed
[  117.141698] sof-audio-pci 0000:00:1f.3: received event 1 for widget DETECT10.0
[  117.141705] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  117.141999] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  117.142000] sof-audio-pci 0000:00:1f.3: error: pcm params failed for DETECT10.0
[  117.142001] sof-audio-pci 0000:00:1f.3: error: failed to set pcm params for widget DETECT10.0
[  117.142002] sof-audio-pci 0000:00:1f.3: ASoC: PRE_PMU: DETECT10.0 event failed: -22
[  117.142002] sof-audio-pci 0000:00:1f.3: received event 2 for widget DETECT10.0
````
</details>

<details>
 <summary>sof_error.log</summary>

````
0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[186713629.427083] (       24.531250) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[186713654.739583] (       25.312500) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186713678.750000] (       24.010416) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[186713703.177083] (       24.427084) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[186713791.562500] (       88.385414) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186713816.770833] (       25.208334) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[186713841.354167] (       24.583334) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[186713866.614583] (       25.260416) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186713890.625000] (       24.010416) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 536870912
[186713922.812500] (       32.187500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186713946.875000] (       24.062500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[186713970.520833] (       23.645834) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[186713995.833333] (       25.312500) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186714019.843750] (       24.010416) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 536870912
[186714052.083333] (       32.239582) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186714076.093750] (       24.010416) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[186714099.843750] (       23.750000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[186714125.989583] (       26.145834) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[186714150.000000] (       24.010416) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 536870912
[186714181.458333] (       31.458334) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
````
</details>

**Environment**
Kernel:https://github.com/keyonjie/linux/commits/keyon-002(commit:082d5f6f2334824388b2e80de98e1f501106b756)
Firmware:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc10 ,(commit: fb801c8e5e3090406955d82e06c20fcdbfd302b4)

TPLG:sof-tgl-max98373-rt5682.tplg
Platform:Chromebook with codec 98373 in I2S mode.

[dmesg_0715.log](https://github.com/thesofproject/sof/files/4923143/dmesg_0715.log)
[sof-logger_0715.log](https://github.com/thesofproject/sof/files/4923144/sof-logger_0715.log)

Liviali155, 2020-07-15:
Error also seen on 
https://github.com/thesofproject/sof/tree/tgl-007-drop-stable   v7.0-rc6,commit: a40b46c1
 
+ PR#3116 for KWD trigger volume

ClarexZhou, 2020-07-16:
Error still seen on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc10, commit: fb801c8e
Kernel: https://github.com/thesofproject/linux/tree/tgl-007-20200716, commit: 70ec4f52b

```
[ 3788.947293] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40080000: GLB_PM_MSG: unknown type
[ 3788.947997] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3788.948019] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3788.948030] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40080000: GLB_PM_MSG: unknown type
[ 3788.948035] sof-audio-pci 0000:00:1f.3: Current DSP power state: D0I0
[ 3788.948042] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 3788.953530] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 3788.953820] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[ 3788.953827]  DMIC16k: ASoC: trigger FE cmd: 0 failed: -22
[ 3788.953891]  DMIC16k: ASoC: hw_free FE DMIC16k
[ 3788.953898] sof-audio-pci 0000:00:1f.3: pcm: free stream 100 dir 1
[ 3788.953905] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 3788.953925] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3788.953940] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3788.957830] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 3788.957834] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 3788.957861] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 3788.957941]  dmic16k: ASoC: close BE dmic16k
[ 3788.958028]  DMIC16k: ASoC: close FE DMIC16k
[ 3788.958035] sof-audio-pci 0000:00:1f.3: pcm: close stream 100 dir 1
[ 3788.958153] sof-audio-pci 0000:00:1f.3: received event 4 for widget DETECT10.0
[ 3788.958158] sof-audio-pci 0000:00:1f.3: received event 8 for widget DETECT10.0
[ 3788.958168] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 3788.958410] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 3788.958418] sof-audio-pci 0000:00:1f.3: error: failed to trigger DETECT10.0
```
```
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[7508069903.020834] (             nan) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[7508069910.416667] (        7.395833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 536870912
[7508070003.177084] (       92.760414) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[7508070010.729167] (        7.552083) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[7508070018.020834] (        7.291667) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM overrun
[7508070026.250000] (        8.229167) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[7508070033.645834] (        7.395833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 536870912
[7508070127.968750] (       94.322914) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 2147483648
[7508070135.729167] (        7.760417) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 671088640
[7508070143.020834] (        7.291667) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1542 ERROR dmic_irq_handler(): full fifo B or PDM ov
```
[sof-trace.txt](https://github.com/thesofproject/sof/files/4930574/sof-trace.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/4930575/dmesg.txt)

slawblauciak, 2020-07-28:
To clarify with the Chrome team if this is to be a supported scenario.

Liviali155, 2020-07-29:
On Chrome JSL with codec in i2s mode also has this issue

[dmesg29.log](https://github.com/thesofproject/sof/files/4993602/dmesg29.log)
[sof-logger29.log](https://github.com/thesofproject/sof/files/4993603/sof-logger29.log)

https://github.com/thesofproject/linux/commits/topic/sof-dev(797eb03a)
https://github.com/thesofproject/sof/tree/releases/jsl/v4.0-rc1(21c8900f)

slawblauciak, 2020-08-10:
Lowered the priority as it's been confirmed that this is not a supported scenario in use with cras.

closed 2020-12-03 by mengdonglin