thesofproject/sof #3209: [BUG] No sound for Intel Cannon Lake PCH cAVS on Lenovo IdeaPad S740

hweom, 2020-07-20:
**Describe the bug**
No sound via speakers or headphones. Both speakers and headphones are detected correctly, I can control volume in `pavucontrol` and I can actually see how the sound bar is moving if a sound is played, but no sound comes out.

**To Reproduce**
No special steps to reproduce (the issue is always there right after boot).

**Reproduction Rate**
100%

**Expected behavior**
Audible sound.

**Impact**
Showstopper.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/torvalds/linux/commit/11ba468877bb23f28956a35e896356252d63c983
    * SOF: https://github.com/thesofproject/sof/commit/88707ca87e782631fd2da11a89e7a89d6bd95763

(I haven't found a way to check which topology file I'm using.)

**Screenshots or console output**
dmesg:
[dmesg.log](https://github.com/thesofproject/sof/files/4949147/dmesg.log)

```
$ inxi -Fxz
System:    Host: ideapad Kernel: 5.8.0-1-MANJARO x86_64 bits: 64 compiler: N/A Desktop: Gnome 3.36.4 Distro: Manjaro Linux 
Machine:   Type: Convertible System: LENOVO product: 81NW v: Lenovo IdeaPad S740-15IRH Touch serial: <filter> 
           Mobo: LENOVO model: LNVNB161216 v: SDK0R32862 WIN serial: <filter> UEFI: LENOVO v: BKCN21WW(V1.03) date: 12/18/2019 
Battery:   ID-1: BAT1 charge: 47.7 Wh condition: 70.1/69.1 Wh (101%) model: 0x53 0x4D 0x50 0x00 0x32 0x30 0x PABAS0241231 
           status: Discharging 
CPU:       Topology: 6-Core model: Intel Core i7-9750H bits: 64 type: MT MCP arch: Kaby Lake rev: A L2 cache: 12.0 MiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 62431 
           Speed: 800 MHz min/max: 800/4500 MHz Core speeds (MHz): 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800 8: 800 
           9: 800 10: 800 11: 800 12: 800 
Graphics:  Device-1: Intel UHD Graphics 630 vendor: Lenovo driver: i915 v: kernel bus ID: 00:02.0 
           Device-2: NVIDIA TU117M [GeForce GTX 1650 Mobile / Max-Q] vendor: Lenovo driver: nvidia v: 440.100 bus ID: 01:00.0 
           Display: x11 server: X.org 1.20.8 driver: intel,nvidia unloaded: modesetting,nouveau resolution: <xdpyinfo missing> 
           OpenGL: renderer: Mesa Intel UHD Graphics 630 (CFL GT2) v: 4.6 Mesa 20.1.3 direct render: Yes 
Audio:     Device-1: Intel Cannon Lake PCH cAVS vendor: Lenovo driver: sof-audio-pci bus ID: 00:1f.3 
           Sound Server: ALSA v: k5.8.0-1-MANJARO 
Network:   Device-1: Intel Wireless-AC 9560 [Jefferson Peak] driver: iwlwifi v: kernel port: 4000 bus ID: 00:14.3 
           IF: wlp0s20f3 state: up mac: <filter> 
Drives:    Local Storage: total: 715.44 GiB used: 199.33 GiB (27.9%) 
           ID-1: /dev/mmcblk0 model: GE8QT size: 238.50 GiB 
           ID-2: /dev/nvme0n1 vendor: Samsung model: MZVLB512HBJQ-000L2 size: 476.94 GiB 
Partition: ID-1: / size: 122.54 GiB used: 11.48 GiB (9.4%) fs: ext4 dev: /dev/nvme0n1p2 
           ID-2: swap-1 size: 15.62 GiB used: 0 KiB (0.0%) fs: swap dev: /dev/nvme0n1p1 
Sensors:   System Temperatures: cpu: 39.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Info:      Processes: 364 Uptime: 1h 28m Memory: 15.36 GiB used: 2.37 GiB (15.4%) Init: systemd Compilers: gcc: 10.1.0 
           Shell: bash v: 5.0.18 inxi: 3.0.37
```

I unfortunately was not able to use sof-logger, since it kept giving me:

```
$ sudo ./sof-logger -l /usr/lib/firmware/intel/sof/v1.5.1/sof-icl-v1.5.1.ldc -e
error: fw version in /usr/lib/firmware/intel/sof/v1.5.1/sof-icl-v1.5.1.ldc file does not coincide with fw version in /sys/kernel/debug/sof/fw_version file.
```

Enerccio, 2020-07-20:
I think it might be related to my issue #3208, I have c940

plbossart, 2020-07-20:
@hweom if you can compile 5.8-rc5, can you try with the SOF development kernel? https://github.com/thesofproject/linux.git branch topic/sof-dev

I don't see a smoking gun here.

hweom, 2020-07-20:
I'll try it, although I don't have experience building a custom kernel on Manjaro (5.8-rc5 is in Manjaro's testing branch repos), so it'll probably take some time.

hweom, 2020-07-21:
I've downgraded `alsa-lib`, `lib32-alsa-lib`, `alsa-topology-conf` and `alsa-ucm-conf` to 1.2.1.2 and now the headphones work, but the speakers still don't.

mengdonglin, 2020-07-21:
@hweom Can you pls provide output of alsa-info.sh when you're doing playback by the speaker although you cannot hear the sound?

lgirdwood, 2020-07-21:
> sudo ./sof-logger -l /usr/lib/firmware/intel/sof/v1.5.1/sof-icl-v1.5.1.ldc -e
> error: fw version in /usr/lib/firmware/intel/sof/v1.5.1/sof-icl-v1.5.1.ldc file does not coincide with fw version in /sys/kernel/debug/sof/fw_version file

@xiulipan - do we have the wrong LDC file for the ICL 1.5.1 binary ? maybe the sof-bin makefile should validate this as a rule.
@hweom als please check alsamixer - some speakers do have separate volume and mute controls to headphones. It's worth a quick poke around here and report any missing settings so they can be added to the UCM profile.

hweom, 2020-07-22:
```
$ sudo alsa-info.sh
ALSA Information Script v 0.4.65
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  aplay
  amixer
  alsactl
  rpm, dpkg
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See '/usr/bin/alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_skl_hda_dsp/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : n

Your ALSA information is in /tmp/alsa-info.txt.f6EFAV77OE
```

[alsa-info.txt.f6EFAV77OE](https://github.com/thesofproject/sof/files/4957092/alsa-info.txt.f6EFAV77OE.txt)

alsamixer does show a master channel that is not enabled by default:

[image]

However even I if max it out, there is still no sound. (Also there is nothing more on the right.)

xiulipan, 2020-07-22:
@lgirdwood Not sure yet. I did not do any test, I will try to do some check for the binary on 1.5.1 release.

xiulipan, 2020-07-22:
@hweom @lgirdwood I am a little confusing here, so what FW should be used on S740 if it is Cannon Lake PCH?

@hweom Could you try to use sof-cnl.ldc instead? and dump the log from `amixer contents` I am think about wrong codec setting.

lgirdwood, 2020-07-22:
@xiulipan agree it should be loading ```sof-cnl.ri``` binary IIRC and not ```sof-icl.ri```. @hweom renaming the sof FW directly should make the kernel complain about the FW file it's trying to load cant be found. This will confirm we are using the correct version based on your PCH ID.

plbossart, 2020-07-22:
@hweom please copy the attached file as /etc/modprobe.d/sof-ddebug.conf and re-attach the dmesg log.
[ddebug.txt](https://github.com/thesofproject/sof/files/4961305/ddebug.txt)

I don't get why we are talking about icl firmware, and the logs with dynamic debug will tell us exactly what the SOF driver does.
Thanks!

hweom, 2020-07-22:
@plbossart here it is. Unfortunately, the beginning of the log got truncated, so not sure if any interesting information was lost.

[dmesg_sof.txt](https://github.com/thesofproject/sof/files/4962260/dmesg_sof.txt)

plbossart, 2020-07-22:
@hweom yeah, all the initial stuff is lost...You seem to have a very chatty kernel configuration.
Maybe use a blacklist snd-sof-pci then after boot dmesg -C; dmesg -w > log.txt &; modprobe snd-sof-pci; 
that should give you the initial log?

hweom, 2020-07-22:
@plbossart done:

[dmesg_sof.txt](https://github.com/thesofproject/sof/files/4962420/dmesg_sof.txt)

hweom, 2020-07-22:
@xiulipan I think the last log confirms that we're indeed using sof-cfl.ri. I actually don't remember where I got the LDC file in my original file -- but sorry for the confusion!

Also, here is output from `amixer contents`:
```
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
```

xiulipan, 2020-07-27:
@hweom It seems nothing wrong with dmesg and logger logs. Can you try to play with the alsamixer setting to check if any codec setting may benefit for your case. @RanderWang have we do any test with the ALC285 codec?

plbossart, 2020-07-27:
Wondering if this is the same issue as https://github.com/thesofproject/linux/issues/1917, can you try the recipe that fixed it?
It seems there is a 'Master' simple control that's not set by UCM

````
$ amixer -Dhw:0 sset Master unmute
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 0 [0%] [-65.25dB] [on]

$ amixer -Dhw:0 sset Master 100%
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 87 [100%] [0.00dB] [on]
````

hweom, 2020-07-27:
Thanks for the suggestions!

I can confirm that doing 
```
$ amixer -Dhw:0 sset Master unmute
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 0 [0%] [-65.25dB] [on]

$ amixer -Dhw:0 sset Master 100%
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 87 [100%] [0.00dB] [on]
```

enabled the _headphones_ output. (I believe the second command is the same as setting the Master channel via `alsamixer` in my screenshot [here](https://github.com/thesofproject/sof/issues/3209#issuecomment-662173203). However, it is the _unmuting_ part that was missing in `alsamixer`.)

However, the speakers are still not working. 

Here is the output of `amixer -Dhw:0 scontents`:
```
$ amixer -Dhw:0 scontents
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 87 [100%] [0.00dB] [on]
Simple mixer control 'Headphone',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 87
  Mono:
  Front Left: Playback 87 [100%] [0.00dB] [off]
  Front Right: Playback 87 [100%] [0.00dB] [off]
Simple mixer control 'Speaker',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 87
  Mono:
  Front Left: Playback 87 [100%] [0.00dB] [on]
  Front Right: Playback 87 [100%] [0.00dB] [on]
Simple mixer control 'Mic Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 3
  Front Left: 0 [0%] [0.00dB]
  Front Right: 0 [0%] [0.00dB]
Simple mixer control 'IEC958',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',1
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'IEC958',2
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 63
  Front Left: Capture 0 [0%] [-17.25dB] [off]
  Front Right: Capture 0 [0%] [-17.25dB] [off]
Simple mixer control 'Auto-Mute Mode',0
  Capabilities: enum
  Items: 'Disabled' 'Enabled'
  Item0: 'Disabled'
Simple mixer control 'Dmic0',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 70
  Front Left: Capture 50 [71%] [0.00dB] [on]
  Front Right: Capture 50 [71%] [0.00dB] [on]
Simple mixer control 'Dmic1 2nd',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 70
  Front Left: Capture 50 [71%] [0.00dB]
  Front Right: Capture 50 [71%] [0.00dB]
Simple mixer control 'PGA1.0 1 Master',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 32
  Mono:
  Front Left: Playback 32 [100%] [0.00dB]
  Front Right: Playback 32 [100%] [0.00dB]
Simple mixer control 'PGA2.0 2 Master',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 80
  Front Left: Capture 50 [62%] [0.00dB]
  Front Right: Capture 50 [62%] [0.00dB]
Simple mixer control 'PGA3.0 3 Master',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 32
  Mono:
  Front Left: Playback 32 [100%] [0.00dB]
  Front Right: Playback 32 [100%] [0.00dB]
Simple mixer control 'PGA4.0 4 Master',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 80
  Front Left: Capture 50 [62%] [0.00dB]
  Front Right: Capture 50 [62%] [0.00dB]
Simple mixer control 'PGA7.0 7 Master',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 32
  Mono:
  Front Left: Playback 32 [100%] [0.00dB]
  Front Right: Playback 32 [100%] [0.00dB]
Simple mixer control 'PGA8.0 8 Master',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 32
  Mono:
  Front Left: Playback 32 [100%] [0.00dB]
  Front Right: Playback 32 [100%] [0.00dB]
Simple mixer control 'PGA9.0 9 Master',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 32
  Mono:
  Front Left: Playback 32 [100%] [0.00dB]
  Front Right: Playback 32 [100%] [0.00dB]
```

plbossart, 2020-07-27:
@hweom maybe try playing with what's disabled (AutoMuteMode) and what's on (SPDIF) to see if this changes anything.
I really don't get what UCM is doing here....

hweom, 2020-07-28:
```
$ amixer -Dhw:0 sset 'Auto-Mute Mode' Enabled
Simple mixer control 'Auto-Mute Mode',0
  Capabilities: enum
  Items: 'Disabled' 'Enabled'
  Item0: 'Enabled'
```
Yields no effect.

Changing SPDIFs to off didn't really change anything either (although I didn't try all the permutations):
```
[mikhail@ideapad ~]$ amixer -Dhw:0 sset 'IEC958',0 off
Simple mixer control 'IEC958',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
[mikhail@ideapad ~]$ amixer -Dhw:0 sset 'IEC958',1 off
Simple mixer control 'IEC958',1
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
[mikhail@ideapad ~]$ amixer -Dhw:0 sset 'IEC958',2 off
Simple mixer control 'IEC958',2
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
```

As an interesting datapoint, I just tried Ubuntu 20.04 live image and it had headphones working out of the box, but not speakers. I can't try Windows, since my laptop is not dual-booting and there is no way to boot windows live, at least not that I know of.

plbossart, 2020-07-28:
> As an interesting datapoint, I just tried Ubuntu 20.04 live image and it had headphones working out of the box, but not speakers. I can't try Windows, since my laptop is not dual-booting and there is no way to boot windows live, at least not that I know of.

Can you dump the mixer contents with the live image and compare with what you have here? It's clearly a configuration issue.

hweom, 2020-07-28:
Yes, here is the diff: http://www.mergely.com/pCTbN8wW/

Ubuntu 20.04 is on the right, my system is on the left.

Master is on by default in Ubuntu, but off by default for me. This explains why headphones work in Ubuntu out of the box.

Again, speakers don't work in both systems.

hweom, 2020-07-28:
I just checked it on a live WinPE image, and speakers work in Windows. So at least we know for sure this is not a broken device.

plbossart, 2020-07-28:
@hweom when you say 'speakers don't work w/ Linux', is this some of the speakers don't work or none of the speakers work?
There are known issues with a series of Lenovo C930/C940 devices, wondering if they are related to your config:

https://bugzilla.kernel.org/show_bug.cgi?id=203409
https://bugzilla.kernel.org/show_bug.cgi?id=205755

hweom, 2020-07-28:
Thanks @plbossart , I'll look at those bugs for potential insights.

To clarify, in my case, **none** of the internal laptop speakers work.

hweom, 2020-07-28:
Looks like this kernel bug is exact match for my situation:

https://bugzilla.kernel.org/show_bug.cgi?id=207019

RanderWang, 2020-07-31:
> @hweom It seems nothing wrong with dmesg and logger logs. Can you try to play with the alsamixer setting to check if any codec setting may benefit for your case. @RanderWang have we do any test with the ALC285 codec?

ALC285 ? I have no idea about it.

RanderWang, 2020-07-31:
I found a fix related to this
````C
From d5d720ae68d713f78afc4f8bf0e0897bb2ccf33c Mon Sep 17 00:00:00 2001
From: Kailang Yang <kailang@realtek.com>
Date: Wed, 29 Jul 2020 15:09:27 +0800
Subject: [PATCH] ALSA: hda/realtek - Fixed HP right speaker no sound

HP NB right speaker had no sound output.
This platform was connected to I2S Amp for speaker out.(None Realtek I2S Amp IC)
EC need to check codec GPIO1 pin to initial I2S Amp.

diff --git a/sound/pci/hda/patch_realtek.c b/sound/pci/hda/patch_realtek.c
index e95fea8430b0..8afd7e07129c 100644
--- a/sound/pci/hda/patch_realtek.c
+++ b/sound/pci/hda/patch_realtek.c
@@ -5963,6 +5963,16 @@ static void alc_fixup_disable_mic_vref(struct hda_codec *codec,
 		snd_hda_codec_set_pin_target(codec, 0x19, PIN_VREFHIZ);
 }
 
+static void  alc285_fixup_hp_gpio_amp_init(struct hda_codec *codec,
+			      const struct hda_fixup *fix, int action)
+{
+	if (action != HDA_FIXUP_ACT_INIT)
+		return;
+
+	msleep(100);
+	alc_write_coef_idx(codec, 0x65, 0x0);
+}
+
 /* for hda_fixup_thinkpad_acpi() */
 #include "thinkpad_helper.c"
 
@@ -6143,6 +6153,7 @@ enum {
 	ALC289_FIXUP_ASUS_GA401,
 	ALC289_FIXUP_ASUS_GA502,
 	ALC256_FIXUP_ACER_MIC_NO_PRESENCE,
+	ALC285_FIXUP_HP_GPIO_AMP_INIT,
 };
 
 static const struct hda_fixup alc269_fixups[] = {
@@ -7375,6 +7386,12 @@ static const struct hda_fixup alc269_fixups[] = {
 		.chained = true,
 		.chain_id = ALC256_FIXUP_ASUS_HEADSET_MODE
 	},
+	[ALC285_FIXUP_HP_GPIO_AMP_INIT] = {
+		.type = HDA_FIXUP_FUNC,
+		.v.func = alc285_fixup_hp_gpio_amp_init,
+		.chained = true,
+		.chain_id = ALC285_FIXUP_HP_GPIO_LED
+	},
 };
 
 static const struct snd_pci_quirk alc269_fixup_tbl[] = {
@@ -7525,7 +7542,7 @@ static const struct snd_pci_quirk alc269_fixup_tbl[] = {
 	SND_PCI_QUIRK(0x103c, 0x84e7, "HP Pavilion 15", ALC269_FIXUP_HP_MUTE_LED_MIC3),
 	SND_PCI_QUIRK(0x103c, 0x869d, "HP", ALC236_FIXUP_HP_MUTE_LED),
 	SND_PCI_QUIRK(0x103c, 0x8729, "HP", ALC285_FIXUP_HP_GPIO_LED),
-	SND_PCI_QUIRK(0x103c, 0x8736, "HP", ALC285_FIXUP_HP_GPIO_LED),
+	SND_PCI_QUIRK(0x103c, 0x8736, "HP", ALC285_FIXUP_HP_GPIO_AMP_INIT),
 	SND_PCI_QUIRK(0x103c, 0x877a, "HP", ALC285_FIXUP_HP_MUTE_LED),
 	SND_PCI_QUIRK(0x103c, 0x877d, "HP", ALC236_FIXUP_HP_MUTE_LED),
 	SND_PCI_QUIRK(0x1043, 0x103e, "ASUS X540SA", ALC256_FIXUP_ASUS_MIC),
````

hweom, 2020-08-08:
I've updated to 5.8.0-1 and just saw this in `dmesg`:

```
[    4.628502] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.677898] ------------[ cut here ]------------
[    4.677904] WARNING: CPU: 0 PID: 433 at kernel/kmod.c:137 __request_module+0x29e/0x41b
[    4.677905] Modules linked in: acpi_cpufreq(-) snd_soc_dmic snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda x86_pkg_temp_thermal snd_sof_xtensa_dsp intel_powerclamp snd_sof_intel_hda coretemp joydev mousedev snd_sof ledtrig_audio kvm_intel snd_soc_skl hid_multitouch snd_soc_sst_ipc iTCO_wdt thunderbolt hid_generic snd_soc_sst_dsp intel_pmc_bxt ee1004 mei_hdcp kvm iTCO_vendor_support snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core intel_rapl_msr snd_compress ac97_bus snd_pcm_dmaengine wmi_bmof intel_wmi_thunderbolt irqbypass snd_hda_intel mmc_block snd_intel_dspcfg iwlmvm snd_hda_codec nls_iso8859_1 nls_cp437 crct10dif_pclmul snd_hda_core vfat crc32_pclmul ghash_clmulni_intel snd_hwdep fat mac80211 i915 snd_pcm aesni_intel libarc4 crypto_simd cryptd glue_helper rapl intel_cstate ofpart cmdlinepart snd_timer intel_spi_pci iwlwifi intel_uncore intel_spi snd spi_nor ucsi_acpi mei_me i2c_i801 pcspkr intel_lpss_pci typec_ucsi
[    4.677937]  input_leds mtd i2c_smbus soundcore processor_thermal_device i2c_hid i2c_algo_bit intel_lpss cfg80211 mei idma64 intel_rapl_common ideapad_laptop intel_gtt intel_pch_thermal hid intel_soc_dts_iosf typec rfkill tpm_crb int3403_thermal int340x_thermal_zone battery ac tpm_tis wmi tpm_tis_core tpm rng_core int3400_thermal acpi_thermal_rel intel_hid evdev mac_hid sparse_keymap uinput nvidia_drm(POE) nvidia_modeset(POE) drm_kms_helper cec rc_core drm agpgart syscopyarea sysfillrect sysimgblt fb_sys_fops nvidia(POE) ipmi_devintf ipmi_msghandler crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 sdhci_pci cqhci sdhci xhci_pci crc32c_intel xhci_pci_renesas mmc_core xhci_hcd i8042 serio
[    4.677968] CPU: 0 PID: 433 Comm: kworker/u24:4 Tainted: P           OE     5.8.0-1-MANJARO #1
[    4.677970] Hardware name: LENOVO 81NW/LNVNB161216, BIOS BKCN21WW(V1.03) 12/18/2019
[    4.677973] Workqueue: events_unbound async_run_entry_fn
[    4.677976] RIP: 0010:__request_module+0x29e/0x41b
[    4.677978] Code: 00 48 85 c0 75 e1 65 ff 0d 3f 78 f5 6e 0f 85 b1 fe ff ff e8 43 47 f4 ff e9 a7 fe ff ff e8 4a eb ff ff 84 c0 0f 84 1a fe ff ff <0f> 0b e9 13 fe ff ff 48 c7 c6 50 11 01 92 48 c7 c7 80 db 64 92 e8
[    4.677979] RSP: 0018:ffffae2bc0b3f978 EFLAGS: 00010202
[    4.677981] RAX: ffff9cf35bb72801 RBX: 0000000000000001 RCX: 0000000000000000
[    4.677982] RDX: ffffffffc172ee96 RSI: ffffffffc1714ec7 RDI: ffff9cf34d5f5d00
[    4.677983] RBP: ffffae2bc0b3fa58 R08: ffff9cf35284f75b R09: 0000000000000000
[    4.677984] R10: ffffae2bc0b3fa68 R11: ffff9cf35284f75b R12: ffffae2bc0b3f9b8
[    4.677986] R13: ffffffffc1714ec7 R14: ffffae2bc0b3fa68 R15: ffff9cf35dce3658
[    4.677987] FS:  0000000000000000(0000) GS:ffff9cf360400000(0000) knlGS:0000000000000000
[    4.677989] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    4.677990] CR2: 00007f3458039020 CR3: 00000003aea0a001 CR4: 00000000003606f0
[    4.677991] Call Trace:
[    4.677999]  ? kobject_set_name_vargs+0x6f/0x90
[    4.678006]  rc_map_get+0x7a/0xe0 [rc_core]
[    4.678011]  rc_register_device+0x120/0x540 [rc_core]
[    4.678016]  cec_register_adapter+0x5c/0x280 [cec]
[    4.678029]  drm_dp_cec_set_edid+0x163/0x1d9 [drm_kms_helper]
[    4.678107]  intel_dp_set_edid+0x8d/0xc0 [i915]
[    4.678175]  intel_dp_detect+0x19d/0x5f0 [i915]
[    4.678188]  drm_helper_probe_single_connector_modes+0xe5/0x6f0 [drm_kms_helper]
[    4.678213]  drm_client_modeset_probe+0x29d/0x13e0 [drm]
[    4.678220]  ? sched_clock+0x5/0x10
[    4.678230]  __drm_fb_helper_initial_config_and_unlock+0x45/0x4b0 [drm_kms_helper]
[    4.678235]  ? _raw_spin_unlock_irq+0x1d/0x30
[    4.678238]  ? finish_task_switch+0x80/0x270
[    4.678240]  ? __switch_to_asm+0x36/0x70
[    4.678302]  intel_fbdev_initial_config+0x14/0x30 [i915]
[    4.678305]  async_run_entry_fn+0x37/0x140
[    4.678308]  process_one_work+0x1da/0x3d0
[    4.678311]  worker_thread+0x4d/0x3d0
[    4.678313]  ? rescuer_thread+0x410/0x410
[    4.678316]  kthread+0x142/0x160
[    4.678318]  ? __kthread_bind_mask+0x60/0x60
[    4.678321]  ret_from_fork+0x22/0x30
[    4.678325] ---[ end trace 1b47830c33d524cd ]---
[    4.685321] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.709246] Registered IR keymap rc-cec
[    4.709284] rc rc0: DP-1 as /devices/pci0000:00/0000:00:02.0/rc/rc0
[    4.709306] input: DP-1 as /devices/pci0000:00/0000:00:02.0/rc/rc0/input16
[    4.764672] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.764673] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.764676] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.775648] fbcon: i915drmfb (fb0) is primary device
[    4.775649] fbcon: Deferring console take-over
[    4.775651] i915 0000:00:02.0: fb0: i915drmfb frame buffer device
```

Not sure if it's relevant. (And audio situation is still the same with new kernel.)

hweom, 2020-08-08:
> I found a fix related to this

I've tried applying this fix to my system (with relevant changes) and confirmed that it's applied, but the speaker sound is still muted:

```
--- a/src/linux-5.8/sound/pci/hda/patch_realtek.c
+++ b/src/linux-5.8/sound/pci/hda/patch_realtek.c
@@ -5981,6 +5981,8 @@ static void  alc285_fixup_hp_gpio_amp_init(struct hda_codec *codec,
        if (action != HDA_FIXUP_ACT_INIT)
                return;
 
+       printk("Fixup entered");
+
        msleep(100);
        alc_write_coef_idx(codec, 0x65, 0x0);
 }
@@ -7669,6 +7671,10 @@ static const struct snd_pci_quirk alc269_fixup_tbl[] = {
        SND_PCI_QUIRK(0x17aa, 0x3151, "ThinkCentre Station", ALC283_FIXUP_HEADSET_MIC),
        SND_PCI_QUIRK(0x17aa, 0x3176, "ThinkCentre Station", ALC283_FIXUP_HEADSET_MIC),
        SND_PCI_QUIRK(0x17aa, 0x3178, "ThinkCentre Station", ALC283_FIXUP_HEADSET_MIC),
+       // !!! 
+       SND_PCI_QUIRK(0x17aa, 0x3827, "Lenovo", ALC285_FIXUP_HP_GPIO_AMP_INIT),
+       SND_PCI_QUIRK(0x8086, 0xa348, "Lenovo", ALC285_FIXUP_HP_GPIO_AMP_INIT),
+       // !!!
        SND_PCI_QUIRK(0x17aa, 0x3902, "Lenovo E50-80", ALC269_FIXUP_DMIC_THINKPAD_ACPI),
        SND_PCI_QUIRK(0x17aa, 0x3977, "IdeaPad S210", ALC283_FIXUP_INT_MIC),
        SND_PCI_QUIRK(0x17aa, 0x3978, "Lenovo B50-70", ALC269_FIXUP_DMIC_THINKPAD_ACPI),
```

(I wasn't sure which PCI device IDs are used from this:
```
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Cannon Lake PCH cAVS [8086:a348] (rev 10)
	Subsystem: Lenovo Device [17aa:3827]
```
so I added both.)

`dmesg` has:
```
[    5.017550] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    5.041100] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    5.041102] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.041102] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    5.041103] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    5.041103] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    5.041104] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    5.087015] Fixup entered
```
So I know the code works.

The change seems to have *some* effect though: I now have a new scontents entry:
```
Simple mixer control 'Mic Mute-LED Mode',0
  Capabilities: enum
  Items: 'On' 'Off' 'Follow Capture' 'Follow Mute'
  Item0: 'Follow Mute'
```

I've tried setting it to all possible values without any noticeable difference.

Any clues on how to debug this further? I can do some experiments, but I have no idea how the sound works in Linux/SOF. Tried to read up some docs, but quickly got overwhelmed :(

lgirdwood, 2020-08-10:
@hweom one more thing to check - can you disable pulseaudio (and make sure it's not running - it can respawn) and check each PCM device directly with aplay for speakers (whilst trying alsamixer). This will rule out any mixer changes via pulseaudio and allow you to target one PCM device at a time.

hweom, 2020-08-11:
Hmm, I've tried to google up some examples of `aplay`/`speaker-test` and I see people have some actual speaker devices in their `aplay -L`, but I have only this:

```
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pipewire
    PipeWire Sound Server
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Default ALSA Output (currently PulseAudio Sound Server)
sysdefault:CARD=sofhdadsp
    sof-hda-dsp, 
    Default Audio Device
usbstream:CARD=sofhdadsp
    sof-hda-dsp
    USB Stream Output
```

And when I try `speaker-test`, no sound comes out:

```
$ speaker-test -Dplug:sysdefault -c 6

speaker-test 1.2.3

Playback device is plug:sysdefault
Stream parameters are 48000Hz, S16_LE, 6 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 4 - Center
 1 - Front Right
 3 - Rear Right
 2 - Rear Left
 5 - LFE
Time per period = 17.523502
 0 - Front Left
 4 - Center
 1 - Front Right
 3 - Rear Right
...
```

(Pulse audio is not running, verified with `ps aux | grep pulse` before and after running `speaker-test`.)

`alsamixer` shows all controls maxed out.

RanderWang, 2020-08-11:
@hweom Do you try legacy hda driver instead of SOF ?  Can you make sure speaker works on this laptop ?  There is no difference between headphone and speaker in SOF driver side. If the speaker works, it is more likely linked to codec drver in patch_realtek.c

lgirdwood, 2020-08-11:
@hweom you can aplay to any of the PCM devices using the ```-DhwX.Y``` cmd line option where X is card and Y is PCM as listed from /proc
```
cat /proc/asound/pcm 
00-00: ALC1220 Analog : ALC1220 Analog : playback 1 : capture 1
00-01: ALC1220 Digital : ALC1220 Digital : playback 1
00-02: ALC1220 Alt Analog : ALC1220 Alt Analog : capture 1
01-03: HDMI 0 : HDMI 0 : playback 1
01-07: HDMI 0 : HDMI 0 : playback 1
01-08: HDMI 0 : HDMI 0 : playback 1
01-09: HDMI 0 : HDMI 0 : playback 1
```

hweom, 2020-08-12:
Got it, thanks. Here's what I have (I assume it should be `-DhwX,Y`, otherwise I get "No such device" error):

```
$ cat /proc/asound/pcm 
00-00: HDA Analog (*) :  : playback 1 : capture 1
00-01: HDA Digital (*) :  : playback 1 : capture 1
00-03: HDMI1 (*) :  : playback 1
00-04: HDMI2 (*) :  : playback 1
00-05: HDMI3 (*) :  : playback 1
00-06: DMIC (*) :  : capture 1
00-07: DMIC16kHz (*) :  : capture 1

$ aplay -Dhw:0,0 -vvv /usr/share/sounds/alsa/Noise.wav 
Playing WAVE '/usr/share/sounds/alsa/Noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available

$ aplay -Dhw:0,1 -vvv /usr/share/sounds/alsa/Noise.wav 
aplay: main:830: audio open error: Invalid argument

$ aplay -Dhw:0,3 -vvv /usr/share/sounds/alsa/Noise.wav 
Playing WAVE '/usr/share/sounds/alsa/Noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available

$ aplay -Dhw:0,4 -vvv /usr/share/sounds/alsa/Noise.wav 
Playing WAVE '/usr/share/sounds/alsa/Noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available

$ aplay -Dhw:0,5 -vvv /usr/share/sounds/alsa/Noise.wav 
Playing WAVE '/usr/share/sounds/alsa/Noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available

$ aplay -Dhw:0,6 -vvv /usr/share/sounds/alsa/Noise.wav 
aplay: main:830: audio open error: No such file or directory

$ aplay -Dhw:0,7 -vvv /usr/share/sounds/alsa/Noise.wav 
aplay: main:830: audio open error: No such file or directory
```

plbossart, 2020-08-12:
@hweom we don't support mono, so either use a 2ch wave file or use -Dplughw: instead of -Dhw: (it will convert for you).

hweom, 2020-08-12:
Ah, thanks. Tried that, the error is gone but still no sound.

RanderWang, 2020-08-12:
The last method: install windows to check whether speaker is broken or not.

lgirdwood, 2020-08-12:
@RanderWang see previous comments - speakers work in windows.
@hweom IIUC the speakers should be hooked up to either PCM 0,0 or PCM0,1 below
```
 $ cat /proc/asound/pcm 
 00-00: HDA Analog (*) :  : playback 1 : capture 1
 00-01: HDA Digital (*) :  : playback 1 : capture 1
```
Please try the following
```
aplay -vvv -Dplughw:0,0 -f dat -d 30 /dev/urandom
aplay -vvv -Dplughw:0,1 -f dat -d 30 /dev/urandom
```

This will play 30 seconds of **loud** noise on both the above PCM devices. If there is no audio after traying all alsamixer settings then it could be an external AMP that is not known to Linux (or something similar betwen the HDA codec and the speakers)..

hweom, 2020-08-12:
> Do you try legacy hda driver instead of SOF 

Yes, I've tried blacklisting the SOF driver, but legacy intel driver is not able to recognize my card (I get only Dummy Output).

> Can you make sure speaker works on this laptop ?

Yes, as commented above, speakers work in Windows.

> Please try the following

```
$ aplay -vvv -Dplughw:0,0 -f dat -d 30 /dev/urandom
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Hardware PCM card 0 'sof-hda-dsp' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
Max peak (8160 samples): 0x00007ffa #################### 99%
Max peak (8160 samples): 0x00007ffb #################### 99%
Max peak (8160 samples): 0x00007ffe #################### 99%
Max peak (8160 samples): 0x00007ffd #################### 99%
Max peak (8160 samples): 0x00007fff #################### 99%
...

$ aplay -vvv -Dplughw:0,1 -f dat -d 30 /dev/urandom
aplay: main:830: audio open error: Invalid argument
```
No sound via speakers. Notice that playing through hw:0,1 fails -- is this expected?

> If there is no audio after traying all alsamixer settings then it could be an external AMP that is not known to Linux

What's the best way to debug this? It looks like this problem is widespread -- I've seen many forum threads for the same laptop model without a solution:
https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/1857198
https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/1857552
https://forums.lenovo.com/t5/Other-Linux-Discussions/Lenovo-ideapad-S740-speakers-in-linux-not-working/m-p/4581648
https://forums.linuxmint.com/viewtopic.php?t=325637

plbossart, 2020-08-12:
> > Do you try legacy hda driver instead of SOF
> 
> Yes, I've tried blacklisting the SOF driver, but legacy intel driver is not able to recognize my card (I get only Dummy Output).

make sure you also don't have a blacklist for snd-hda-intel, and you may also need to use
"options snd_intel_dspcfg dsp_driver=1"

hweom, 2020-08-12:
Right, I thought about the same thing. With `options snd_intel_dspcfg dsp_driver=1` intel driver is loaded:

```
$ sudo dmesg | grep "snd\|hda\|sound\|audio"
[    3.384575] snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.384754] snd_hda_intel 0000:00:1f.3: Applying patch firmware 'hda-jack-retask.fw'
[    4.589126] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.640387] Modules linked in: acpi_cpufreq(-) joydev mousedev x86_pkg_temp_thermal intel_powerclamp coretemp iTCO_wdt intel_pmc_bxt hid_multitouch hid_generic ee1004 iTCO_vendor_support snd_soc_skl kvm_intel snd_soc_sst_ipc snd_soc_sst_dsp mei_hdcp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi intel_rapl_msr wmi_bmof intel_wmi_thunderbolt snd_soc_core kvm snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg irqbypass iwlmvm nls_iso8859_1 snd_hda_codec nls_cp437 crct10dif_pclmul crc32_pclmul vfat mac80211 ghash_clmulni_intel fat snd_hda_core aesni_intel libarc4 snd_hwdep snd_pcm crypto_simd cryptd glue_helper rapl intel_cstate snd_timer ofpart iwlwifi i915 cmdlinepart snd intel_uncore intel_spi_pci intel_spi spi_nor i2c_i801 pcspkr input_leds mtd i2c_smbus soundcore cfg80211 thunderbolt tpm_crb mei_me processor_thermal_device mei intel_lpss_pci ucsi_acpi intel_rapl_common i2c_algo_bit intel_lpss typec_ucsi ideapad_laptop intel_gtt idma64 intel_soc_dts_iosf
[    4.782398] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=1 (0x1e/0x0/0x0/0x0/0x0) type:speaker
[    4.782402] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.782404] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.782406] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[    4.782408] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[    4.782410] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
[    4.856631] input: HDA Digital PCBeep as /devices/pci0000:00/0000:00:1f.3/sound/card0/input17
[    4.856690] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input18
[    4.856738] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input19
[    4.856787] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input20
[    4.856818] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input21
[    4.856857] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input22
[    4.856886] input: HDA Intel PCH HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input23
[    4.856938] input: HDA Intel PCH HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input24
```

However, the intel driver behaves exactly like the SOF -- I have sound via headphones but not through speakers. Tried `aplay -vvv -Dplughw:0,0 -f dat -d 30 /dev/urandom` too.

plbossart, 2020-08-12:
@hweom it could be simple (bad mixer settings) or really complicated (bad driver settings). 
For the first case, try and look with amixer -Dhw:0 controls or alsamixer, if for some reason you see a muted output.
For the second case you'd need to work with Realtek/Lenovo to understand the codec configuration.

hweom, 2020-08-12:
I've tried toggling every single control in amixer, without luck.

Do you know if there is a usual path/channel to talk to Lenovo about this kind of stuff? I've opened a support ticket, but it was immediately closed with a comment that Lenovo doesn't support Linux on this model. The required fixes *do* get implemented  eventually, right? How this is usually done?

plbossart, 2020-08-12:
@hweom i've alerted my contacts to this issue, but it's not my place to comment on 3rd party support and which models are supported.

hweom, 2020-08-12:
Thanks Pierre-Louis!

Yes, the usual workflow for getting Linux support for new HW is a complete black box to me, so I apologize if I ask too many questions or create too much commotion :) 

If there is anything I can do to help, please let me know. I do C/C++ programming for life, but zero experience/knowledge with Linux kernel or sound.

mrhpearson, 2020-08-12:
Hi,

Consider Lenovo notified :) (thanks Pierre-Louis for bringing it to my attention). I'll see what I can track down. 

Just to set expectations - the S470 is not a Linux supported platform. That means I have quite limited levers I can pull on to get support. I've asked if I can get details on the audio topology and any information that might help Linux audio experts figure out what is missing.  I will report back with the results. I don't have access to a S470 to play with myself.

If anybody wants to ask about this in the future (give me a nudge etc if I go silent for too long) it is being tracked internally with "LO-478". Note - that's an internal reference for the PC Linux team; don't go to Lenovo support with it as they will just stare at you blankly ;)

Mark

hweom, 2020-08-13:
Thanks Mark and Pierre-Louis, I really appreciate your help!

(I think you just mistyped, but just in case - it's S740, not S470.)

RanderWang, 2020-08-13:
SOF driver & legacy HDA driver share the same codec driver patch_realtech.c.  If both can't work, you may need to ask realtech for help.

AtjonTV, 2020-08-17:
I am having the same issues on a HP Omen 17 (2019 Edition with nVidia 2080 [TU104]) running Gentoo Linux on Kernel 5.8.1.

It also has the Intel Cannon Lake PCH cAVS audio device. The only thing I managed to get is the nocodec dummy driver to load.

Without the dummy installed the right driver cant be found:

```
# dmesg | grep -i audio
[    0.815286] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    3.148440] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    3.509609] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.509611] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.509624] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.509800] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.511975] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.516484] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    3.516485] sof-audio-pci 0000:00:1f.3: error: no matching ASoC machine driver found - aborting probe
[    3.516486] sof-audio-pci 0000:00:1f.3: error: failed to get machine info -19
[    3.516664] sof-audio-pci 0000:00:1f.3: error: failed to probe DSP hardware!
```

I have nearly every SOF flag in the Kernel enabled.

The speakers work (tested on Windows 10 1809) yet not with Gentoo or Arch.

AtjonTV, 2020-08-17:
I have got it working by setting `options snd-intel-dspcfg dsp_driver=1` in a file in `/etc/modprobe.d/` and then reloading the `snd_hda_intel` kernel module.

It only shows stero output and the mics wont show/work, but I dont care about those anyway ..

plbossart, 2020-08-17:
> I have nearly every SOF flag in the Kernel enabled.

You're probably missing CONFIG_SND_SOC_INTEL_SKL_HDA_DSP_GENERIC_MACH

AtjonTV, 2020-08-17:
Nope, its build as a Module

plbossart, 2020-08-17:
can you add this file 
[ddebug.txt](https://github.com/thesofproject/sof/files/5086909/ddebug.txt)
as /etc/modprobe.d/sof-ddbug.conf and retry (remove the options snd-intel-dspcfg dsp_driver=1), then attach the dmesg log.
I am really curious as to why there's no machine here, it's been a long time since we saw this.

AtjonTV, 2020-08-18:
There is not much:

```
[ 5817.169199] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x1f0500
[ 5817.169202] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x1f0500
[ 5817.169469] snd_hda_intel 0000:01:00.1: spurious response 0x3:0x0, last cmd=0x473500
[ 5817.169470] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x473500
[ 5817.169708] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x4f0015
[ 5817.169709] snd_hda_intel 0000:01:00.1: spurious response 0x3:0x0, last cmd=0x4f0015
[ 5817.169969] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x470700
[ 5817.169970] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x470700
[ 5817.170219] snd_hda_intel 0000:01:00.1: spurious response 0x3:0x0, last cmd=0x473502
[ 5817.170220] snd_hda_intel 0000:01:00.1: spurious response 0x0:0x0, last cmd=0x473502
[ 5818.268142] udevd[899]: invalid key/value pair in file /lib/udev/rules.d/60-steam-input.rules on line 42, starting at character 82 ('u')
[ 5850.883174] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[ 5850.883177] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[ 5850.883300] snd_hda_intel 0000:01:00.1: Disabling MSI
[ 5850.883303] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[ 5850.962093] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input44
[ 5850.962182] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input45
[ 5850.962245] input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input46
[ 5850.962299] input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input47
[ 5850.962376] input: HDA NVidia HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input48
[ 5850.962417] input: HDA NVidia HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0/input49
```

I have added that config, removed that other option, killed pulseaudio, unloaded and loaded the `snd_hda_intel` module and started pulseaudio again.

There is only the HMDI output device listed by pulse.

AtjonTV, 2020-08-18:
I tried to revert what I just did to get audio back and noticed that the deprecated `options snd_hda_intel dmic_detect=0` was the cause of the audio working, not the one option I said earlier.

H-Park, 2020-09-09:
Is there an update on this? I've had a IdeaPad S740 since mid March and I've not been able to get my speakers on microphone to work.

lgirdwood, 2020-09-09:
@H-Park sorry it's not working for you, please do let us know if 5.9 fixes this for you. There is also the topic/sof-dev branch on Linux git here (that has patches not in 5.9 and destined for 5.10+) that can also be tried if 5.9 does not work. Lets also keep @mrhpearson in the loop too.

H-Park, 2020-09-09:
Update: AUR/linux-git did not fix the sound issues out of the box, unfortunately. I have not played around with modprobe.d conf files, and I have ```snd_hda_intel dmic_detect=0``` as a kernel parameter. Lots of knobs to tweak regarding this. I'll play around with it on my flight this afternoon and see what I can get done. As it stands now, my laptop does not see the Intel Cannon Lake PCH cAVS as an audio device (gnome settings say "Speakers - Built-in Audio:). At one point it did, but I've messed with so many settings over the months. I suppose it's unfair of me to say AUR/linux-git didn't fix my sound issue when I have misconfigured settings preventing the sound card from being detected. 

@lgirdwood Great! I'll watch that branch and hopefully all this can get figured out and the appropriate patches start getting  pushed upstream and shipped.

plbossart, 2020-09-09:
@H-Park  snd_hda_intel dmic_detect=0  seems like an obsolete setting to me.

First remove all settings, and add options snd-intel-dspcfg dsp_driver=1
reboot and see if your speakers work.

Then if you want to use the SOF driver, remove all settings, add this file 
[ddebug.txt](https://github.com/thesofproject/sof/files/5197521/ddebug.txt) as /etc/modprobe.d/sof-ddebug.conf and attach the dmesg log.

[25 more replies]