thesofproject/sof #3221: CML : DMIC capture has initial 350 ms muted

sathyap-chrome, 2020-07-22:
On CML devices, we see 350 ms delayed start of DMIC capture.
Customer is interested to understand below things

(a) why is there is 350 ms delay ?
(b) is it specific to CML or can this be modified if required ?

I'm attaching the graph for reference ( to confirm there is no additional processing added in topology)
captured wav files from 2 customer platforms is also attached.
[DMIC_mute_issue.zip](https://github.com/thesofproject/linux/files/4924848/DMIC_mute_issue.zip)

The delay can be seen in both of them
[image]

slawblauciak, 2020-08-10:
@singalsu is the ramp up time configurable?

singalsu, 2020-08-10:
The ramp time is configurable in topology. It's logarithmic shape so it may look muted but it's not. Please plot logaritmic level vs. time so it becomes visible.

The ramp time has been optimized in known platforms to conceal best the start transient. Without it some rattle that is louder than speech is heard in the beginning.

In e.g. file tools/topology/platform/intel/intel-generic-dmic.m4 it's configured to 200 ms for 48 kHz DAI.

```
DAI_CONFIG(DMIC, 0, DMIC_DAI_LINK_48k_ID, dmic01,
	   DMIC_CONFIG(1, 500000, 4800000, 40, 60, 48000,
		DMIC_WORD_LENGTH(s32le), 200, DMIC, 0,
		PDM_CONFIG(DMIC, 0, FOUR_CH_PDM0_PDM1)))
```

For the 16 kHz DAI it's configured to 400 ms.

```
DAI_CONFIG(DMIC, 1, DMIC_DAI_LINK_16k_ID, dmic16k,
	   DMIC_CONFIG(1, 500000, 4800000, 40, 60, 16000,
		DMIC_WORD_LENGTH(s32le), 400, DMIC, 1,
		PDM_CONFIG(DMIC, 1, FOUR_CH_PDM0_PDM1)))
```

slawblauciak, 2020-08-10:
@sathya-nujella is this answer sufficient for you?

sathya-nujella, 2020-08-10:
> The ramp time is configurable in topology. It's logarithmic shape so it may look muted but it's not. Please plot logaritmic level vs. time so it becomes visible.
> 
> The ramp time has been optimized in known platforms to conceal best the start transient. Without it some rattle that is louder than speech is heard in the beginning.
> 
> In e.g. file tools/topology/platform/intel/intel-generic-dmic.m4 it's configured to 200 ms for 48 kHz DAI.
> 
> ```
> DAI_CONFIG(DMIC, 0, DMIC_DAI_LINK_48k_ID, dmic01,
> 	   DMIC_CONFIG(1, 500000, 4800000, 40, 60, 48000,
> 		DMIC_WORD_LENGTH(s32le), 200, DMIC, 0,
> 		PDM_CONFIG(DMIC, 0, FOUR_CH_PDM0_PDM1)))
> ```
> 
> For the 16 kHz DAI it's configured to 400 ms. It's because the DC compensation filter in HW uses the same coefficients for any rate so the internal DC level settling is shorter.
> 
> ```
> DAI_CONFIG(DMIC, 1, DMIC_DAI_LINK_16k_ID, dmic16k,
> 	   DMIC_CONFIG(1, 500000, 4800000, 40, 60, 16000,
> 		DMIC_WORD_LENGTH(s32le), 400, DMIC, 1,
> 		PDM_CONFIG(DMIC, 1, FOUR_CH_PDM0_PDM1)))
> ```
> 
> 400 ms is quite long but unfortunately that's pretty much minimum to conceal the start noise sounds. If the audio capture is not for humans but instead for machine hearing and any level control algorithms tolerate the DC pulse the ramp could be omitted. Also adding IIR EQ or dcblock component to suppress more the DC pulse could be experimented for faster time to normal capture. The DCCOMP filter in HW is only first order and it's tuned for very low cutoff. If it's known that lower frequencies are not needed, e.g. less than 100 Hz then the settling could be made a lot faster with those components.

Good to know Seppo that it is configurable.

sathya-nujella, 2020-08-10:
> @sathya-nujella is this answer sufficient for you?

@sathyap-chrome was discussing with Customer.. so requesting him for update.

singalsu, 2020-08-26:
Any updates to this?

sathyap-chrome, 2020-09-08:
sorry was on leave. yes. we can close this

closed 2020-09-08 by sathyap-chrome