thesofproject/sof #3222: [BUG][Fuzzing]Testbench: Segmentation Fault: when running testbench with a topology file

ymdatta, 2020-07-22:
**Describe the bug**
This bug was found when fuzzing the testbench using AFL fuzzer. For a fuzzed topology file input (i.e this topology was generated by the fuzzer) testbench exits with segmentation fault(core dumped) error.

**To Reproduce**
1. Build the testbench from the using `host-build-all.sh` script from scripts directory.
2. Download the topology file from [here](https://github.com/ymdatta/sof_testbench_fuzz_files/blob/master/id_02_tplg.bin).
3. Run the `host-testbench.sh` script from scripts directory to see if testbench works.
4. Run the following command:
   
    `./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volume_out.raw -t ./id_02_tplg.bin -b S16_LE`

**Reproduction Rate**
All the time.

**Expected behavior**
If there is something wrong with topology file, the testbench should exit with some error code.

**Screenshots or console output**
1. Segmentation fault screenshot.

[image]

2. gdb trace for the segmentation fault.

[image]

cujomalainey, 2020-07-22:
We have no codeowners for tplg_parser, @ranj063, @xiulipan and @singalsu look like closest owners.

lgirdwood, 2020-07-23:
> We have no codeowners for tplg_parser, @ranj063, @xiulipan and @singalsu look like closest owners.

Ack - we should add some owners to CODEOWNERS. 
@ymdatta could our array index be too large ? (and does it come from topology ?)

ymdatta, 2020-07-23:
@lgirdwood: Yes, i think index size is too large. 

If you take a look at this [line](https://github.com/thesofproject/sof/blob/master/tools/tplg_parser/tplg_parser.c#L1305) from the ```tplg_parser.c```, the value of  ```array->num_elems``` was  ```826361680``` when checked via gdb. With this, i came to the above conclusion that index size is too large.

Yes, the array comes from the topology file. This can be be confirmed from this [line](https://github.com/thesofproject/sof/blob/master/tools/testbench/topology.c#L178).

singalsu, 2020-07-28:
Thanks for reporting! I think I can fix this.

singalsu, 2020-07-28:
I created PR #3246 . It returns now an error for the fuzzed topology.

```
./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volume_out.raw -t ./id_02_tplg.bin -b S16_LE
unknown ipc_init()
unknown edf_scheduler_init()
debug: loading widget PCM0P id 11
error: illegal array number of elements 826361680
error: parse comp tokens 164
error: load AIF IN failed
error: loading widget
error: parsing topology
```

cujomalainey, 2020-08-04:
Fixed

closed 2020-08-04 by cujomalainey