thesofproject/sof #3241: [BUG][jsl-004-drop-stable] IPC timeout on StreamTrigPause on Demux Capture Topology

zrombel, 2020-07-27:
**Describe the bug**
IPC timeout when sending StreamTrigPause to playback pipeline on topology with demux in capture pipeline.
```
    +-------------------------------+
    | +------+   +------+   +-----+ |                                                 +------+
    | | Host |-->| Buff |-->| Dai |-------------------------------------------------->| SSPx |
    | +------+   +------+   +-----+ |                                                 +------+
    +-------------------------------+                                                    |
                                                                                         |
                                                                                         |
        pipe_cap                                                                         |
    +----------------------------------------------------------------------------+       |
    | +-------+   +------+   +-----+   +------+   +-------+   +------+   +-----+ |    +------+
    | | Host1 |<--| Buff |<--| Vol |<--| Buff |<--| DeMux |<--| Buff |<--| Dai |<-----| SSPx |
    | +-------+   +------+   +-----+   +------+   +-------+   +------+   +-----+ |    +------+
    +-------------------------------------------------|--------------------------+
                                                      |
        pipe_cap                                      |
    +-------------------------------------------+     |
    | +-------+   +------+   +-----+   +------+ |     |
    | | Host2 |<--| Buff |<--| Vol |<--| Buff |<------+
    | +-------+   +------+   +-----+   +------+ |
    +-------------------------------------------+
```
**To Reproduce**
Run Slim Driver python test:
14_04_TestDemuxSynchronizedCapture48000Hz16b16b8ch
parameters:
--playback-iterations=100

**Reproduction Rate**
1/50

**Environment**
1) FW Branch: jsl-004-drop-stable, commit: 21c8900
2) FW Config: target: JSL, config: jasperlake_defconfig+ jasperlake_chrome_overridedefconfig
2) Test environment: brach: master, commit: 303aa33b3
2) Python test: 14_04_TestDemuxSynchronizedCapture48000Hz16b16b8ch --playback-iterations=100
3) Platform: JSL RVP

**LOGS**
[JSL DEMUX CAP IPC TIMEOUT.zip](https://github.com/thesofproject/sof/files/4981398/JSL.DEMUX.CAP.IPC.TIMEOUT.zip)

slawblauciak, 2020-07-28:
Please help to verify 24 bit and 32 bit formats.

zrombel, 2020-07-29:
From further observations: 
- bug can only be reproduced on JSL. 
- bug occurs only with 48kHz audio formats
- there is no error logged in dma_trace_log, but dma_trace_sink is filled with **timer_report_delay** and  **dtrace_add_event** logs

[dma_trace_1_sink.log](https://github.com/thesofproject/sof/files/4994251/dma_trace_1_sink.log)

jajanusz, 2020-08-03:
Fixed with https://github.com/thesofproject/sof/commit/ed35a46a5380e0a12fa2df1d54456f4ed3f77f8d

closed 2020-08-03 by jajanusz