thesofproject/sof #3243: [BUG] sof-logger -e option is broken: firmware/debug ABI confusion

plbossart, 2020-07-28:
**Describe the bug**

the sof-logger error checks are broken:

````
root@chrx:~# ./sof-logger -e -l sof-cht.ldc 
error: fw version in sof-cht.ldc file does not coincide with fw version in /sys/kernel/debug/sof/fw_version file.
error: version_file ABI Version is 3:17:0
error: snd ABI Version is 5:0:0
````
This is obvious confusion between firmware and debug ABI

**To Reproduce**
use sof-logger -e option

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
firmware master 1c53b8e9dd97f10a10bae81c15974f0ff1935699

plus added patch to show the issue:

````diff
diff --git a/tools/logger/convert.c b/tools/logger/convert.c
index 1f26e4411..94d24f29b 100644
--- a/tools/logger/convert.c
+++ b/tools/logger/convert.c
@@ -678,6 +678,19 @@ static int verify_fw_ver(const struct convert_config *config,
                log_err(config->out_fd,
                        "fw version in %s file does not coincide with fw version in %s file.\n",
                        config->ldc_file, config->version_file);
+
+               log_err(config->out_fd,
+                       "version_file ABI Version is %d:%d:%d\n",
+                       SOF_ABI_VERSION_MAJOR(ver.abi_version),
+                       SOF_ABI_VERSION_MINOR(ver.abi_version),
+                       SOF_ABI_VERSION_PATCH(ver.abi_version));
+
+               log_err(config->out_fd,
+                       "snd ABI Version is %d:%d:%d\n",
+                       SOF_ABI_VERSION_MAJOR(snd->version.abi_version),
+                       SOF_ABI_VERSION_MINOR(snd->version.abi_version),
+                       SOF_ABI_VERSION_PATCH(snd->version.abi_version));
+
                return -EINVAL;
        }

````

ktrzcinx, 2020-07-28:
https://github.com/thesofproject/sof/pull/3195

lgirdwood, 2020-08-03:
@slawblauciak can we close now ?

marc-hb, 2020-08-04:
A pointer to the fix would not hurt.

slawblauciak, 2020-08-10:
@lgirdwood I'd leave it open until the PR gets merged.
@marc-hb apologies, it's the PR mentioned above, https://github.com/thesofproject/sof/pull/3195

slawblauciak, 2020-08-18:
@plbossart PR has been merged, can you please verify?

closed 2020-08-18 by plbossart