thesofproject/sof #3244: [BUG]Build error with GCC9 in smart_amp_test.c

xiulipan, 2020-07-28:
**Describe the bug**
A new build bug when using GCC9
```
/home/sof/work/sof.git/src/audio/smart_amp_test.c: In function 'smart_amp_set_config':
/home/sof/work/sof.git/src/audio/smart_amp_test.c:93:39: error: taking address of packed member of 'struct sof_abi_hdr' may result in an unaligned pointer value [-Werror=address-of-packed-member]
   93 |  cfg = (struct sof_smart_amp_config *)cdata->data->data;
```

**To Reproduce**
```
docker pull thesofproject/sof:20200728
docker tag thesofproject/sof:20200728 sof
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh cnl
```

**Reproduction Rate**
10/10

**Expected behavior**
The GCC build should pass

xiulipan, 2020-07-28:
@bkokoszx @mmaka1 I did not find a simple fix in code for this `-Werror=address-of-packed-member`. Any idea about how to solve the issue here?

paulstelian97, 2020-07-28:
Are you modifying the configuration in any way? If not, you might want to make an on-stack copy of the configuration (a new memcpy_s call).

You shouldn't take the address of any field that has alignof(field) > 1 if the embedding structure is packed. Thus the error itself is legit.

Or maybe you could make that final data[] array to be an array of chars. Not sure exactly where that structure is though (IPC structure? Ugh...)

PR #3233 (2020-08-06): smart amp: fix alignment warning of data passesed in
Fix GGC warning on aligned packed data.

Fixes #3244
closed 2020-08-06 by lgirdwood