thesofproject/sof #3247: [BUG][Fuzzing] Testbench: Segmentation fault due to accessing pointer without null pointer checks

ymdatta, 2020-07-28:
**Describe the bug**
This bug was found when fuzzing the testbench using AFL fuzzer. For a fuzzed topology file input (i.e this topology was generated by the fuzzer) testbench exits with segmentation fault(core dumped) error.

**To Reproduce**
1. Build the testbench from the using `host-build-all.sh` script from scripts directory.
2. Download the topology file from [here](https://github.com/ymdatta/sof_testbench_fuzz_files/blob/master/id_26_tplg.bin).
3. Run the `host-testbench.sh` script from scripts directory to see if testbench works.
4. Run the following command:

```   
./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volume_out.raw -t ./id_26_tplg.bin -b S16_LE
```

**Reproduction Rate**
All the time.

**Expected behavior**
If there is something wrong with topology file, the testbench should exit with some error code.

**Screenshots or console output**

1. Segmentation Fault.
```
root@52607466acdc:/home/sof/work/sof.git# ./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testben[16/784]
e_out.raw -b S16_LE -t ./id_26_tplg.bin
unknown ipc_init()                  
unknown edf_scheduler_init()           
debug: loading widget PCM0P id 11
unknown comp new (null) type 1 id 1.0                                                           
debug: loading widget PGA1.0 id 4                                                               
unknown comp new (null) type 5 id 1.1
unknown vol->initial_ramp = 20, vol->ramp = 2, vol->min_value = 41, vol->max_value = 65536
debug: loading widget BUF1.0 id 16
unknown buffer new size 0xc00 id 1.2 flags 0x0
debug: loading widget BUF1.1 id 16
unknown buffer new size 0xc00 id 1.3 flags 0x0
debug: loading widget SSP5.OUT id 13
unknown comp new (null) type 2 id 1.4
debug: loading widget PIPELINE.1.SSP5.OUT id 17
unknown pipeline new pipe_id 1 period 1000 priority 0
loading route PCM0P -> BUF1.0                   
unknown connect buffer 2 as sink
loading route BUF1.0 -> PGA1.0
unknown connect buffer 2 as source
loading route PGA1.0 -> BUF1.1
unknown connect buffer 3 as sink
loading route BUF1.1 -> SSP5.OUT
unknown connect buffer 3 as source
unknown pipeline complete, clock freq 0Hz
<malloc version="1">
<heap nr="0">
<sizes>
</sizes>
<total type="fast" count="0" size="0"/>
<total type="rest" count="0" size="0"/>
<system type="current" size="135168"/>
<system type="max" size="135168"/>
<aspace type="total" size="135168"/>
<aspace type="mprotect" size="135168"/>
</heap>
<total type="fast" count="0" size="0"/>
<total type="rest" count="0" size="0"/>
<total type="mmap" count="0" size="0"/>
<system type="current" size="135168"/>
<system type="max" size="135168"/>
<aspace type="total" size="135168"/>
<aspace type="mprotect" size="135168"/>
</malloc>
unknown pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 48000
unknown pipe params stream_tag 6 channels 2 sample_valid_bytes 2 sample_container_bytes 2
unknown file_hw_params()
unknown file_params()
unknown file_params()
unknown pipe prepare
unknown file_prepare()
unknown file_prepare()
unknown pipe trigger cmd 1
unknown file_trigger()
unknown file_trigger()
Segmentation fault (core dumped)
```

2. gdb trace log

```
Program received signal SIGSEGV, Segmentation fault.
sa_set_panic_on_delay (enabled=false) at /home/sof/work/sof.git/src/include/sof/lib/agent.h:48
48                      atomic_sub(&sa->panic_cnt, 1);
(gdb) bt
#0  sa_set_panic_on_delay (enabled=false) at /home/sof/work/sof.git/src/include/sof/lib/agent.h:48
#1  pipeline_schedule_copy (p=0x555555773f80, start=0) at /home/sof/work/sof.git/src/audio/pipeline.c:1104
#2  0x00007ffff7622237 in pipeline_schedule_triggered (cmd=1, ctx=0x7fffffffe0e0) at /home/sof/work/sof.git/src/audio/pipeline.c:766
#3  pipeline_trigger (p=p@entry=0x555555773f80, host=host@entry=0x5555557711b0, cmd=cmd@entry=1) at /home/sof/work/sof.git/src/audio/pipeline.c:815
#4  0x000055555555937b in tb_pipeline_start (ipc=<optimized out>, ipc_pipe=<optimized out>, tp=<optimized out>) at /home/sof/work/sof.git/tools/testbench/common_test.c:85
#5  0x00005555555572ce in main (argc=<optimized out>, argv=<optimized out>) at /home/sof/work/sof.git/tools/testbench/testbench.c:260
```

ymdatta, 2020-07-28:
Some additional information: 

My reasoning to why this happens is because pointer  `sa` from this line [here](https://github.com/thesofproject/sof/blob/master/src/include/sof/lib/agent.h#L43) is not checked whether it's a null pointer or not. So, when it's a null pointer, since we are accessing the pointers contents in the subsequent lines, this leads to segmentation fault.

ymdatta, 2020-07-28:
@cujomalainey , @singalsu : Please take a look at this.

cujomalainey, 2020-07-28:
Looks like we have our first SOF failure :+1: 
No code owner for that file, blame shows @tlauda as probably best person to look at this

cujomalainey, 2020-07-28:
Looks like we are not handling errors in sa_init which in theory could cause a NULL pointer https://github.com/thesofproject/sof/blob/master/src/lib/agent.c#L85

slawblauciak, 2020-08-04:
@cujomalainey it's not possible to receive a NULL pointer when allocating on the system zone, should an allocation there fail, an immediate DSP panic would occur.

slawblauciak, 2020-08-04:
This is a possible fault in topology parsing code in test bench.

cujomalainey, 2020-08-04:
@slawblauciak does that apply for the testbench? If not we should handle that case so the fuzzer can push deeper.

slawblauciak, 2020-08-04:
Testbench has its own topology parser, which is based on the one used in the Linux driver. Having this in mind, it is a testbench specific issue.

cujomalainey, 2020-08-04:
Interesting, because based on the stacktrace my thoughts were that we simply drained all the memory for the app and the malloc for sa failed as a result and we just don't have a path to catch that in the DSP given the above handling you mentioned.

singalsu, 2020-08-26:
Thanks for reporting, I just repeated the issue with c0b969d784ee77646e1235c66ce8a70371d57925 .

singalsu, 2020-08-26:
This seems to be possible by adding sa_init() and dependencies to testbench build.

PR #3352 (2020-08-27): Tools: Testbench: Add system agent initialization
This patch fixes a segfault with null sof->sa pointer in
pipeline_schedule_copy() function. The bug has not been observed
with test topologies for processing pipelines but with other
topologies with fuzzer varied parameters.

The sa_init() is added to tb_pipeline_setup(). The scheduler_init_edf()
is a dependency. The two dummy functions platform_timer_get() and
clock_ms_to_ticks() are also added as dependency. The CMakelist for
library is edited to include agent.c build for testbench.

Fixes #3247
commit 40fd801c (2020-08-27): Tools: Testbench: Add system agent initialization

This patch fixes a segfault with null sof->sa pointer in
pipeline_schedule_copy() function. The bug has not been observed
with test topologies for processing pipelines but with other
topologies with fuzzer varied parameters.

The sa_init() is added to tb_pipeline_setup(). The scheduler_init_edf()
is a dependency. The two dummy functions platform_timer_get() and
clock_ms_to_ticks() are also added as dependency. The CMakelist for
library is edited to include agent.c build for testbench.

Fixes #3247
closed 2020-08-27 by lgirdwood