thesofproject/sof #3259: [bug][JSL] DMIC48kHz no  volume control in alsamixer

Liviali155, 2020-07-30:
**Describe the bug**
DMIC48kHz no  volume control in alsamixer

**To Reproduce**
1."sudo reboot" to reboot sysytem
2.alsamixer

```
ubuntu@ubuntu-Waddledoo:~/sof-test/test-case$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofda7219max983 [sof-da7219max98360a], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofda7219max983 [sof-da7219max98360a], device 5: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofda7219max983 [sof-da7219max98360a], device 6: DMIC16k (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**Reproduction Rate**
10/10

**Expected behavior**
DMIC48kHz has  volume control in alsamixer

**Impact**
DMIC48kHz no  volume control in alsamixer

[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev(797eb03a)
    * SOF:https://github.com/thesofproject/sof/tree/releases/jsl/v4.0-rc1(21c8900f)
 
Topology:sof-jsl-da7219-mx98360a.tplg
Name of the platform(s) on which the bug is observed.
Platform: JSL  Chromebook  with codec da7219max98360a   in i2s mode

[dmesg0729.log](https://github.com/thesofproject/linux/files/4993420/dmesg0729.log)
[sof-logger0729.log](https://github.com/thesofproject/linux/files/4993422/sof-logger0729.log)

mengdonglin, 2020-07-30:
@yongzhi1 Can you please check if this volume control is needed for JSL Chromebook? If yes, please help add volume control into topology.

udaymb, 2020-07-30:
On JSL chrome waddledoo, mic is seen

alsamixer -c 0 
capture tab screen shot - mic is seen
[image]

amixer -c 0 command response
[amixer-card0.log](https://github.com/thesofproject/sof/files/5000640/amixer-card0.log)

SOF: https://github.com/thesofproject/sof/tree/releases/jsl/v4.0-rc1(21c8900f)
Kernel: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.4

yongzhi1, 2020-07-30:
Hi @xiulipan 

Is passthrough capture used here for a special purpose? can we change it to volume?
https://github.com/thesofproject/sof/blob/master/tools/topology/platform/intel/intel-generic-dmic-kwd.m4#L54

xiulipan, 2020-07-31:
@yongzhi1 Yes, you can. It is removed as we used to use VOL comp for format conversation. Later we have some update to let DAI to have format conversation, so we remove the VOL in KWD pipeline by default.

As we have request for volume change in KWD pipeline, I have re-add those feature. Please take reference in this PR https://github.com/thesofproject/sof/pull/2405

You only need to change `define(KFBM_TYPE, `kfbm')` to  `define(KFBM_TYPE, `vol-kfbm')` if you think VOL comp is needed.

yongzhi1, 2020-07-31:
Hi, @xiulipan 

Added a make option for DMIC01 in the same spirit:
https://github.com/thesofproject/sof/pull/3266

yongzhi1, 2020-08-03:
Hi, @mengdonglin and @Liviali155, with #3266 and -DDMICPROC=eq-iir-volume option, we can add the following volume control:

localhost ~ # amixer -c0 cget name='Dmic0 Capture Volume'
numid=91,iface=MIXER,name='Dmic0 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=50,50
  | dBscale-min=-50.00dB,step=1.00dB,mute=1

From review comments of #3266, Google prefers pass-through w/o reprocessing, can we close the issue?

mengdonglin, 2020-08-04:
Close this issue, as the test result is aligned with customer request.

closed 2020-08-04 by mengdonglin