thesofproject/sof #3265: [BUG] error trace debugfs entry (etrace) is always empty

keyonjie, 2020-07-31:
The error trace debugfs entry (/sys/kernel/debug/sof/etrace) is always empty which lead to we can't get any error trace with sof-logger (sof-logger -l *.ldc), this is bad for debugging when DMA trace doesn't work because of FW hanged or crashed.

**To Reproduce**
Do anything can lead to FW reporting tr_err() or comp_err(), and check the "sof-logger -l *.ldc".

**Reproduction Rate**
100%

**Expected behavior**
We should see the error trace logs in sof-logger output.

**Impact**
This will make it difficult for FW debugging when FW is hang or crashed.

**Environment**
Any Intel platforms

paulstelian97, 2020-07-31:
Either this is a regression (I have used it on i.MX8 initially before adding the DMA trace support -- heck, before adding the host DMA in the first place) or something is different between Intel and NXP platforms.

plbossart, 2020-07-31:
We should really have a first message to check that the trace works. An empty trace is not-testable, it can mean the trace is not working or there is no error message. An 'Hello SOF" seems required for CI/daily tests to check the trace functionality, no?

@mmaka1 @xiulipan @marc-hb FYI

marc-hb, 2020-07-31:
@keyonjie does revert #3245 help?

Related, test fixes: https://github.com/thesofproject/sof-test/issues/297 https://github.com/thesofproject/sof-test/issues/298

slawblauciak, 2020-08-04:
Issue not apparent when testing with the slim driver, maybe a kernel regression?

dbaluta, 2020-08-04:
Cc: @IulianOlaru249 @crazoes 

I also noticed this last week but I thought it's just me being unable to use sof-logger properly.

lyakh, 2020-08-04:
I hardly remember seeing error trace data in "etrace" ever. Looking through both kernel and firmware sources it seems like the kernel expects two memory window regions - SOF_IPC_REGION_DEBUG for debug and SOF_IPC_REGION_TRACE for errors, in the firmware e.g. on CAVS platforms they are initialised by `EXT_MAN_PORT()` in src/platform/intel/cavs/platform.c - so far so good. But I don't see the DEBUG region being used. Both `_trace_error_with_ids()` and `_trace_event_with_ids()` seem to translate to the same code in the end. I also checked one of daily tests on ICL from yesterday, where there is an "ERROR" entry in debug trace from `hda_dma_link_check_xrun()` where `tr_err()` is called, but "etrace" is empty. So, it seems like a global error to me.

lgirdwood, 2020-08-04:
> We should really have a first message to check that the trace works. An empty trace is not-testable, it can mean the trace is not working or there is no error message. An 'Hello SOF" seems required for CI/daily tests to check the trace functionality, no?
> 
> @mmaka1 @lgirdwood @xiulipan @marc-hb FYI

Ack - "Hello SOF". @marc-hb do you fancy making this change, it should be a one liner for each platform in the FW ready function to add a trace_error call with the above string..

xiulipan, 2020-08-05:
@lgirdwood I am OK with this!
@marc-hb And I think you can also make patches to sof-test scripts to make a valid test for error logger now.

lyakh, 2020-08-05:
@xiulipan @akloniex so how is the error trace supposed to work and when did it work last time? I looked at v1.0 and v1.1 and even there I couldn't  so far find that out.

xiulipan, 2020-08-06:
@lyakh To understand how the trace work, we need to know we have 2 kinds of trace: 

1.  error trace: a simple shared memory window between DSP and host (same as the IPC mailbox), which means this kind of trace can only be directly access when DSP in D0. To make it work with PM, we have some method to cache this filed into debugfs when enter suspend, but I think it is not prefect at now and will lose data in some case (tplg load fail message will lost becaise of unfinished multiple core support feature)
2. DMA trace: something like a capture stream. We have set up buffer in debugfs to store data copied by DMA from DSP. As the buffer is limited size, so it can only hold limit size logs in the past. But if you open the sof-logger before the test, it can record all logs.

To understand more about error trace, we need first to know where is the buffer. Following code take APL as example:
We have about 8KB buffer for error log:
https://github.com/thesofproject/sof/blob/cf74566a331c5ee5d0878cccaee4df31de3b5eca/src/platform/apollolake/include/platform/lib/memory.h#L175-L181
https://github.com/thesofproject/sof/blob/cf74566a331c5ee5d0878cccaee4df31de3b5eca/src/platform/intel/cavs/include/cavs/lib/mailbox.h#L38-L40
Function for trace written to mailbox:
https://github.com/thesofproject/sof/blob/cf74566a331c5ee5d0878cccaee4df31de3b5eca/src/trace/trace.c#L68-L97

I am also look into this issue, it seems we have some error in trace implement.

lyakh, 2020-08-06:
> @lyakh To understand how the trace work, we need to know we have 2 kinds of trace:
[snip]
> I am also look into this issue, it seems we have some error in trace implement.

@xiulipan thanks for the explanation, that's also my impression, that the current implementation isn't right and I don't understand whether and when and how it ever worked in the past.

xiulipan, 2020-08-06:
@lyakh I did some debug and test, root cause to be we never copy anything into mailbox even if the trace is a ERROR. But all the functionality for MBOX trace is still OK. Have a simple fix: https://github.com/thesofproject/sof/issues/3265

xiulipan, 2020-08-06:
@plbossart @lgirdwood Now all trace is refined to link with comp ctx and uuid, I found it is not that easy to just add one line **Hello SOF** directly into our code.
Any idea here? Which comp we can used here to let the hello message linked to?

paulstelian97, 2020-08-06:
> @plbossart @lgirdwood Now all trace is refined to link with comp ctx and uuid, I found it is not that easy to just add one line **Hello SOF** directly into our code.
> Any idea here? Which comp we can used here to let the hello message linked to?

I typically use IPC for such messages in my internal tests that don't have an obvious component.

dbaluta, 2020-08-06:
@xiulipan thanks a lot for your information about traces. This is the kind of information we need to write it in sof-docs.

lyakh, 2020-08-06:
> @lyakh I did some debug and test, root cause to be we never copy anything into mailbox even if the trace is a ERROR. But all the functionality for MBOX trace is still OK. Have a simple fix: #3278 

@xiulipan nice, just as I said - we never actually copy the data to the mailbox, no idea how it has been tested with the slim driver... Wondering whether it ever worked and since when it broke down.

closed 2020-08-06 by lgirdwood

xiulipan, 2020-08-07:
@lyakh I tried to figure out why and when the trace is broken, but no luck because of the refinement
@dbaluta Yes, I think we may missed this in sof-docs. Would you like to review my comments above and add them into doc?