thesofproject/sof #3271: [BUG][tgl][v7.0-rc10]"error: ERROR kpb_init_draining(): not enough data in history buffer"

RDharageswari, 2020-08-05:
**Describe the bug**
While integrating the Google Hotword module, the streaming doesnt start after the Key Phrase is detected.
arecord -Dhw:0,100 -M -N -r 16000 -c2 -f s16_le --buffer-size=96000 -vvv /tmp/wov.wav -d10
At times when I say “OK Google” the detection works and the capture also starts well, but this is not the case always.
There are scenarios where even if I say “OK Google” the capture doesn’t start. I took the fw & driver logs.
From the logs, it looks like Hotword is detected, but there is some issue in draining the data.

**To Reproduce**

 Checkout the SOF fw branch:
1.https://github.com/bzhg/sof/commits/hotword-tgl-006
2. Build SOF for TGL + Google module with xcc
3. Find the mixer control and set the language model on DUT:
./sof-ctl -n 73 -t 1 -b -r -s language_model
4.Open the WoV PCM device for listening:
arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s16_le --buffer-size=96000 -vvv /tmp/wov.wav
Say "OK Google" and streaming should start.

**Reproduction Rate**
At times it works for the first/second time, but fails after that

**Expected behavior**
Say "OK Google" and streaming should start.

**Impact**
Blocked for the integration

**Environment**
Kernel: 5.4 kernel + https://github.com/thesofproject/linux/pull/2276/commits/19912c1c25f2bad5019fefb00bdd1c4faaec0467
Fw: https://github.com/thesofproject/sof/commits/releases/tgl/v7.0 + patches from Google. (https://github.com/bzhg/sof/commits/hotword-tgl-007 )
Topology, kcontrol, logger all built from v7.0

Fw build Steps
make tigerlake_defconfig 
cp ../src/arch/xtensa/configs/override/tigerlake_chrome.config override.config 
make override.config 
make bin -j10

**Screenshots or console output**
[235412854.322917] (  2236586.250000) c0 kd-test      10.53 /google_hotword_detect.c:263  Hotword detected 2000ms
[235412863.906250] (        9.583333) c0 kd-test      10.53 /google_hotword_detect.c:60   notify_host()
[235412878.802083] (       14.895833) c0 kd-test      10.53 /google_hotword_detect.c:69   notify_kpb()
[235412888.437500] (        9.635417) c0 kpb          9.48           src/audio/kpb.c:915  kpb_event_handler(): received event with ID: 2 
[235412898.072917] (        9.635417) c0 kpb          9.48           src/audio/kpb.c:1010 kpb_init_draining(): requested draining of 2000 [ms] from history buffer
[235412905.937500] (        7.864583) c0 kpb                         src/audio/kpb.c:1021 ERROR kpb_init_draining(): not enough data in history buffer
[249711068.333333] ( 14298162.000000) c0 ll-schedule        ./schedule/ll_schedule.c:334  task add 0x9e120080 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690>
[249711096.562500] (       28.229166) c0 ll-schedule        ./schedule/ll_schedule.c:336  task params pri 4 flags 0 start 0 period 500000

RDharageswari, 2020-08-05:
[fw_log.txt](https://github.com/thesofproject/sof/files/5025459/fw_log.txt)
[wov_failure_dmesg.txt](https://github.com/thesofproject/sof/files/5025461/wov_failure_dmesg.txt)

ranj063, 2020-08-05:
@RDharageswari just to clarify, there's no S0ix in the picture here? keyword detection fails even in S0?

RDharageswari, 2020-08-05:
yes it fails on S0 itself. I tried with & without gain module also, still i see the same behavior.

RDharageswari, 2020-08-05:
i verified GHWD for more than 10times and it worked with the below steps:
arecord -Dhw:0,100 -M -N -r 16000 -c2 -f s16_le –buffer-size=67200 -vvv /tmp/wov.wav -d10
wait for 10sec // this is must! If you don’t wait and say “ok google” things will not work.
Say “OK Google”, the recording will start. Now Check the /tmp/wov.wav for captured data

Additionally, there was also a change to the build steps i had used:
make tigerlake_defconfig
cp ../src/arch/xtensa/configs/override/tigerlake_chrome.config override.config
make override.config // modified this to overrideconfig.
make bin -j10

Thanks @ranj063 for helping to debug!

closed 2020-08-05 by RDharageswari