thesofproject/sof #3272: [BUG] WoV cannot be triggered with some specific sets of history_depth and preamble_time

keqiaozhang, 2020-08-05:
**Describe the bug**
When testing wov with different values of history depth and preamble time, we found that most combinations can work, but we also found that with some certain values of preamble time and history depth, wov cannot be triggered. 
For example, below two sets do not work.

Preamble-time =2000
History-depth =2000
Or 
preamble-time =1900
History-depth=1900

I haven't traversed all the possible combinations yet，but these two sets work with 003 release firmware. it's looks like a regression.

**To Reproduce**
1. change the history depth and preamble time to 2000 in config blob.
2. run wov:
    arecord -Dhw:0,6 -M -N -c 2 -f s24_le --buffer-size=1599840 -r 16000 -vvv  -d 10
3. try to trigger the wov pipeline.

**Reproduction Rate**
10/10

**Expected behavior**
wov can be triggered with all valid combinations of history depth and preamble time.

**Impact**
wov cannot be triggered 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/1b2f260
    * SOF: https://github.com/thesofproject/sof/tree/releases/jsl/v4.0-rc2
2) Name of the topology file
    * Topology: sof-jsl-da7219-mx98360a.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: JSL-chromebook

** Trace **
[error-trace.txt](https://github.com/thesofproject/sof/files/5026616/error-trace.txt)

slawblauciak, 2020-08-05:
In order to make WoV more stable on JSL platforms and fix issues stemming from memory pool limitations the history depth/preamble should not be longer than 1800

keqiaozhang, 2020-08-06:
Thanks @slawblauciak  for clarification. Due to the hardware restriction on JSL, this bug can be closed.

closed 2020-08-06 by keqiaozhang