thesofproject/sof #3274: [BUG] Platform & CPU timer should be distinct

jajanusz, 2020-08-05:
**Describe the bug**
`cpu_timer_get` and `timer_get` that get timers from `struct sof` should return distinct physical or virtual timers, so user can register handler in one of them with not overriding another one.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
100%

**Expected behavior**
I should get distinct timers or at least should be able to use them like they would be distinct.

**Impact**
Needed for implementation of budgeting in scheduler.
Some context: https://github.com/thesofproject/sof/pull/3268

**Environment**
IMX and pre-CAVS platforms (it works as expected only on CAVS platforms).

**Side note**
Probably doxygen should be more verbose and specify that these timers should be distinct.

jajanusz, 2020-08-05:
@mengdonglin please assign some1 who supports baytrail, thanks

lgirdwood, 2020-08-05:
@jajanusz ack - but the new API must map easily onto Zephyr clock APIs i..e naming and calling convention @finikorg has Zephyr clock support for all IP on his TODO list IIRC.

paulstelian97, 2020-08-05:
On the i.MX one, I think we only have one, so we should be able to somehow reuse part of the infrastructure for cascaded IRQs (the part that allows multiple handlers per same interrupt) within the firmware itself.

PR #3346 (2020-08-26): byt: use an internal timer for "cpu_timer"
Using the same hardware timer for both "cpu_timer" and
"platform_timer" doesn't work, switch the "cpu_timer" to use
TIMER1 on Baytrail. This also fixes performance counters on
the platform.

Fixes #3274
closed 2020-08-26 by lgirdwood