thesofproject/sof #3276: [BUG] How can I recover from DSP Panic?

EvanCarroll, 2020-08-05:
**Describe the bug**
User gets a DSP, not clear how to recover.

**To Reproduce**
Happens sporatically

**Reproduction Rate**
About once in an hour of use.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I am getting a dsp panic,

```
error : DSP panic!
status: fw entered - code 00000005
error: can't enter idle
error: trace point 00004000
error: panic at src/lib/agent.c:62
```

Then sound stops working and I can't get it back on without restarting. Is there any way to recover from this?

plbossart, 2020-08-05:
@EvanCarroll does this happen when you are continuously playing sound?
Or just when a sound gets played after some period of idleness? Just trying to correlate with other cases, we've seen this panic in other bug reports but haven't triangulated what might cause this.
Can you also point us to the results of alsa-info so that we know what platform this happens on?
@lgirdwood FYI

plbossart, 2020-08-05:
Possible duplicate of https://github.com/thesofproject/sof/issues/2828 ?

paulstelian97, 2020-08-06:
If SOF is built as a kernel module, removing and reinserting the module should also reset the DSP and get sound back on, for a workaround. No guarantees that it works (for some reason for i.MX there are issues in this flow?)

lgirdwood, 2020-08-06:
@slawblauciak @mengdonglin fyi.

ilia-kats, 2020-08-07:
I'm also getting hit by this. I usually notice it when trying to use audio (playback or mic) after several hours of not using audio, but today this happened right in the middle of playback. The sound suddenly just went silent and this was in the kernel log:
```
[263553.357824] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[263553.357837] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[263553.358092] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[263553.358096] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[263553.358100] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62
[263553.358104] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[263553.358109] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
[263553.358113] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d1fe EPC3     0x00000000 EPC4    0x00000000
[263553.358117] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[263553.358121] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[263553.358124] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[263553.358127] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05a110
[263553.358135] sof-audio-pci 0000:00:1f.3: 0xbe05a110: be05a140 00000001 be013760 00000001
[263553.358140] sof-audio-pci 0000:00:1f.3: 0xbe05a114: 00000000 00000000 0000003e be064400
[263553.358145] sof-audio-pci 0000:00:1f.3: 0xbe05a118: b1712600 c11fd8dd 000c0800 00000000
[263553.358149] sof-audio-pci 0000:00:1f.3: 0xbe05a11c: 0dead000 00000000 e14c6018 ffff9e17
[263553.358154] sof-audio-pci 0000:00:1f.3: 0xbe05a120: c1383044 ffffffff e41a3180 ffff9e17
[263553.358158] sof-audio-pci 0000:00:1f.3: 0xbe05a124: b1712600 c11fd8dd 00000000 00000000
[263553.358163] sof-audio-pci 0000:00:1f.3: 0xbe05a128: 004f7c48 ffffaac0 e41a3180 ffff9e17
[263553.358167] sof-audio-pci 0000:00:1f.3: 0xbe05a12c: 00000000 00000000 00000000 00000000
```
This is on a Thinkpad X1 Carbon 7th using SOF firmware 1.5.1 on Arch linux. Here is alsa-info output:
```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.65
!!################################

!!Script ran on: Fri Aug  7 12:13:18 UTC 2020

!!Linux Distribution
!!------------------

Arch Linux \r (\l) NAME="Arch Linux" PRETTY_NAME="Arch Linux" ID=arch HOME_URL="https://www.archlinux.org/" DOCUMENTATION_URL="https://wiki.archlinux.org/" SUPPORT_URL="https://bbs.archlinux.org/" BUG_REPORT_URL="https://bugs.archlinux.org/" LOGO=archlinux

!!DMI Information
!!---------------

Manufacturer:      LENOVO
Product Name:      20QES01M00
Product Version:   ThinkPad X1 Carbon 7th
Firmware Version:  N2HET43W (1.26 )
Board Vendor:      LENOVO
Board Name:        20QES01M00

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0003:00/status 	 15
/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/ACPI000E:00/status 	 15
/sys/bus/acpi/devices/INT0E0C:00/status 	 15
/sys/bus/acpi/devices/INT3400:00/status 	 15
/sys/bus/acpi/devices/INT3403:00/status 	 15
/sys/bus/acpi/devices/INT34BB:00/status 	 15
/sys/bus/acpi/devices/INT3F0D:00/status 	 15
/sys/bus/acpi/devices/LEN0100:00/status 	 15
/sys/bus/acpi/devices/LEN0111:00/status 	 15
/sys/bus/acpi/devices/LEN0268:00/status 	 15
/sys/bus/acpi/devices/LEN0300:00/status 	 15
/sys/bus/acpi/devices/LNXPOWER:00/status 	 1
/sys/bus/acpi/devices/LNXPOWER:01/status 	 1
/sys/bus/acpi/devices/LNXPOWER:02/status 	 15
/sys/bus/acpi/devices/LNXPOWER:03/status 	 1
/sys/bus/acpi/devices/LNXPOWER:04/status 	 1
/sys/bus/acpi/devices/LNXPOWER:05/status 	 1
/sys/bus/acpi/devices/LNXPOWER:06/status 	 1
/sys/bus/acpi/devices/LNXPOWER:07/status 	 1
[... 1502 lines cut ...]
/sys/class/sound/hwC0D2/user_pin_configs:

/sys/class/sound/hwC0D2/init_verbs:

/sys/class/sound/hwC0D2/hints:

!!ALSA/HDA dmesg
!!--------------

[263763.742460] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[263763.742470]  HDA Analog: ASoC: trigger FE cmd: 5 failed: -110
[263763.743310] wlp0s20f3: deauthenticating from xx:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
--
[263767.656572] sof-audio-pci 0000:00:1f.3: ASoC: platform prepare error: -22
[263767.656576]  HDA Analog: ASoC: prepare FE HDA Analog failed
[263767.670431] PM: hibernation: hibernation exit
--
[263768.691484] sof-audio-pci 0000:00:1f.3: ASoC: platform prepare error: -22
[263768.691489]  HDA Analog: ASoC: prepare FE HDA Analog failed
[263768.696893] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
--
[263768.696910] sof-audio-pci 0000:00:1f.3: ASoC: platform prepare error: -22
[263768.696915]  HDA Analog: ASoC: prepare FE HDA Analog failed
[263768.914744] audit: type=1111 audit(1596802281.974:1946): pid=743 uid=0 auid=4294967295 ses=4294967295 msg='op=radio-control arg="wwan-enabled" pid=808150 uid=0 result=success exe="/usr/bin/NetworkManager" hostname=? addr=? terminal=? res=success'
```

paulstelian97, 2020-08-07:
@ilia-kats I recommend that you move those very large logs into separate files and attach them to your comment (yeah, somehow it is possible to do that). Also, I recommend you open another issue with your specific panic details, so it can be checked.

Yikes, another agent panic (DSP load got too high and the agent, a component that keeps the load in check, saw that the DSP was overloaded and stopped everything).

EvanCarroll, 2020-08-07:
I've updated with the result of `alsa-info`. However, I'm more concerned about knowing if recovery is possible than fixing any single issue. This issue however also happens on the X1 Carbon 7th Gen.

plbossart, 2020-08-07:
@EvanCarroll In theory if you do a suspend operation the DSP context is lost and the firmware re-downloaded.
We do not have a recovery in place at the moment, it's been an ask for ages but we never got to it.
https://github.com/thesofproject/linux/issues/452

paulstelian97, 2020-08-07:
@tlauda Are you aware of any clock speed woes on KBL? Maybe that's why the agent is crying for these two and causing panics...

@EvanCarroll @ilia-kats I'd like some kernel boot logs (dmesg) from both of you to identify the exact system, topology, firmware version etc. It doesn't need to be the very log from the crashes (although it's always better), but they must have the same version (no updates or anything) as with the crash so we can identify it. There are several already known issues with the agent in older versions of the firmware (as far as I know, they've been patched in the current development version) and I'd like to know that it isn't one of those that you're hitting.

EvanCarroll, 2020-08-07:
I'll provide that next time it crashes, I can't guarantee my firmware hasn't already updated. I use `fwupdagent`.

paulstelian97, 2020-08-07:
Sure thing. Maybe the update already fixed the issue though. But if it didn't, you're welcome to post all the logs so that I (or, more importantly, the devs that know your platform specifically; I don't know more than some generalities about the Intel platforms) can look into it and identify the reason for the crash. Again, with exact information we can provide a proper solution or workaround.

EvanCarroll, 2020-08-08:
It just did it again. I was watching a movie and `xscreensaver` kicked in.

I will update the topic with the logs.

paulstelian97, 2020-08-09:
The only thing that is half-suspect is "FW ABI is more recent than kernel/topology ABI is more recent than kernel" (kernel is 3:13:0 while FW/topology are 3:16:0). See if you can somehow update the SOF kernel module (if you have a module) or the kernel itself (if it's built in). Not sure if it will actually help but I'd say it's worth a shot.

slawblauciak, 2020-08-26:
The FW should no longer panic like that in the upcoming 1.6 release.

mengdonglin, 2020-09-08:
Close the bug now. Recovery solution from DSP panic is a big topic and will not be tracked in this bug.

closed 2020-09-08 by mengdonglin