thesofproject/sof #3281: [FEATURE]Add an error trace message for functionality check

xiulipan, 2020-08-07:
**Is your feature request related to a problem? Please describe.**
Add a message like **Hello SOF** into error trace to show error trace works

**Describe the solution you'd like**
- create new component to show this messge
- Add this into one existing component 

**Additional context**
Discussion moved from https://github.com/thesofproject/sof/issues/3265

xiulipan, 2020-08-07:
@plbossart @lgirdwood @lyakh @paulstelian97 @marc-hb moved the discussion in #3265 here.
 We have 2 proposal:
- create new component to show this message (need to send the message after trace init)
- Add this into one existing component (eg. Add this message in IPC comp, but may show multiple time)

Please have some comments. Then we can refine out logger check test case to track error trace status.

paulstelian97, 2020-08-07:
I think the message could be shown in either the generic SOF init code or in the platform init code. I suggested IPC only because messages have to be tied to some component.

Or perhaps trace init itself could show a message (configurable, of course) or just dump the message into the mailbox as some sort of initialization.

xiulipan, 2020-08-07:
@paulstelian97 I tried to add this kind for message into trace init. It used to be very easy in past if we just send trace message into mailbox. But now the trace in mailbox is actually some address and date, then we need a ldc file to parse it out.

I will have try and RFC PRs

paulstelian97, 2020-08-07:
We don't need ldc file to verify that there is some message, only to actually decode it. You can put in some fixed parameters in the message and the parameters don't need the LDC file to check, if you know how many there should be beforehand.

lyakh, 2020-08-07:
> @plbossart @lgirdwood @lyakh @paulstelian97 @marc-hb moved the discussion in #3265 here.
> We have 2 proposal:
>
>     * create new component to show this message (need to send the message after trace [...]

Use `&sa_tr` from agent.c?

lgirdwood, 2020-08-07:
@xiulipan keep it simple. Just emit trace error level in platform_boot() for each platform.

marc-hb, 2021-04-27:
Was this fixed by commit 67a0a6937 / PR #3319?

marc-hb, 2021-04-28:
Assuming fixed and closing, re-open if needed.

closed 2021-04-28 by marc-hb