thesofproject/sof #3308: [BUG] Are "no bytes to {produce,consume}" cases actually errors?

lyakh, 2020-08-13:
`comp_update_buffer_produce()` and `comp_update_buffer_consume()` in buffer.c print an "information" severity message "no bytes to {produce,consume}" when called with `bytes == 0`. In many tests these messages accompany an error, in others, like those seen in CI tests and reported by @plbossart, execution seems to continue and it is unclear whether functionality actually was affected. Are there cases when these messages are indeed "harmless?" Shouldn't they always be treated as errors?

paulstelian97, 2020-08-13:
A case that isn't implemented but will become not-an-error will arrive with Compress support I guess. Or should the components explicitly not call the functions at all if bytes == 0?

dbaluta, 2020-08-13:
Maybe for PCM that there could be a problem, but as @paulstelian97 said for Compress we will be having this situation pretty often, where we need an amount of bytes greater than the current period so we will need to wait for the next period to get all bytes needed.

I wonder if we can change buf_info to bug_dbg for when this case happens.

lgirdwood, 2020-08-13:
@lyakh whilst these are sometimes  helpful, they can flood logs - can we make them debug as suggested by @dbaluta

closed 2021-06-11 by lgirdwood