thesofproject/sof #3310: [BUG][DSM][TGL] LSB rounding in S32_LE in smart_amp pipeline

aiChaoSONG, 2020-08-14:
**Describe the bug**
A nocodec topology is created to test DSM in nocodec mode: #3212.
With this topology, play a sine wave through DSM playback pipeline, and record it back through DSM capture pipeline.
The recorded data is binary same with playback data under S16_LE, but not binary same under S32_LE.
Under S32_LE, the LSB seems rounded. for example: 
    `9F 75 53 0D` becomes `00 76 53 0D` 
    `01 97 86 A6` becomes `00 97 86 A6`  (LSB-->MSB)

**To Reproduce**
1. compile and use topology in #3212  (SSP2 should be enabled for DSM)
2. run `arecord -Dhw:0,2 -f S32_LE -r 48000 -c 4 -v -q recorded.wav & aplay -Dhw:0,2 -f S32_LE -r 48000 -c 2 -v -q a-sine-wave.wav `
3. compare recorded.wav and a-sine-wave.wav. 
Note: zeros at the beginning of recorded.wav should be ignored. compare the real data.

**Expected behavior**
recorded wave and playback wave binary same.

**Impact**
LSB rounding By SOF will impact sound quality, the same mechanism as quantization error in ADC.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {355c0de4}
    * SOF: {04ba7ba4}
2) Name of the topology file
    * Topology: {#3212}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL_RVP with ubuntu 20.04}

playback wave:
[image]

captured wave, channel 1/2 (at that time, data from channel 3/4 are zero)
[image]

captured wave, channel 3/4
[image]

[playback_and_captured_wav.zip](https://github.com/thesofproject/sof/files/5072423/playback_and_captured_wav.zip)

paulstelian97, 2020-08-14:
Something in the pipeline may be 24 bits and that could be the reason why we notice this. The data looks oddly like S24_4LE format.

lgirdwood, 2020-08-14:
@singalsu any comments ?

singalsu, 2020-08-14:
Is there processing in the pipeline, e.g. volume? A plot of the topology would be nice to attach for this issue.

aiChaoSONG, 2020-08-17:
@singalsu There is no processing component, only BUF and MUXDEMUX.
[image]

aiChaoSONG, 2020-09-04:
24 valid bits is configured in the DSM SSP, that's the reason we get 32 bit data rounding to 24 bit.  but some codec do require 24 bit data, so close this for won't fix.

```
DAI_CONFIG(SSP, SMART_SSP_INDEX, SMART_BE_ID, SMART_SSP_NAME,
	SSP_CONFIG(DSP_B, SSP_CLOCK(mclk, SSP_MCLK, codec_mclk_in),
		      SSP_CLOCK(bclk, 12288000, codec_slave),
		      SSP_CLOCK(fsync, 48000, codec_slave),
		      SSP_TDM(8, 32, 15, 255),
		      SSP_CONFIG_DATA(SSP, SMART_SSP_INDEX, 24, 0, SMART_SSP_QUIRK)))
')

```

closed 2020-09-04 by aiChaoSONG