thesofproject/sof #3314: [BUG] First period_bytes are not copied by Host in COMP_COPY_ONE_SHOT mode to source buffer

dbaluta, 2020-08-17:
**Describe the bug**

When implementing the compress audio feature I noticed that first 768 bytes (one period) are not copied by Host via DMA to source buffer.

**To Reproduce**
This is not noticeable when running a simple playback, because first period very likely is very small (1/10 of a second).
One can notice this by printing the sink buffer before dma start copying data.

In our IMX specific case this is easily reproducible by printing the values of sink buffer in ```dummy_dma_copy_crt_elem``` function.

**Reproduction Rate**
Always.

**Expected behavior**
All the bytes from the initial input files should be copied via the audio pipeline to destination DAI.

**Impact**
While for PCM this is barely noticeable, for Compress it has a huge impact as we are not able to correctly parse the compress file header.

**Solution and questions**

For IMX which uses dummy DMA to copy data from AP to DSP, it looks like the following solution works:

```
diff --git a/src/audio/host.c b/src/audio/host.c
index 5ff22106..940d3547 100644
--- a/src/audio/host.c
+++ b/src/audio/host.c
@@ -284,10 +284,6 @@ static int host_copy_one_shot(struct comp_dev *dev)
 
        comp_dbg(dev, "host_copy_one_shot()");
 
-       /* update first transfer manually */
-       if (!dev->position)
-               host_one_shot_cb(dev, hd->dma_buffer->stream.size);
-
        copy_bytes = host_get_copy_bytes_one_shot(dev);
        if (!copy_bytes) {
                comp_info(dev, "host_copy_one_shot(): no bytes to copy");

```

I wonder, what are the assumptions for using ONE_SHOT MODE.  I do not understand this comment 
```update first transfer manually```. It seems that this assume that ``someone`` already copied the first period bytes and now we only need to update our internal counters and current working elements.

dbaluta, 2020-08-17:
@tlauda @marc-hb @mrajwa @ktrzcinx @akloniex @lgirdwood any clues?

lgirdwood, 2020-08-17:
It could be that the first period is not being reported downstream ? I guess you are seeing a copy callback for it ? i.e. if you were to play 10 periods you should see 10 copies ?

dbaluta, 2020-08-17:
That's the strange thing, that I don't see a copy callback for the first period. Not sure who should call it for our Dummy DMA driver.

I only see that Host copy callback (for ONE_SHOT_MODE) assumes that the first period is copied from somewhere else.

```
host_copy_one_shot
   -> if (!dev->position) //first period
           host_one_shot_cb() //this updates the host counters, current positions
   -> dma_copy() period bytes

```

slawblauciak, 2020-08-25:
@dbaluta is this issue still relevant? The patch you mentioned earlier has been merged.

dbaluta, 2020-08-25:
@slawblauciak thanks for mentioning this. I'm closing the issue as it has been fixed.

closed 2020-08-25 by dbaluta