thesofproject/sof #3318: [BUG][TGL] unable to install params when aplay/arecord on DMIC16k in S24_LE/S32_LE

aiChaoSONG, 2020-08-18:
**Describe the bug**
On a TGL_RVP_SDW device, run `arecord -Dhw:0,4 -r 16000 -c 4 -f S24_LE -d 3 /dev/null -v -q`, you will see 'unable to install hw params', check dmesg, you get `ipc error for 0x60010000`, in sof-logger, see `ERROR volume_prepare(): sink buffer size 256 is insufficient < 2 * 256`

**To Reproduce**
run command: `arecord -Dhw:0,4 -r 16000 -c 4 -f S24_LE -d 3 /dev/null -v -q`

**Reproduction Rate**
100%

**Impact**
DMIC16k cannot support S24_LE and S32_LE

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev 69d2199b}
    * SOF: {releases/tgl/v8.0 d4e8548}
2) Name of the topology file
    * Topology: {sof-tgl-rt711-rt1308-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL_RVP_SDW}

@keyonjie confirmed that this issue is related with M4.
The DMIC said it only support S16_LE in sof-tgl-rt711-rt1308.m4:
```
PIPELINE_PCM_ADD(sof/pipe-volume-capture-16khz.m4,
        5, 4, CHANNELS, s16le,
        1000, 0, 0,
        16000, 16000, 16000)
DAI_ADD(sof/pipe-dai-capture.m4,
        5, DMIC, 1, dmic16k,
        PIPELINE_SINK_5, 2, s16le,
        1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)
```
But what actually in the `pipe-volume-capture-16khz.m4`:
```
PCM_CAPABILITIES(Passthrough Capture PCM_ID, `S32_LE,S24_LE,S16_LE',
	16000, 16000, PIPELINE_CHANNELS, PIPELINE_CHANNELS,
	2, 16, 192, 16384, 65536, 65536)
```

plbossart, 2020-08-18:
@singalsu can you look into this? I am not sure why some platforms use s16le for the dmic16kHz.
I vaguely recall it was the plan initially, but we've transitioned to support larger word sizes for kpb anyways, so wondering if we can align all dmic16k uses to rely on S32LE?

keyonjie, 2020-08-19:
Worth to try if the PR https://github.com/thesofproject/sof/issues/3318 can fix the issue.

slawblauciak, 2020-08-25:
@aiChaoSONG please try with the suggested patch.

aiChaoSONG, 2020-08-26:
Checked with #3324 on TGL_RVP_SDW, works perfectly.

aiChaoSONG, 2020-08-31:
close for fix(#3324) merged

closed 2020-08-31 by aiChaoSONG