thesofproject/sof #3322: [BUG] APL/CNL Build failed due to bigger SOF_SRC_HASH

fredoh9, 2020-08-18:
**Describe the bug**
Build failed with warnings for APL and CNL today in my local build. Not sure this is my env problem. I used to build FW well.

```
./scripts/xtensa-build-all.sh cnl
or
./scripts/xtensa-build-all.sh apl

[  2%] Building C object src/init/CMakeFiles/ext_manifest.dir/ext_manifest.c.o
cc1: warnings being treated as errors
/home/fredoh/work/sof/20200818/src/init/ext_manifest.c:35: warning: integer constant is too large for ‘long’ type
/home/fredoh/work/sof/20200818/src/init/ext_manifest.c:35: warning: large integer implicitly truncated to unsigned type
make[2]: *** [src/init/CMakeFiles/ext_manifest.dir/build.make:63: src/init/CMakeFiles/ext_manifest.dir/ext_manifest.c.o] Error 2

```

I checked SOF_SRC_HASH, it is 5 bytes, although src_hash variable is uint32_t.
```
$ grep -rn SOF_SRC_HASH
generated/include/version.h:7:#define SOF_SRC_HASH 0x96a474897
```

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
100%

**Expected behavior**
Build should be done without warning/error

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev
    * SOF: 96a474897a02b2af93e42894773e8e35c7632e9a

fredoh9, 2020-08-18:
Looks my case, it hit else part of below

```
# Calculate source hash value, used to check ldc file and firmware compatibility
if(EXISTS ${CMAKE_SOURCE_DIR}/.git/)
...
else()
        set(SOF_SRC_HASH ${GIT_LOG_HASH})
        message(WARNING "Source content hash can't be calculated, use GIT_LOG_HASH")
endif()
```

My current build tree is based on `git worktree add` and it has `.git` also.

jdrzaszcz, 2020-08-19:
+ TGL, ICL.
Fixed by #3323

PR #3331 (2020-08-24): cmake: set git hash to 8 digit
GIT_LOG_HASH can be more than 8 digits. Limit hash value up to
first 8 digits.

fixes: #3322
commit 7a10da09 (2020-08-24): cmake: set git hash to 8 digit

GIT_LOG_HASH can be more than 8 digits. Limit hash value up to
first 8 digits.

fixes: #3322
closed 2020-08-24 by lgirdwood