thesofproject/sof #3327: [BUG]no audio

quiricada, 2020-08-19:
dmesg | grep -i audio                                                                                                          [5/36]
[    0.213041] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)            
[    4.263234] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    4.263236] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.265210] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    5.430602] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.437057] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    5.458299] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    5.458303] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    5.458312] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    5.557029] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[    5.557115] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:16:0
[    5.557503] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    5.557597] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    5.562506] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:16:0
[    5.562598] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    5.564364] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    5.564465] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    5.566258] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    5.566359] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    5.567576] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    5.567676] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    5.568916] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    5.568955] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    5.568995] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    5.569032] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    5.569072] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    5.569108] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    5.569948] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    5.569985] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    5.570819] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    5.570857] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    5.571717] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    5.579441] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    5.594675] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[    5.594714] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.594744] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    5.594773] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    5.594799] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    5.594825] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    5.636662] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    5.636722] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten

inxi -F
System:    Host: lenovox1 Kernel: 5.8.1 x86_64 bits: 64 Desktop: Xfce 4.12.5 Distro: Slackware 14.2 
Machine:   Type: Laptop System: LENOVO product: 20R10015US v: ThinkPad X1 Carbon 7th  
           Mobo: LENOVO model: 20R10015US v: SDK0J40697  
Battery:   ID-1: BAT0 charge: 9.8 Wh condition: 46.6/51.0 Wh (91%) 
CPU:       Topology: 6-Core model: Intel Core i7-10710U bits: 64 type: MT MCP L2 cache: 12.0 MiB 
           Speed: 4699 MHz min/max: 400/4700 MHz Core speeds (MHz): 1: 4231 2: 4413 3: 4700 4: 4692 5: 4047 6: 3996 7: 4102 
           8: 4108 9: 4371 10: 4400 11: 4383 12: 4066 
Graphics:  Device-1: Intel driver: i915 v: kernel 
           Display: server: X.Org 1.20.8 driver: intel unloaded: fbdev,modesetting,vesa resolution: 3840x2160~60Hz 
           OpenGL: renderer: Mesa Intel UHD Graphics (CML GT2) v: 4.6 Mesa 20.1.5 
Audio:     Device-1: Intel driver: sof-audio-pci 
           Sound Server: ALSA v: k5.8.1 
Network:   Device-1: Intel Wireless-AC 9462 driver: iwlwifi 
           IF: wlan2 state: up mac: 04:ed:33:d1:e3:d6 
           Device-2: Intel Ethernet I219-V driver: e1000e 
           IF: eth2 state: down mac: 98:fa:9b:d1:c2:7b 
Drives:    Local Storage: total: 953.87 GiB used: 318.43 GiB (33.4%) 
           ID-1: /dev/nvme0n1 vendor: Intel model: SSDPEKKF010T8L size: 953.87 GiB 
Partition: ID-1: / size: 40.00 GiB used: 34.91 GiB (87.3%) fs: btrfs dev: /dev/nvme0n1p6 
           ID-2: /home size: 759.44 GiB used: 240.40 GiB (31.7%) fs: btrfs dev: /dev/nvme0n1p7 
Swap:      ID-1: swap-1 type: partition size: 16.00 GiB used: 0 KiB (0.0%) dev: /dev/nvme0n1p5 
Sensors:   System Temperatures: cpu: 59.0 C mobo: N/A 
           Fan Speeds (RPM): cpu: 3042 
Info:      Processes: 301 Uptime: 30m Memory: 15.30 GiB used: 1.18 GiB (7.7%) Shell: Bash inxi: 3.1.05

lgirdwood, 2020-08-20:
@kv2019i @plbossart fyi.

codewiz, 2020-08-20:
I *do* have functioning audio on the X1G7 using Fedora 32 (sof 1.5.1, alsa-ucm 1.2.3.2, linux 5.8.1, pulseaudio-13.99.1)

codewiz, 2020-08-20:
The DMIC, however, has a constant background noise, very loud.

kv2019i, 2020-08-20:
@codewiz There was a regression in 1.5 (see   https://github.com/thesofproject/sof/issues/2937 ) that should be fixed in 1.5.1. I wonder if you just have high gains in place for the dmic (check "alsamixer -Vcapture")? Otherwise FYI @singalsu and @juimonen we could have a new case.

@quiricada Some more details would be needed. Can you post "alsa-info.sh" output? Can you try a Ubuntu 20.04 or Fedora 32 live image (these are known to work on X1gen7 out-of-the-box). At first hand this seems like a config issue (kernel build options, firmware or something), but we'd need a lot more info to judge.

singalsu, 2020-08-24:
That's a strange problem. The left channel sounds clean and OK, while right channel is fairly white noise with only tiny amount of speech signal heard. Can you confirm this is captured directly from DMIC 48 kHz device without Pulseaudio? There's more elegant ways but I just temporarily rename /usr/bin/pulseaudio to /usr/bin/pulseaudio.disabled and kill all current instances.

singalsu, 2020-08-24:
@codewiz Could you please file a new issue for the noise in capture, we can continue there with that.