thesofproject/sof #3330: [BUG] [WHL] KD notification timeout on D0ix with chrome build.

jdrzaszcz, 2020-08-21:
**Describe the bug**
Kd notification timeout on WHL with chrome build on kd_with_dmic_d0ix test.
Test fail since CI build SOF_FW_NIGHLTY_CHROME_01.01.00.85 (commit: 04ba7ba4ba27432ad03c3908143343723aa04b31).
Last pass on SOF_FW_NIGHLTY_CHROME_01.01.00.84 (commit: 5b4b5532833b25efebfa587a844fa670c1422a31)

**Topology**
```
        +----------+   +------------------+
        |    OS    |-->|   AUDIO DEVICE   |-----------------------+
        +----------+   +------------------+                       |
                                                                  |
                   +-------------------------+                    |
        +--------+ | +---+   +-----+   +---+ |                    |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                  |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap        |
                   +-------------------------+ |                  |
                                               |                  |
                             +-----------------+                  |
        +--------------------|----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-| DMIC |---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +------+ |
        +-------------------------------------------------------+
```
**To Reproduce**
Run one of python test  from file:
test_05_kd_with_dmic_d0ix.py

**Reproduction Rate**
100 %

**Expected behavior**
Test should pass without kd notification timeout.

**Environment**
WHL RVP
CHROM BUILD from SOF master commit: 87e4f5aad80c22a1e82fa2bf78133640e39b8741

**Example logs**
[TestKdDmicD0ix16000Hz16b16b2ch_dma_trace_1.log](https://github.com/thesofproject/sof/files/5107479/TestKdDmicD0ix16000Hz16b16b2ch_dma_trace_1.log)
[TestKdDmicD0ix16000Hz24b32b2ch_dma_trace_1.log](https://github.com/thesofproject/sof/files/5107508/TestKdDmicD0ix16000Hz24b32b2ch_dma_trace_1.log)

[TestKdDmicD0ix16000Hz16b16b2ch_2020_08_21_09_11_56.zip](https://github.com/thesofproject/sof/files/5107474/TestKdDmicD0ix16000Hz16b16b2ch_2020_08_21_09_11_56.zip)
[TestKdDmicD0ix16000Hz24b32b2ch_2020_08_21_09_13_08.zip](https://github.com/thesofproject/sof/files/5107475/TestKdDmicD0ix16000Hz24b32b2ch_2020_08_21_09_13_08.zip)

slawblauciak, 2020-08-25:
@jdrzaszcz please verify on other platforms like TGL

jdrzaszcz, 2020-08-25:
@slawblauciak I checked it on TGL RVP platform and the issue doesn't occur there.

slawblauciak, 2020-09-10:
@jdrzaszcz can you verify if the bug is still reproducible on master? You will need https://github.com/thesofproject/sof/pull/3399 applied.

jdrzaszcz, 2020-09-10:
@slawblauciak issue didn't occur after applying pr: https://github.com/thesofproject/sof/pull/3399 on SOF master branch, commit: 64a1657bdd76858273a3e689c7ed20ff0c134029

slawblauciak, 2020-09-10:
Alright. Closing the issue.
That PR is only required to prevent an unrelated problem from occurring.

closed 2020-09-10 by slawblauciak