thesofproject/sof #3334: [BUG][Fuzzing] Testbench: Program exit with SIGFPE signal, Arithmetic exception

ymdatta, 2020-08-23:
**Describe the bug**
This bug was found when fuzzing the testbench using AFL fuzzer. For a fuzzed topology file input (i.e this topology was generated by the fuzzer) testbench exits with Floating point exception (core dumped) error.

**To Reproduce**
1. Build the testbench from the using `host-build-all.sh` script from scripts directory.
2. Download the topology file from [here](https://github.com/ymdatta/sof_testbench_fuzz_files/blob/master/id_fuzz2_82_tplg.bin).
3. Run the `host-testbench.sh` script from scripts directory to see if testbench works.
4. Run the following command:

```   
./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volume_out.raw -t ./id_fuzz2_82_tplg.bin -b S16_LE
```

**Reproduction Rate**
All the time.

**Expected behavior**
If there is something wrong with topology file, the testbench should exit with some error code.

**Screenshots or console output**

1. Floating point exception.
```
root@96ed1adedfaf:/home/sof/work/sof.git# ./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volum
e_out.raw -b S16_LE -t ./id_fuzz2_82_tplg.bin 
unknown ipc_init()                  
unknown edf_scheduler_init()           
debug: loading comp_id 0: widget PCM0P id 11
unknown comp new (null) type 1 id 1.0                                                           
debug: loading comp_id 1: widget PGA1.0 id 4                                                    
unknown comp new (null) type 5 id 1.1
unknown vol->initial_ramp = 84, vol->ramp = 2, vol->min_value = 41, vol->max_value = 65536
debug: loading comp_id 2: widget BUF1.0 id 16
unknown buffer new size 0xc00 id 1.2 flags 0x0
debug: loading comp_id 3: widget BUF1.1 id 16
unknown buffer new size 0xc00 id 1.3 flags 0x0
debug: loading comp_id 4: widget SSP5.OUT id 13
unknown comp new (null) type 2 id 1.4
debug: loading comp_id 5: widget PIPELINE.1.SSP5.OUT id 17
unknown pipeline new pipe_id 1 period 488 priority 0
loading route PCM0P -> BUF1.0
unknown connect buffer 2 as sink
loading route BUF1.0 -> PGA1.0
unknown connect buffer 2 as source
loading route PGA1.0 -> BUF1.1
unknown connect buffer 3 as sink
loading route BUF1.1 -> SSP5.OUT
unknown connect buffer 3 as source
unknown pipeline complete, clock freq 0Hz
<malloc version="1">
<heap nr="0">
<sizes>
</sizes>
<total type="fast" count="0" size="0"/>
<total type="rest" count="0" size="0"/>
<system type="current" size="135168"/>
<system type="max" size="135168"/>
<aspace type="total" size="135168"/>
<aspace type="mprotect" size="135168"/>
</heap>
<total type="fast" count="0" size="0"/>
<total type="rest" count="0" size="0"/>
<total type="mmap" count="0" size="0"/>
<system type="current" size="135168"/>
<system type="max" size="135168"/>
<aspace type="total" size="135168"/>
<aspace type="mprotect" size="135168"/>
</malloc>
unknown pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 48000
unknown pipe params stream_tag 17216 channels 2 sample_valid_bytes 2 sample_container_bytes 2
unknown file_hw_params()
unknown file_params()
unknown file_params()
unknown pipe prepare
unknown file_prepare()
Floating point exception (core dumped)
```
2. gdb trace log
```
Program received signal SIGFPE, Arithmetic exception.
0x00007f688bafdd8a in volume_prepare (dev=0x5612197d5e50) at /home/sof/work/sof.git/src/audio/volume/volume.c:869
869             cd->vol_ramp_frames = dev->frames / (dev->period / ramp_update_us);
(gdb) bt
#0  0x00007f688bafdd8a in volume_prepare (dev=0x5612197d5e50) at /home/sof/work/sof.git/src/audio/volume/volume.c:869
#1  0x00007f688c310836 in comp_prepare (dev=0x5612197d5e50) at /home/sof/work/sof.git/src/include/sof/audio/component_ext.h:188
#2  pipeline_comp_prepare (current=0x5612197d5e50, calling_buf=calling_buf@entry=0x5612197d6030, ctx=ctx@entry=0x7fff6c416330, dir=dir@entry=0)
    at /home/sof/work/sof.git/src/audio/pipeline.c:579
#3  0x00007f688c310e2f in pipeline_for_each_comp (dir=<optimized out>, ctx=0x7fff6c416330, current=0x5612197d51b0) at /home/sof/work/sof.git/src/audio/pipeline.c:184
#4  pipeline_comp_prepare (current=0x5612197d51b0, calling_buf=<optimized out>, ctx=0x7fff6c416330, dir=<optimized out>) at /home/sof/work/sof.git/src/audio/pipeline.c:583
#5  0x00007f688c315423 in pipeline_prepare (p=p@entry=0x5612197d7f80, dev=dev@entry=0x5612197d51b0) at /home/sof/work/sof.git/src/audio/pipeline.c:602
#6  0x000056121871ab8b in tb_pipeline_start (ipc=<optimized out>, ipc_pipe=<optimized out>, tp=<optimized out>) at /home/sof/work/sof.git/tools/testbench/common_test.c:82
#7  0x0000561218718752 in main (argc=<optimized out>, argv=<optimized out>) at /home/sof/work/sof.git/tools/testbench/testbench.c:305
```

ymdatta, 2020-08-23:
Some additional information that may help:

```
(gdb) p dev->frames
$8 = 24
(gdb) p dev->period
$9 = 488
(gdb) p pga->initial_ramp 
$10 = 84
```

From the logic at this point in src/audio/volume/volume.c: [#L860](https://github.com/thesofproject/sof/blob/master/src/audio/volume/volume.c#L860). 

```
ramp_update_us is set to VOL_RAMP_UPDATE_SLOW_US (which is 500 from include/sof/audio/volume.h)
```

In the above gdb log from issue description, substituting these values results in denominator (dev->period / ramp_update_us) becoming zero thus leading **division by zero** exception.

Hope this helps. Please feel free to comment if i have to include additional information.

ymdatta, 2020-08-23:
@cujomalainey @ranj063 : Please take a look at this crash found during fuzzing.

cujomalainey, 2020-08-23:
Excellent find, we should probably be running sanity checks in the component to make sure the values are valid before we use them.

ymdatta, 2020-08-23:
@cujomalainey : Since, this touches the core of SOF, i am waiting for suggestions on what kind of checks would be good to add here. But, i definitely agree on the point that we need to be adding checks in the code.

cujomalainey, 2020-08-23:
Yep, hence why I added @singalsu as he will be able to help identify the checks needed

singalsu, 2020-08-24:
Thanks for reporting, good catch! Need to think what is the proper fix for this. Rounded arithmetic might work for this case. If it's not feasible and for cases where fraction would not round up an error must be returned from prepare().

singalsu, 2020-08-26:
I'll do today the fix for this.

PR #3351 (2020-08-26): Audio: Volume: Avoid possible divide by zero trap
If in a topology the volume component ramp length (milliseconds) and
from it derived ramp update period is longer than pipeline copy()
schedule period the dev->frames (frames to process per per period) is
divided by zero. This error is not triggered by any of current
topologies.

The fix is to use dev->frames count as gain ramp update rate for
situations where the integer divider would round to zero. There is no
harm to update the gain at higher rate than suggested by topology.

Returning an error for such topologies also is not necessary since [... cut]
commit 009714bd (2020-08-26): Audio: Volume: Avoid possible divide by zero trap

If in a topology the volume component ramp length (milliseconds) and
from it derived ramp update period is longer than pipeline copy()
schedule period the dev->frames (frames to process per per period) is
divided by zero. This error is not triggered by any of current
topologies.

The fix is to use dev->frames count as gain ramp update rate for
situations where the integer divider would round to zero. There is no
harm to update the gain at higher rate than suggested by topology.

Returning an error for such topologies also is not necessary sinc [... cut]
closed 2020-08-26 by lgirdwood