thesofproject/sof #3335: [BUG] [Fuzzing] testbench program exits with segmentation fault

ymdatta, 2020-08-23:
**Describe the bug**
This bug was found when fuzzing the testbench using AFL fuzzer. For a fuzzed topology file input (i.e this topology was generated by the fuzzer) testbench exits with segmentation fault(core dumped) error.

**To Reproduce**
1. Build the testbench from the using `host-build-all.sh` script from scripts directory.
2. Download the topology file from [here](https://github.com/ymdatta/sof_testbench_fuzz_files/blob/master/id_fuzz2_90_tplg.bin).
3. Run the `host-testbench.sh` script from scripts directory to see if testbench works.
4. Run the following command:

```   
./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volume_out.raw -t ./id_fuzz2_90_tplg.bin -b S16_LE
```

**Reproduction Rate**
All the time.

**Expected behavior**
If there is something wrong with topology file, the testbench should exit with some error code.

**Screenshots or console output**

1. Segmentation Fault.
```
root@96ed1adedfaf:/home/sof/work/sof.git# ./tools/testbench/build_testbench/install/bin/testbench -r 48000 -R 48000 -i ./tools/test/audio/zeros_in.raw -o ./tools/testbench/build_testbench/volum
e_out.raw -b S16_LE -t ./id_fuzz2_90_tplg.bin 
unknown ipc_init()
unknown edf_scheduler_init()
debug: loading comp_id 0: widget PCM0P id 11
unknown comp new (null) type 1 id 1.0
debug: loading comp_id 1: widget PGA1.0 id 20
unknown comp new (null) type 8 id 1.1
unknown src_new()
debug: loading comp_id 2: widget BUF1.0 id 16
unknown buffer new size 0xc00 id 1.2 flags 0x0
debug: loading comp_id 3: widget BUF1.1 id 16
unknown buffer new size 0xc00 id 1.3 flags 0x0
debug: loading comp_id 4: widget SSP5.OUT id 12
unknown comp new (null) type 1 id 1.4
debug: loading comp_id 5: widget PIPELINE.1.SSP5.OUT id 17
unknown pipeline new pipe_id 1 period 1000 priority 0
loading route PCM0P -> BUF1.0
unknown connect buffer 2 as sink
loading route BUF1.0 -> PGA1.0
unknown connect buffer 2 as source
loading route PGA1.0 -> BUF1.0
unknown connect buffer 2 as sink
loading route BUF1.1 -> SSP5.OUT
unknown connect buffer 3 as source
unknown pipeline complete, clock freq 0Hz
Segmentation fault (core dumped)
```

2. gdb trace log

```
Program received signal SIGSEGV, Segmentation fault.
0x00007fef7a7d226c in pipeline_for_each_comp (dir=0, ctx=0x7ffce8515750, current=0x564e8f9707b0) at /home/sof/work/sof.git/src/audio/pipeline.c:162
162             struct list_item *buffer_list = comp_buffer_list(current, dir);
(gdb) bt
#0  0x00007fef7a7d226c in pipeline_for_each_comp (dir=0, ctx=0x7ffce8515750, current=0x564e8f9707b0) at /home/sof/work/sof.git/src/audio/pipeline.c:162
#1  pipeline_comp_complete (current=0x564e8f9707b0, calling_buf=<optimized out>, ctx=0x7ffce8515750, dir=0) at /home/sof/work/sof.git/src/audio/pipeline.c:213
#2  0x00007ffce8515750 in ?? ()
#3  0x0000000000000000 in ?? ()
```

ymdatta, 2020-08-23:
Additional information that might help along with gdb trace log from issue description:

```
(gdb) p dir
$11 = 0
(gdb) p current
$12 = (struct comp_dev *) 0x564e8f9707b0
(gdb) p *current
$13 = {state = 1, position = 0, frames = 0, pipeline = 0x564e8f972f70, min_sink_bytes = 0, min_source_bytes = 0, task = 0x0, size = 216, period = 1000, priority = 0, is_shared = false, 
  tctx = {uuid_p = 0, level = 3}, direction = 0, drv = 0x7fef7a1c5d40 <comp_src>, bsource_list = {next = 0x564e8f971070, prev = 0x564e8f971070}, bsink_list = {next = 0x564e8f971060, 
    prev = 0x564e8f971060}, priv_data = 0x564e8f970e50, comp = {hdr = {size = 76, cmd = 805371904}, id = 1, type = SOF_COMP_SRC, pipeline_id = 1, core = 0, ext_data_length = 0}}
```

ymdatta, 2020-08-23:
@cujomalainey @ranj063 : Please take a look at this crash found during fuzzing.

cujomalainey, 2020-08-23:
Hmm, this one looks tricky. Something is wonky with the pipeline

cujomalainey, 2020-08-24:
@tlauda you are last person to edit this code that is segfaulting, do you have any guesses what might be going on here?

juimonen, 2020-08-24:
@cujomalainey tlauda has left the company... I think @singalsu and @ranj063 are the best experts here.

cujomalainey, 2020-08-24:
@juimonen thanks for the heads up. Based on blame @lyakh might also be able to help.

ranj063, 2020-08-24:
@ymdatta looing at the error it seems like the SRC component is were we're seeing the segfault. But looking at the logs in the first comment, there's no logs graph elems for the connections BUF1.0 -> SRC -> 1.1 ?