thesofproject/sof #3336: [BUG] Unable to build topology

emilchudzik, 2020-08-24:
**Describe the bug**
Unable to build topology using script build-tools.sh. There is many errors from alsa lib.

**To Reproduce**
cd sof repo
./scripts/build-tool.sh

**Reproduction Rate**
100%

**Environment**
Branch: master
Commit: 2e07bb9d

**Screenshots or console output**
[image]

lgirdwood, 2020-08-24:
oh, I think we need a check for alsa-lib version inthe tobology cmake. IIUC it needs a recent commit in alsa-lib that is upstream but not part of the last release. @keyonjie fyi - we need a cmake check for this commit or later version of alsa-lib

emilchudzik, 2020-08-24:
with new alsa-lib, topology building is ok:
```
cd alsa-lib
git pull
./gitcompile
sudo make install
```

slawblauciak, 2020-08-25:
@emilchudzik good to close?

emilchudzik, 2020-08-25:
@slawblauciak yes, it can be close

closed 2020-08-25 by emilchudzik